fra589/grbl-Mega-5X #1: ERROR 9

hzxiaoyu, 2018-04-18:
I flashed this firmware at my mega2560 ramps1.4 board, successed, and connected with grbl control .It shows error 9 when I try to move any axis. What's wrong?

fra589, 2018-04-18:
error9 : G-code locked out during alarm or jog state

It seems that Grbl is not ready (homing not done or not in Idle stat)
Perhaps you need to disable homing and limits by setting $20, $21 and $22 to 0...
Please, post your config settings ($$ result) for more investigations.

@++;
Gauthier.

hzxiaoyu, 2018-04-18:
I tried ,and now the axis can work. But the machine coordinates shows 0 after I move any axis. And which control software do you use?  I use grbl controll only 4 axis.

fra589, 2018-04-18:
I used "Universal Gcode sender for the firsh tests" only 3 axes :-(... 
Then, Putty console mode connected to serial,
Actually, I am writing a new 5/6 axis Grbl control panel to implement all the grbl-Mega-5X capabilities...
It will be full open source under GPL licence and I think it will be soon available on github.com/fra589. I haven't make the choice of it's name, but interface begin to be operational for Jogging, manualy send gcode, sending gcode files and giving Grbl's status.
[image]
First version will be only in french :-S ... Be patient

liushuya7, 2018-04-18:
Looks AWESOME!

fra589, 2018-04-18:
@hzxiaoyu No, it work well, it's just that the screenshot was taken just after Grbl's init. This one was taken since it was in full 5 axis movment :
[image]
Tell me if you want to be pre-alpha tester... It work under Python 3, PyQT5 on Linux

hzxiaoyu, 2018-04-19:
Sorry, I use windows only,and don't know python.   : (
I still don't know how to make the coordinates  shows correct，maybe the grbl controller 3.6 is low? I flashed the older version 0.9 ,the coordinates shows but stepper not work because it is not for ramps.

fra589, 2018-04-19:
So, you need to wait for my first version to be available... It will be available both for Linux and Windows, but it's not ready for now.
I think it will be possible in few short weeks.
Watch for  https://github.com/fra589 :-)

hzxiaoyu, 2018-04-19:
Ok，I will keep waitting.

jflapao, 2018-04-19:
I don't know if any body read this but is arduino don't report the
coordinates of any axis try this $10=1, I have the same issues and work for
me

José Lapão

jflapao, 2018-04-20:
@fra598 hello i can be your alphatester, i can use booth linux and windows, i don't know python for programing, but i know install and configuring all the dependencies in booth OS (linux and windows) i can help you to translate to English and Portuguese (i'm from Portugal) so if you want send me a link for download

jflapao, 2018-04-20:
I know that is not the best way for direct comunication.

You version of linux can run on a ARM processor like raspberry pi or in android tablet and communicate with the machine by tcp/ip or even by Bluetooth?
If you have a big machine like my 1000x1000x200 mill in XYZ some time you need the screen and the controls near the job especially to adjust the zero points

fra589, 2018-04-21:
Hi @jflapao, I just put the first alpha (not versioned yet :smiley: ) on https://github.com/fra589/cn5X

The design is made to be capable to run on raspberry pi or equivalent with touchscreen of 1024x600. The goal is to have capability to be a cnc machine console.

See you on https://github.com/fra589/cn5X/issues

jflapao, 2018-04-21:
Hi @fra589 I have download you pre-alpha version, and only now I realise that work under python 3.0, with QT5, I have on my machine computer the python 2.7 and no QT, version, the bCNC only work under python 2.7 and don't need QT to work, I gona put on another computer with python 3 and the QT5 to test your software, then I give you a feedback.

PS: This maybe take a wille to do, I'm in a job and I'm short on time

hzxiaoyu, 2018-04-21:
By the way ,how to change the B axis to C axis in your firmware? My cam soft only have XYZAC ,not XYZAB

fra589, 2018-04-21:
Hi @jflapao I am also in a job, time is not that important :smiley: 
Hi @hzxiaoyu Choice of axis names in in my todo list of grbl-Mega-5X... I wil try to do something for you soon.

fra589, 2018-04-23:
Hi @hzxiaoyu Choice of axis names is now possible :smile: 
You can change B axis to C axis, in nuts_bolts.h :  
- In line 36, change `#define AXIS_NAMES "XYZAB"` to `#define AXIS_NAMES "XYZAC"`
- In line 50, change `#define AXIS_5_NAME 'B'` to `#define AXIS_5_NAME 'C'`
Tell me if this work well for you.
@++;
Gauthier.

fra589, 2018-04-28:
Hi @liushuya7 ,
You can see new settings for updated grbl-Mega-5X in the updated settings.md file in the "doc" directory :
https://github.com/fra589/grbl-Mega-5X/blob/edge/doc/markdown/settings.md#2--step-port-invert-mask

| Setting Value | Mask | Invert X | Invert Y | Invert Z | Invert A | Invert B |
|:-------------:|:--------:|:--------:|:--------:|:--------:|:--------:|:--------:|
| 0 | 00000000 | N | N | N | N | N |
| 1 | 00000001 | Y | N | N | N | N |
| 2 | 00000010 | N | Y | N | N | N |
| 3 | 00000011 | Y | Y | N | N | N |
| 4 | 00000100 | N | N | Y | N | N |
| 5 | 00000101 | Y | N | Y | N | N |
| 6 | 00000110 | N | Y | Y | N | N |
| 7 | 00000111 | Y | Y | Y | N | N |
| 8 | 00001000 | N | N | N | Y | N |
| 9 | 00001001 | Y | N | N | Y | N |
| 10 | 00001010 | N | Y | N | Y | N |
| 11 | 00001011 | Y | Y | N | Y | N |
| 12 | 00001100 | N | N | Y | Y | N |
| 13 | 00001101 | Y | N | Y | Y | N |
| 14 | 00001110 | N | Y | Y | Y | N |
| 15 | 00001111 | Y | Y | Y | Y | N |
| 16 | 00010000 | N | N | N | N | Y |
| 17 | 00010001 | Y | N | N | N | Y |
| 18 | 00010010 | N | Y | N | N | Y |
| 19 | 00010011 | Y | Y | N | N | Y |
| 20 | 00010100 | N | N | Y | N | Y |
| 21 | 00010101 | Y | N | Y | N | Y |
| 22 | 00010110 | N | Y | Y | N | Y |
| 23 | 00010111 | Y | Y | Y | N | Y |
| 24 | 00011000 | N | N | N | Y | Y |
| 25 | 00011001 | Y | N | N | Y | Y |
| 26 | 00011010 | N | Y | N | Y | Y |
| 27 | 00011011 | Y | Y | N | Y | Y |
| 28 | 00011100 | N | N | Y | Y | Y |
| 29 | 00011101 | Y | N | Y | Y | Y |
| 30 | 00011110 | N | Y | Y | Y | Y |
| 31 | 00011111 | Y | Y | Y | Y | Y |

@++;
Gauthier

hzxiaoyu, 2018-04-28:
@fra589 Hi, I havn't test yet ,because now I am working on xyz axis frame.
Today I just found when I homing X axis, and I pull on the endstop, the axis stop moving back a little and show alarm 8 on UGS. Is that normal?

fra589, 2018-04-28:
Hi @hzxiaoyu 
Alarm 8 : Homing fail. Cycle failed to clear limit switch when pulling off. Try increasing pull-off setting or check wiring. 

The homing cycle on axis first do a rapid travel to the axis home switch, then stop when the switch change his status, then go back a little to change the switch status again, then go slowly to have a precise position of the switch.
The alarm 8 in triggered when the switch does'nt change his state during the back move.
@++;
Gauthier

jflapao, 2018-04-29:
Hi @hzxiaoyu 

What is you 5axis g-code sender, for the GRBL firmware, I use the bCNC for the 3 axis, and work very well but the mod for the 4 and 5 axis don't work so good, stop randomly  with no reason, i Use them because the cam part and the visualizer of the work, the estlcam is good with the cam, but i don't like to have a proprietary firmware, and we can't command a GRBL firmware, only create the gcode.

hzxiaoyu, 2018-04-29:
@fra589 thank you ,I fixed it.
@jflapao I use JAVA UGS,tested XYZ axis,work well.

jflapao, 2018-04-29:
@hzxiaoyu so you only use the 3 axis XYZ even with the 5 axis firmware?
OK, i'm search for the 5 axis g-code sender, i'm gone try the @fra589 application, but now i have problems with the connection, the bCNC moded for 4 axis some time work, some time don't. I need at this time 4 axis, 3 for standard XYZ and another for my lathe, because i moded my mpCNC to engrave on a round material.

fra589, 2018-04-29:
Hi @jflapao , 
I haven't tested it on windows for a long time... Since this last Win test, I rewrite lot off serial protocol to optimize comunications. So, I am not surprised :-S
I will test it on Windows 10 tomorrow for needed debug

jflapao, 2018-04-29:
@fra589 OK, for now I can't use linux because the only machine with linux that i have now is the one with the python 2.7 for the bCNC and my work machine. But I gone try make dual boot for windows and linux. But if you want that your application work in both OS, is good debug in both OS

fra589, 2018-04-30:
@jflapao Hi Jose,
Unfortunately, It work well on my Windows 10 box. So, it will be more dificult to debug... But interesting !-)
On the picture you post, it seem that the serial port is normaly open, but the init string from Grbl is not received or understand. So, I add some debug in the opening process (I putted the new code few minutes ago) can you test it with debug mode on (menu : Machine + Debug mode) and post the result of the debug tab ?

jflapao, 2018-04-30:
@fra589 Hi,
I do not know what happened, but even after doing the new download, cn5X ++ did not recognize the arduino, to try to misdirect if the firmware was really flashed, I used the GRBL Controller 3.6.1 and it was really well flashed after I returned the application and voila already works. Very strange, but it looks like it already works.

Now I would like to give you some ideas,

I think there should be a viewer of the work in drawing mode and not only in code G, let's imagine using the app in some machines for newbies does not matter to see or know the code G, but to see the evolution of the drawing.

Have a CAM mode such as bCNC that uses plugins to do simple things like flatten, pocketing, drill, etc, or even make a mix between that CAM and EstlCam that is very good.

I do not know if it is complicated because I do not understand programming, but I'm thinking of automating a real machine that cuts in stone with a disk in XYZ, but it's very old and move by oil pressure, the idea is to put an automaton, an arduino, drivers an inverts and a app as it is, the question is need of encoders to report the position of the machine, so that even if it loses its way it retrieves them, to be as precise as possible, I have not yet chosen the hardware at the level of drivers, encoders, inverts, because I do not know how to solve the question of the encoders and also of jogging because it will be necessary to move the machine through joysticks

luisluna2, 2018-05-01:
@fra589 you have a tutorial for Use your software 5 axis in Windows??? is amazing!!!!

fra589, 2018-05-01:
Hi @luisluna2 ,
I not yet have a tutorial... The software is still in active development and not already finished. Tutorials will be come soon after the first stable version.
@++;
Gauthier.

luisluna2, 2018-05-01:
Wow great!, friend have a diagram of PINOUT Firmaware for start with you amazing project?

fra589, 2018-05-02:
Hi @luisluna2 
I began to do it, but there is only 24h a day :) 
It will be available soon in wiki section... Stay connected!
@++;
Gauthier

luisluna2, 2018-05-02:
@fra589 of course I will follow your work!!!!, and check this, maybe can put and use with out PC:

TFTScreen:
https://www.marlinkimbra.it/?p=741

LCD Screen Arduino UNO Controller:
http://www.bartvenneker.nl/Arduino/index.php?art=0025

Have a great day, I hope help you :+1:

fra589, 2018-05-02:
since I don't found a 5 axis grbl controler / sender, I create cn5X++ (https://github.com/fra589/cn5X)  
Witch is intended to have capacity to work on Raspberry PI or equivalent (Raspbian) with 1024x600 touch screen (https://fr.aliexpress.com/item/7-inch-Raspberry-pi-touch-screen-1024-600-7-inch-Capacitive-Touch-Screen-LCD-HDMI-interface/32729473252.html?spm=a2g0w.10010108.1000016.1.15977444ihwK2r&isOrigTitle=true)
This will be a cnc hardware console equivalent to industrials one...
[image]

fra589, 2018-05-02:
@luisluna2 
Pinout diagram is here :  
https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout
@++;
Gauthier.

jflapao, 2018-05-02:
@luisluna2 Nop, connect you arnuino mega 2560 by USB cable to the Raspbery Pi, the Pi here is the like a ordinary computer, only small, more e efficient energetically, and more portable. My first DIY CNC was made with a raspberry pi B one alamode (a GPIO raspberry 2 arduino uno converter) and a CNC shield 3.0 and work very well, the the raspberry pi 1st generation is quite slow and if the machine is work i can't nothing on the sender. Now i have a netbook is more rapid but not enough for me, I gone use my old laptop i5 8gb ram to run my CNC but for now without touch screen.
@fra589 Hi, I thing the 7" is quite small for the interface, can you make a connection to the arduino by Bluetooth or even by Wi-Fi, I can run your app on a android tablet, but because i can't install the arduino driver (i can't find nothing about) but if you  connection tipe by Bluetooth or Wi-Fi maybe we can use the tablet like a screen and a mobile joystick.

luisluna2, 2018-05-02:
I'm working with Ramps 1.4 and 128x64 LCD now, I have 4Axis but I have problems with make 5 axis :( when I move E0 (4th axis) +10mm when go to see E1 (5th Axis) this axis have the 10mm move in the E0 (4th axis).

But now you finish 5 axis GRBL and your controller (in Linux Really?, I use Widnows haha), now only Need make program for a Big LCD or run with Rabsberry PI like you comment and the work is finish, Really is a amazing work Friend, Congratilations!

Only for see:

1.- Add more extruders to RAMPS 1.4(TEXT for Ramps 1.4, check with CNC SHIELD use AUX-1 but need 1 pin of AUX-2)
http://reprap.org/wiki/Adding_more_extruders

2.- Maybe add some Relay 5V output in SERVOS 
http://3deeprint.blogspot.mx/2016/01/first-article.html

3.- Maybe modify for "Coolant Mist" for use a PWM Laser (TEXT for RAMPS 1.4)
http://dreammakercnc.blogspot.mx/2015/06/marlin-firmware-modification-for-laser.html

Sorry for late reply, I'm from Mexico, I hope that all works fine :')

luisluna2, 2018-05-02:
Thank you for reply @jflapao
If you wanna more big screen: https://youtu.be/VqqcMB_zGYA

I think that 7In is very good and more if is touch :)

liushuya7, 2018-05-02:
I'm having trouble on limit switches for A & B Axis.
I installed a limit switch on each axis, and connected it to D42 and D44 respectively. When the A axis limit switch triggered, on cn5x it shows both A and B have been triggered. When the B axis limit switch triggered, there is nothing happening i.e no message.. 
[image]
In [cpu_map.h](https://github.com/fra589/grbl-Mega-5X/blob/edge/grbl/cpu_map.h):
`#if N_AXIS > 3
   #define MIN_LIMIT_BIT_3 7 // Axis number 4 : RAMPS AUX2 pin D42
 #endif
 #if N_AXIS > 4
   #define MIN_LIMIT_BIT_4 5 // Axis number 5 : RAMPS AUX2 pin D44
 #endif

 #if N_AXIS > 3
   #define MAX_LIMIT_BIT_3 1 // Axis number 4 : RAMPS AUX2 pin D40
 #endif
 #if N_AXIS > 4
   #define MAX_LIMIT_BIT_4 5 // Axis number 5 : RAMPS AUX2 pin D59`
Where could be the problem?

luisluna2, 2018-05-02:
@liushuya7 maybe is a problem in the Linux Controller or mayb is in the Code. Only need wait for our friend @fra589 make reply :+1:

jflapao, 2018-05-02:
@luisluna2 
Hahahahah, connect a pi to a TV ou a big monitor it's easy with the hdmi connection , but you lose the touch on this type of screen, at this time i have my pi connected to 26"TFT screen. If you think the 7" screen is ok take a look to this photos, on my 10,1" netbook
[image]
[image]

if you see i lose some bottoms because the small screen

The next photos is my Pi 1st generation, the alamode (pi 2 arduino uno GPIO converter) and the CNC Shield, is the hardware of my 1st generation DIY CNC, now I use the mega to use 4 axis.
[image]
[image]

I whant use the mega with 5 axis to use my mpCNC with a lathe, but the 4th axis (A axis) and the 5th the (B axis) are normal axis. If you use a standard CNC the A axis is a rotation over the X axis and is from 0 to 90 degrees, the B axis is the rotation over the Y axis and normaly we don't you this, because we use the C axis that is a rotation over the Z axis and can rotate 360 degrees, but i need a linear rotation axis for the lathe, that can be interpolated with the XYZA and C and if we need a continus rotation to round peces, but for now I can use like a normal axis moded the g-code or create my g-code to round the pece and the aplay the warp g-code to create the pece.

luisluna2, 2018-05-02:
Hello friends, I have problems to compile GRBL, with what arduino version can upload???

luisluna2, 2018-05-03:
Hello friends, one question: How can SWAP the ZAXIS motor outputs to YAXIS motor?

I try change pins in cpu_map.h but when I change the pins it does not work any changed that in this case is Y and Z, what I want to do is get 2 Y and 1 Z motors.

Thank you for your coments!

luisluna2, 2018-05-03:
I do not see A9 in the PinOut :(

https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout

fra589, 2018-05-03:
Reply in reverse order :smile:
Hi @luisluna2 :  
A9 is on the Aux-2 :  
[image]
@++;
Gauthier.

fra589, 2018-05-03:
hi @luisluna2
Reply is coming on your [issue #2](https://github.com/fra589/grbl-Mega-5X/issues/2)
@++;
Gauthier.

fra589, 2018-05-03:
Hi @luisluna2,
I just checked compiling with Arduino IDE v1.8.2 and it work fine !
The method used is those describe here : https://github.com/gnea/grbl/wiki/Compiling-Grbl  
Perhaps you forgot to supress old grbl library before starting compiling ?
@++;
Gauthier.

fra589, 2018-05-03:
Hi @liushuya7,

This is a strange behavior ! In my home, end stops for A & B axis works well...
The detection by cn5X is made by an analisys of the grbl status message. Inside this message, ther is the "Pn:" sentence wich list the switchs thas are activated...
In the last version (yesterday) of cn5X, the grbl's status message is writed in the status area on the bottom of the window.
If "...Pn:AB..." in writed in the status, then, the both A and B switch heve been seen triggered by grbl.

Then, you have to look for your hardware.

Warning, all the switch for end stops limits need to have the same normaly open or normaly closed for the whole homing system to work.

@++;
Gauthier.

closed 2018-05-24 by fra589