fra589/grbl-Mega-5X #3: End Stop's

jflapao, 2018-05-03:
@fra589 Hi,

Today I have some time and I'm testing your app, now I realize that the end stop's don't stop the machine, does not matter if you put the end stop on the + or the -, I tested the X Y and Z axis, the light goes on when you press the end stop, but the motor don't stop.
I have done this g-code

G1 X400 F1500
Then I press the end stop and the motor don't stop until reach the 400mm in X then I do
G0 X0
And press the end stop and nothing, i even put to end stop's, one in X- and other in X+, press one press tow press booth noting appends.

luisluna2, 2018-05-03:
@jflapao  Check $21 - Hard limits, boolean:

 https://github.com/fra589/grbl-Mega-5X/blob/edge/doc/markdown/settings.md#2--step-port-invert-mask

fra589, 2018-05-03:
Hi @jflapao 
You need to enable "Hard limits" in grbl config use $21=1  or do it with cn5X++
use Machine + Configuration de Grbl, Tab Limites :  
[image]
@++;
Gauthier.

luisluna2, 2018-05-03:
@fra589 You make some software for GRBL 6 Axis Controller for Windows?

fra589, 2018-05-04:
Hi @luisluna2,
[cn5X++](https://github.com/fra589/cn5X) is able to work on Windows... 
I will post a Windows install release today or tomorow...
@++;
Gauthier.

luisluna2, 2018-05-04:
@fra589 Great!!!, thank you very much dear friend! :1st_place_medal: 

I hope you can help me to program a 128x64 LCD screen with SD (the one used by 3D printers) to move the equipment in a simple way. If can't, thank you very much for your attention and time.

Fantastic project!

fra589, 2018-05-04:
Hi @luisluna2,

For you, one good news ans one bad one 😄😈 

The bad one is grbl doesn't support LCD screen neither SD card reader like some 3D printers firmwares do. The goals of grbl and grbl-Mega-5X are not to implement thoses items wich are not in use on standards milling machines or other machining machines,

The good news is : I released a Windows (W64) install in https://github.com/fra589/cn5X/releases.

@++;
Gauthier.

luisluna2, 2018-05-04:
But can go to LCD friend :)

Check 

LCD Screen Arduino UNO Controller:
http://www.bartvenneker.nl/Arduino/index.php?art=0025

luisluna2, 2018-05-05:
@fra589 I have this problem in 2 PC one with Windows 7 and other with Windows 8.1 :C 

[image]

fra589, 2018-05-05:
HI @luisluna2,
This is not à cn5X++ problem... It seems that is à classic problem you have on your computers.  
See it on [Google](https://www.google.com/search?q=The+entry+point+of+the+ucrtbase+procedure+can+not+be+found.&ie=utf-8&oe=utf-8&client=firefox-b-ab)  
And one fix procedure on [Youtube](https://www.youtube.com/watch?v=5554vP3IZxo)  

Download Visual C++ Redistributable for Visual Studio 2017 for your PC with the right architecture.  
[Visual C++ Redistributable for Visual Studio 2017 - 32 bit](http://download.microsoft.com/download/9/3/F/93FCF1E7-E6A4-478B-96E7-D4B285925B00/vc_redist.x86.exe)  
[Visual C++ Redistributable for Visual Studio 2017 - 64 bit](http://download.microsoft.com/download/9/3/F/93FCF1E7-E6A4-478B-96E7-D4B285925B00/vc_redist.x64.exe)  
Run the vcredist_x32.exe (32 bit) or vcredist_x64.exe (64 bit) and select Uninstall  
Run the Visual C++ Redistributable setup again and select Install  

@++;  
Gauthier.

luisluna2, 2018-05-05:
@fra589  I try this some hours, but Check this XC

[image]

fra589, 2018-05-05:
@luisluna2 You realy have a problem on your computer... [Google is to be your friend for that](https://www.google.com/search?q=The+entry+point+of+the+ucrtbase+procedure+can+not+be+found.&ie=utf-8&oe=utf-8&client=firefox-b-ab)
It seems thats it be possibility of multiples causes...
Another items on the Microsoft answer site : https://answers.microsoft.com/en-us/windows/forum/windows_7-performance/the-procedure-entry-point-ucrtbaseterminate-could/e45c2162-bb43-4082-a3dd-d715c7ea1f46
I can't do more help to you for this... Sorry.
@++;
Gauthier.

jflapao, 2018-05-05:
Hi @fra589,
I have download your pre-release and instalit on my windows 10 pro 64bit PC, but something is not ok with the layout, maybe is my screen resolution (3840x2160 is 4K native) but the layout appear like this

[image]

and I can't resize the lateral panel.

jflapao, 2018-05-05:
@luisluna2 Hello, Holla,
I have see the link about the 20x4 screen you ont put in or GRBL machine:
1st I don't understand why you want a small screen, for me is better a big screen, then a very small screen, and about a SD card reader is better have a raspberry pi doing the jog of a PC sending the lines to the arduino, and give the arduino only the task of execute the g-code, and have a big screen connected to the Pi where you see all the information you need.
2nd in the link i see a UNO that have the GRBL and is the heart of the X-CRAVE and then is conected to the mega by the TX, RX, and is the mega with the code that is showing that do the job of shown the information on the screen. I think if you want is possible have the GRBL 5axis firmware in one mega, and the connect this mega to another mega with the code and you have the 20x4 screen like show in your like, and if you look on the web search for 128x64 screen codes you can put this kind of screen showing the information, but i steel think is better for me have a 13", 14", or even a 15" touch screen with all the information and big bottoms and numbers that have a very small screen with a stupid rotatory encoder, and all no information.

jflapao, 2018-05-05:
Hi @luisluna2 
I have try do the think i see in LCD Link, but I don't have all the parts, I only have a extra mega and a 20x2 SPI LCD, but I write the code uploaded  to the mega connect the LCD, but the only thing the LCD show is "Connect?" the I analyse the code, and I that we wait for you press a bottom, but because I do't have any, I can't go further, but if you analyse the code, I think that is for replicate the 128x64 LCD screen  replacing the rotatory encoder by the joystick, but I don't know for what is the four bottoms, the card reader is like the card reader of the 128x64 LCD. With a little of programming and time we can create a code similar to use a 128x24 LCD and all the components, the rotatory encoder and the card reader
[image]

luisluna2, 2018-05-06:
@jflapao Yes, is best Rabs and Touch Screen, I wanna use LCD 128x64 because is more cheap and have SD reader. 

Check this GRBL controller, I think is best because have SD and Remote Contrlol: https://proyectosarduinoivan.blogspot.mx/2018/01/control-de-cnc-mediante-tarjeta-sd-y.html

luisluna2, 2018-05-06:
@jflapao Yes, is best Rabs and Touch Screen, I wanna use LCD 128x64 because is more cheap and have SD reader.
You need add butons for go to next screen. in the link exist videos (I think is in Germany)

fra589, 2018-05-06:
@jflapao Hi,

Yes, I think your layout problem come from your screen configuration...
I think to have reasonable font size with your very high screen resolution, you have to configure the text size at more than 100% int the Windows display parameters panel. If so, try to test and tell me if the layout is correct with the text size parameter at 100%. If yes, I will try to correct this behavior as soon as possible, but I am not shure that it will be easy and can give a perfect result...

@++;
Gauthier.

jflapao, 2018-05-06:
@fra589 Hi,

My screen resolution is 250% the one that is recommended by windows, i know that is no easy, because i have the same problems in autocad, the welcome screen of autocad the text is so small that i can't read, but all the bottoms are ok,them the cota and all the text is small to I have to put 250% more to see, the problem is when i send a file to another co-worker the text is huge. I have't test the python version on this computer because have the python 2.7 to see if we have the same problem. I think if we can adjust the panel manually we can resolve the problem, or wen you resize de windows all the panels and the text resize with them proportional .

jflapao, 2018-05-06:
@luisluna2 
Hi, if you only want work with a DIY CNC with 3 axis you have all ready a firmware working with the 128x64 LCD screen, the mpcnc by vicius https://www.v1engineering.com/marlin-firmware/ and like that you only use one arduino mega with a ramps 1.4, he use the marlin firmware, the repetier host to send the g-code, and the estcam to create the code, is not pure GRBL, but work, if you want you can go the the repetier website and personalize a firmware for your DIY CNC with welcome screen, brand, etc, etc, but all way you have to use a sender that recognize the repetier, or the marlin firmware. if you want to use a normal GRBL and the UGS or other like cn5X or even the bCNC or estlcam you need to do the thing you see in your like use 2 arduinos, uno for the GRBL and other for the LDC and the card reader, because like @fra589 say the GRBL firmware is not compatible with LCD screen is more a interpreter of g-code.

fra589, 2018-05-07:
Hi @jflapao,

Ok, the origin of the problem is found. It's a good start 😃 
In fact, it's not a Python problem, but rather a Qt5 one... For the layout be like I want with the differents size of widgets, I need to place some resizing limits wich are actually incompatible with your huge out of scale font... 
I don't want all the pannels be proportionals, because I prefer that the output zone  and the text zones have priority.
I began to update drawing of the left panel (adding an 6th axis button) and I take this opportunity to try to correct this wrong behavior.  
@++;
Gauthier.

fra589, 2018-05-26:
Hi @jflapao,
Latest release 0.3 (https://github.com/fra589/cn5X/releases/tag/v0.3-alpha) should correct your font problem.
@++;
Gauthier.

closed 2018-06-04 by fra589