fra589/grbl-Mega-5X #4: Clone Axis?

LukeBonathan, 2018-05-08:
Is there a way to clone the Y axis when using a ramps 1.4 board? 

I found the incredibly handy pinout on the wiki.

https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout

However my cnc is currently 4 axis  - a single X/Z and dual Y. 

The only way I can think to do it would be by taking 2 sets of pins from the Y and feed them both into individual drivers.

Thanks

fra589, 2018-05-09:
Hi @LukeBonathan,
Standard RAMPS Z axis have 2 motors output for cloning, so, you can swap Z motor to another in cpu_map.h to do what you want.
See [grbl-Mega-5X issue #2](https://github.com/fra589/grbl-Mega-5X/issues/2) for doing it.
@++;
Gauthier.

LukeBonathan, 2018-05-10:
Thank you, I have a bit of work to understand how to edit a hex file. 

Is there a recommended site I can read up on this?

rasciodc, 2018-05-29:
@fra589 Congrats for your effort! This is a great fork for GRBL. About your solution to clone Y axis:
> Standard RAMPS Z axis have 2 motors output for cloning, so, you can swap Z motor to another in cpu_map.h to do what you want.

Your solution is very clever and should work great, but I have only one question: How should I regulate the A4988 limiting current?

If I understand correctly, there will be 2 motors being driven by only 1 A4988 driver....should I double the current? What are other impacts of having 2 steppers in 1 drive in terms of speed, torque etc?

Tks!

fra589, 2018-05-30:
Hi @rasciodc 

The A4988 (or DRV8825) are "current regulators", so you don't need to change the output current value, each motor receive the current he need... Only tension will not be the same, due to the global resistor result change.
Perhaps the driver will be more hot, so you can use a better driver like the DRV8825 or use a fan.

@++;
Gauthier.

rasciodc, 2018-06-04:
I might be wrong but, if the steppers are connected in parallel and each
one is rated for 1.5A, the driver needs to be able to output 3A.
Given that DRV8825 is only capable of outputing 2.1A, I would be somewhat
limited...
Would there be an easy way to implement the axis cloning via code? If you
point me on the right direction I believe I could do it...and share it if
it works.

fra589, 2018-06-07:
Effectively, you're probably right...
It seems to me rather complicated to clone by code.  
Another solution would be to connect 2 drivers on the same output of the Arduino board.  
In Grbl-Mega-5X, I plan to add a 6th output axis on the Aux-3 port of the RAMPS. it will work by connecting a driver (step, pulse & enable pins) on Aux-3. You can then connect 2 drivers in parallel.  
@++;  
Gauthier.

fra589, 2018-07-03:
hi @rasciodc,  
You're right to insist, I think I have a good idea how to do it ...  
But unfortunately, I'm on a business trip on the other side of the world, and I do not have access to my development system...  
I will be able to do it after August.  
Be patient :-)

rlwoodjr, 2018-07-04:
Hello, I am considering a 4th axis for the E3 CNC router that I have designed. (see BobsCNC.com) Currently, the E3 CNC router has a clone on the X-axis. I would also like to see a configurable clone for each axis.  This would make it possible to develop a 4th axis.

rasciodc, 2018-07-04:
@fra589 Thanks for replying. Hope you are successful on you business trip! Don't worry, I'm really not waiting for you or somebody to implement a solution and hand it over ready to use... Actually I really hope I can come up with a good solution and contribute with the project!

I really believe it's possible and I have two different approaches to it. Will check which one works or is easier to implement.

The first approach is to modify the Stepper.h file: So far, I was able to clone the stepping command but not the direction command (i.e. the cloned axis moves as expected but does not change direction - it only moves in one direction). Apperently I'm having trouble on setting the direction port invert bits correctly.

If that fails, the second approach would be to modify the Planner.h file by calculating the steps for the clone axis using the X,Y or Z target position (this should be easy to implement but will have the arduino repeat calculations that were already made for the original axis... not very efficient).

I'll keep you guys updated! =]

fra589, 2018-07-06:
Hi @rasciodc,  
Your solution should work (I don't study it completely), but it will not be easily portable and configurable...  
It's sure, my solution need to change about 15 files... But it's not to hard to do...  
Next week, I will come back home... I will try to make the stuff and publish the solution on special Github branch.  
@++;  
Gauthier.

rlwoodjr, 2018-07-08:
I know a little about programming, but this is above my skill level. If the cloning feature is added will I be able to change a header file to make it work for a specific axis?

rasciodc, 2018-07-11:
@fra589 it's my pleasure! Like I said, just want to contibute!

Regarding cloning the axis, I think it is safe to have GRBL clone the Limit Pins as well... So in the cpu_map.h file, (let's say we will clone AXIS_4 to X for example) it is necessary to make MIN_LIMIT_PORT_3= MIN_LIMIT_PORT_1 and MIN_LIMIT_BIT_3=MIN_LIMIT_BIT_1.

The code would look like

```
#define MIN_LIMIT_PORT_3 MIN_LIMIT_PORT_1
#define MIN_LIMIT_BIT_3 MIN_LIMIT_BIT_1
```
My question is, how can I do that if I was to set a variable (VAR=1) and use it to clone the limit pins?Something like this: (which does not work)
```
#define VAR 1
#define MIN_LIMIT_PORT_3 MIN_LIMIT_PORT(VAR)
#define MIN_LIMIT_BIT_3 MIN_LIMIT_BIT(VAR)
```
It's more a programming question/issue but I can't find a solution for it! I was hoping you could give me some advice!

Tks!!!

fra589, 2018-07-11:
Hi @rasciodc,  
I am not fluent with this, but search for "#define macro string concatenation" in Google perhaps will help you.  
About cloning Limit pins, I think it's not necessary with my cloning method because I made a bijection between AXIS_Number and associated letter (AXIS_N_NAME) so, if the axis number 3 is a clone of Y axis, the limit pin activation of the axis number 3 will be interpreted as the Y limit.  
@++;  
Gauthier.

rasciodc, 2018-07-11:
Thanks @fra589, you really pointed me in the right direction!

Even though it is 100% sure I'll use your final version (which I'm very eager to check out btw) it's been a good opportunity for me to practice some coding and get familiar with GRBL code... I was able to get it working but your solution is way more efficient and elegant than mine!

One thing I noticed when debugging the limit switches: the '?' command (to report position, pin states etc) would not show the AXIS_4 and AXIS_5 limit pin status... Only XYZ were displayed.

It really is a minor thing, but, that report proved to be very useful when things don't seem to be working fine... not a Must but Nice to Have....

Tks again! =]

fra589, 2018-07-12:
Hi @rasciodc,
I just pushed the last v1.1h.20180712 version wich having the ability of renaming and cloning axis.  

You can change any axis place without changing the cpu_map.h by simply change the axis names in the config.h file.
You can clone an axis by using the same name for more than one axis in the config.h file. For example, if you use : 
```
#define AXIS_1_NAME 'X'
#define AXIS_2_NAME 'Y'
#define AXIS_3_NAME 'Z'
#define AXIS_4_NAME 'Y'
#define AXIS_5_NAME 'A'
```
You will obtain a 4 axis system "XYZA" with the Y output cloned on axis number 2 and 4.  
  
For the moment, I have only tested the code on a simulator, not on a mecanical machine... I haven't any time for now to make full tests, so be carrefull ! And tell me if it work as you want.  
  
About the end-stop status, it work for all axis for me.

@++;  
Gauthier.

rasciodc, 2018-07-13:
@fra589 You do work fast! Your code looks and works really fine! I saw you implemented a 6th axis, that's nice!

I've made some tests on real stepper motors and the cloning feature does work and is super easy to change, Yay!

I would reinforce to new users the absolut necessity of homing the cloned axes at the same cycle, as this is not done automagically by the code and could cause damage in some machines (like a gantry machine with motors on each side.....one would move but the other wouldn't.... no fun!)

I also would like to know your opinion on something: What would be your approach to limit switches? 1) Would you install one on each cloned axis and during homing each axis would trigger its own switch (most likely and ideally at the same time if machine is true, symmetrical and parallel) or 2) Would you have the 2 axes share just one limit switch so that during homing both axes obey to the same switch (stopping exactly at the same time)?

My question comes from a practical point of view, where, in a real world and real machine you can't really install 2 different limit switches on each cloned axis in the exactly same way and exactly same position, thus, giving different homing position to axes that are meant to clone each other. (and from this thinking came my idea of cloning the limit switches....and now I'm thinking if that's really necessary).

Again, you've made a relly nice work!

fra589, 2018-07-13:
Hi @rasciodc,  
  
Your remark on homing both cloned axis in the same time is very relevant. I will see the homing code to protect it from this problems when axis are cloned.  
  
My opinion about limit switch is to use separate switch on both axis. It is sure than they don't work exactly at the same time, but the first switch who will detect the limit will stop the both axes. The second switch will act as redundancy if the first fail. this will increase the security.

@++;
Gauthier.

rasciodc, 2018-07-16:
Hi @fra589!

I don't know if it is extremely important to implement code to do the homing automatically for cloned axes. I believe a clear warning and example instruction should do the job.

Additionally, I'm trying to make my machine as "user friendly" as possible, in a way that any one can use it without being affraid of messing something up (like my wife and some friends...)

To accomplish that the hard limits are very useful and somewhat important. Would love some directions from you to implement them to the code.

I believe it would be necessary to "adapt" lines 115 to 117 on Limitc.c file. Is that right or am I missing some other part that need to be changed also?

closed 2018-08-27 by fra589

tshilling, 2018-12-29:
I'm sorry if this is a bit delayed, but I am moving over from Mach 3 to GRBL.  My machine is a standard 3 axis gantry with two motors for the Y-Axis (Y and A).  In Mach 3, I home my Y axis, which drives Y and A axis toward two independent home switches (Both located in the Y+ direction).  Both drive independently (At the same rate and direction toward Y+) until their corresponding Home switch is triggered.  Meaning, the Y motor moves until the Y limit switch is triggered and the A motor moves until the A limit switch is triggered.

This is to achieve auto squaring of the axis and is fully supported by Mach 3.  When the motors are not powered there is no holding torque and it is easy to knock the gantry out of square.  When I power everything up and run my home cycle, it ensures everything is nice a square.

Before I go digging into the code to try and implement this behavior I wanted to make sure I am not misunderstanding the current implementation.  As current of the time of this posting, it appears that this is not how slave/cloned axis are implemented.  

As I imagine it, you would configure A as a slave of Y.  Then when homing, Y and A homing would initiate at the same time.  Each would drive until its corresponding limit switch was triggered.  They should trigger very close together in time as long as the axis is not totally out of square.  Then every future command to move Y would auto move A in the corresponding way.  This behavior has worked for me for 10+ years with Mach and is the only thing currently keeping me from switching over to the more open source GRBL.

Thanks for the time and looking,
V/r,
Tim

RaphaelDives, 2018-12-29:
Hey @tshilling 
this should work like you wrote above!

On my MPCNC I have two steppers at each axis, therefore I cloned X to A and Y to B,
simply by changing the AXIS_4_NAME to 'X' and AXIS_5_NAME to 'Y'. (config.h:60ff)
Then I defined my homing cycle as (config.h:152ff):
`	#define HOMING_CYCLE_0 ((1<<AXIS_2)|(1<<AXIS_5))
	#define HOMING_CYCLE_1 ((1<<AXIS_1)|(1<<AXIS_4))`
(I have my Y Axis first because it's the longer axis and I couldn't home the X axis when Y is not parallel)

By the way - since I have only limit switches, I changed the MIN_LIMIT_PORT and _BIT variables to use the max limit of my RAMPS board for the 'A' and 'B' limit switches. If you want to do so, take care to change the MAX_LIMIT variables to other values… :D (cpu_map.h:247ff)

And - another pitfall: Every parameter for the Y Axis has to be the same for the cloned axis!!
(like $101 = $104, $111 = $114 and so on...)

For me this works very well! Just give it a try! :)

Cheers
Raphael

fra589, 2019-01-01:
Hi @RaphaelDives, your reply is perfect, thanks ! I don't have to do it myself 😄 .
@tshilling, You have understood the principle for the homing of cloned axes.

Happy new year to ALL !!!

@++;
Gauthier.

rasciodc, 2019-01-01:
Happy new year guys! 

I have been using this branch of grbl, a ramps board and cloned Y axis machine.

The homing cycle works like a charm! Go for it, you will not regret it. 

I have a friend running his machine on a semi professional shop. He used to run grbl, switched to mach3 but today he is back with grbl again and using this branch specifically.

lesaux, 2019-02-23:
Hi Guys,
I've setup a dual Y axis cnc, with this firmware, by just changing the `axis_4_name` to `Y`.
Now I am trying to setup Homing, with a single switch on Y1.

I've setup homing like
```
#define HOMING_CYCLE_0 ((1<<AXIS_4)|(1<<AXIS_2))
```
But when Y1 hits the switch, Y2 keeps on going. I'm not too sure how to figure this out, so I'm looking for pointers.

RaphaelDives, 2019-02-24:
Hi @lesaux, 
that's not a problem at all. Just set the "missing" limit switch to the same port and pin as the y-axis.
Take a look at cpu_map.h:248+260. These should be the parameters for your y-axis. 
For the cloned (4th) axis in 251+263.

So I would change:
```
251: #define MIN_LIMIT_PORT_3 J
263: #define MIN_LIMIT_BIT_3 1
```

Cheers,
Raphael

lesaux, 2019-02-24:
Thank you so much @RaphaelDives  this worked perfectly! I actually missed that this was mentioned earlier in this thread.

yauravy, 2022-06-05:
Hi Guys

i have also a cnc with 5 axis with  2x and 2y, i'm planning to use grbl 5x but i don’t know how to clone axis, can’t  find the number of axis on config file maybe they updated a lot since this thread. Thanks

dmikor, 2025-01-04:
all i read - not work.. I can't understand how to clone Y axis. Help please
i have only XY axis for my cnc based on MKS Gen_L v2.1 
for my Y i use 2 motors and 2 drivers (Y and E0 port)
HOW CLONE Y to E0 ?????????

fra589, 2025-01-05:
Hello @dmikor,

If you have only X and Y axis, why not to clone your second axis to the Z port?
In this case, write in config.h (line 55 to 64):
```
#define N_AXIS 3        // Number of axes (3 to 6)
#define N_AXIS_LINEAR 3 // Number of linears axis, must be <= N_AXIS

// Axis indexing and names
#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V, W, D, E & H.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Y'
```

Output of $I with this config:
```
[VER:1.2h.20220109:]
[AXS:3:XYY]
[OPT:VNMGH,35,255,48]
ok
```
Which mean 3 axis named X, Y and Y => Y is cloned on the Z port.
  
  
  
If you absolutely want to use E0, then you will have unused, but present Z axis.
To do this, write in config.h  (line 55 to 71):
```
#define N_AXIS 4        // Number of axes (3 to 6)
#define N_AXIS_LINEAR 4 // Number of linears axis, must be <= N_AXIS

// Axis indexing and names
#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V, W, D, E & H.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'
#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'Y' // Letter of axis number 4
#endif
```

Output of $I with this config:
```
[VER:1.2h.20220109:]
[AXS:4:XYZY]
[OPT:VNMGH,35,255,65]
ok
```

Which mean 4 axis named X, Y, Z and Y => Y is cloned on the H0 port.

@++;
Gauthier.