fra589/grbl-Mega-5X #6: 4 linear axis (Hotwire Foam Cutter)

rasciodc, 2018-05-29:
A great application of the Mega-5X would be on 4 axis hotwire cutter machines, which have 2 "linear towers" (XY axis and AB axis).

Do you think this version of GRBL could handle it?

I know that in terms of number of ports it would be ok but my concern would be regarding the calculation of movements and accelerations, given that one tower (XY) should have its movements completely decoupled (independent) from the second tower (AB)...

fra589, 2018-05-30:
Hi @rasciodc,

From a performance point of view, the number of axis is not a problem.
Movement management is the same regardless of the number of axes (from 3 to 5 (6 in the next prototype version)).
The computation is done on all the axes with each GCode order, and the computation time is very little different according to the number of axes.
The calculation result is then queued in the motion planner, and the movements take the same time from the planner regardless of the number of output axes.
But,
The movements are always synchronized between all the axes in one GCode order.
If you send G1X10Y20A10B30, the resulting computation will made for that all the axes finish their movements in the same time.
If you send G1X10Y20, then G1A10B30, the system will move simultaneously the X and Y axis, it will wait for X and Y move finish, then move simultaneously A and B.
So, if you need completely decoupled (independent) movements between the two towers, I think you need 2 arduino cards with 2 GRBL, then you will need to make the sync between them with the sending computer.

@++;
Gauthier.

closed 2018-06-04 by fra589

johnopsahl, 2019-01-05:
Gauthier,

> So, if you need completely decoupled (independent) movements between the two towers, I think you need 2 arduino cards with 2 GRBL, then you will need to make the sync between them with the sending computer.
> 

Do you know of a gcode sender software that can send gcode to two controllers with grbl? Movements of the two cnc machines need to be sequenced but not simultaneous. The axes of each machine do need to be simultaneous. The reason I need two controllers with grbl is because one cnc machine is a 6 axis and the other is a 5 axis.

fra589, 2019-01-06:
Hi @johnopsahl,  

I don't think that such software exists. 
When I searched a GCode sender with more than 4 axis capability, I don't found it, so I began to write [cn5X++](https://github.com/fra589/cn5X).  
I think you need to write your own sender.  

@++;
Gauthier.