fra589/grbl-Mega-5X #7: Kinematics

liushuya7, 2018-06-05:
Hi Gauthier,

I wonder how to add my customized kinematics in grbl-Mega-5X?
I've successfully implemented it in **Linuxcnc** [5-Axis Kinematics](http://linuxcnc.org/docs/master/html/motion/5-axis-kinematics.html)
However, I am not sure if I could modify to add my own 5-axis kinematics.
In [#791](https://github.com/grbl/grbl/issues/791), it says currently not possible.
Could you give some instructions?

Thanks,
Joshua

fra589, 2018-06-07:
Hi Joshua,

You can get inspired by how COREXY works (search for COREXY in all .c and .h files)
The main motion calculation is made in planner.c :  
`#ifdef COREXY`
`    position_steps[X_AXIS] = system_convert_corexy_to_x_axis_steps(sys_position);`
`    position_steps[Y_AXIS] = system_convert_corexy_to_y_axis_steps(sys_position);`
with the "system_convert_corexy_to_x_axis_steps()" and "system_convert_corexy_to_y_axis_steps()" functions defined in system.c

But in my opinion, these calculations are out of place in Grbl. They must be made at the post-processor level, which will save valuable computing time in favor of speed and smoth of movement.

@++;
Gauthier.

liushuya7, 2018-06-07:
Thanks for the explanation. From my understanding from what you just said, all the axes right now are independent, and you would recommend others to implement any 5-axis kinematics within the post-processor so that the generated gcode (X Y Z A B) would directly command each individual axis with already calculated right amount of movement instead occupying Grbl to do the kinematics calculation because that would slow down the system and not be accurate. Is that correct? If that is so, I guess the challenge would be developing a 5-axis post-processor. I used to use Fusion 360 for tool path generation with default linuxcnc post-processor. Now maybe I should dive in more to write a post-processor of my version. Do you think I am on the right direction?

Thanks,
Joshua

qiwalker, 2019-01-28:
hi @fra589 
can i use grblmega-5x for scara arm ?

thanks
rizki