fra589/grbl-Mega-5X #11: limit switch not recognized

jhkaho, 2018-08-17:
Hi,
Sth very strange happens on my Ramps1.4 + Mega.
Stepper is still running when limit sensor be triggered, and no alarm. But if I send '?' grbl recognizes the triggered limit switches.

<Run|MPos:69.913,0.000,0.000,0.000,0.000|FS:250,0>
ok
<Run|MPos:82.113,0.000,0.000,0.000,0.000|FS:250,0>
ok
<Run|MPos:118.375,0.000,0.000,0.000,0.000|FS:250,0|Pn:X>
ok
<Run|MPos:125.975,0.000,0.000,0.000,0.000|FS:250,0|Pn:X>
ok

Here are part of my setting:
Grbl 1.1h
$5=1
$20=0
$21=1
$22=0

According to the Wiki, a hard limit event is considered to be critical event, where steppers immediately stop.  
Can somebody help me? Thanks!

jhkaho, 2018-08-23:
OK，I found hard limit is disable(or missing, exactly) in the code. Limit switches only work for homing. In the homing process, software method (check limit states every cycle) is used instead of hardware interrupts.
So, only soft limt ($20) can be used.
Could you please update Grbl Configuration page?

kahoch, 2018-08-30:
Hi Gauthier,
I use pin 53 52 51 50 12 as the XYZAB limit(+&- combined), it works so far.

jwags123, 2018-09-24:
Hi Kahoch, how did you get this fixed? I'm having the same issue of limit switches only working during homing.

fra589, 2018-09-25:
See [Issue #15](https://github.com/fra589/grbl-Mega-5X/issues/15) wich stay in the "TODO" state while the solution is not implemented...

closed 2018-09-25 by fra589

kahoch, 2018-09-26:
> 
> 
> Hi Kahoch, how did you get this fixed? I'm having the same issue of limit switches only working during homing.

I modified the code a little bit. 
Briefly speaking:
1. Change the limit pins to those which support PCINT in cpu_map.h
2. Enable them
3. Some changes in limits.c
I can share with you my code if you need it, but it's very bad because I'm really a noob in programing.

jwags123, 2018-09-26:
Please!  I guarantee its better than mine....

tkurbad, 2018-11-09:
@kahoch I'd also be interested in your code! Thank you!

fra589, 2019-12-15:
Hi all,
New version "v1.1m.20191215" with hardware limits available for RAMPS pushed today !
See new cpu_map.h file at line 321 to enable new feature.
@++;
Gauthier.