fra589/grbl-Mega-5X #13: 5th Axis Issue

liushuya7, 2018-09-13:
Hi Gauthier,

Whenever I jog or send Gcode commands to move the 5th axis (only the 5th axis has this issue), the grbl-Mega-5X always froze. I have to reset to make it work again. Have anyone ever encountered this? I checked all my wire connection, there is no problem. I suppose something within the grbl-Mega-5X firmware is not functioning correctly? I need some help. Thank you!

Joshua

liushuya7, 2018-09-14:
I reloaded GRBL into Arduino. Problem solved. Please discard this issue. Thx.

closed 2018-09-14 by liushuya7

fra589, 2018-09-15:
Hi Joshua,

I already have this type of bug... 

With many tests loading and reloading Grbl onto the Arduino, sometime, there is memory corrupt that need another reload for correct it.  
  
Some other times, when you change the number of axis or the axis names or the axis positions, you need to reset the parameters eeprom memory to "factory profile" because the number of parameters is not the same from the anterior version to the new one, so the eeprom mapping is not the same.  
  
However, you had the right reflex to reload Grbl on your Ardiono board 😃   
  
@++;  
Gauthier.