fra589/grbl-Mega-5X #14: Pin Out in the Arduino Mega (Not use ramps 1.4) ???

luisluna2, 2018-09-23:
Hello dear @fra589 

I wanna know if you have the Pinout in image format of the Arduino Mega for you GRBL-Mega-5X with out Ramps.

Tahnk you for your answer friend.

fra589, 2018-09-25:
Hi @luisluna2,

Just ask Google for "arduino mega pinout" picture...  
You will find toons of picts like this :  
https://protosupplies.com/wp-content/uploads/2017/12/Arduino-Mega-2560-Board-Pinout.jpg  
  
@++;  
Gauthier.

closed 2018-09-25 by fra589

fra589, 2018-09-25:
4, 5 & 6 axis work only with the Ramps code.  
Just use the cpu_map.h comments to known the pin number of an item.  
For example :  
`#define X_STEP_BIT 2 // MEGA2560 Digital Pin 24`  
Then, search for "Digital Pin 24" on the [arduino pinout picture](https://protosupplies.com/wp-content/uploads/2017/12/Arduino-Mega-2560-Board-Pinout.jpg) to know where to connect directly on the Arduino board without Ramps.

luisluna2, 2018-09-25:
@fra589 Yes, I see this now, thank you for you time and attention, I ask this because only I can add 2 outputs of relays to the board, If add 3 or 4 only work 2, maybe is a interference with other pin or pin in function, I will check for this.

I'm working on adding a full graphics LCD to GRBL. They have done it before, but first tenog to see yes with its current configuration in arduino this is possible.