fra589/grbl-Mega-5X #15: Hard limits remain disabled

jwags123, 2018-09-24:
Hi!  My hard limits and homing are both enabled however the switches are only recognized while homing.  During normal operation I can see the switches are working appropriately when pressed however an alarm is not thrown and machine continues to move.  Do you have any suggestions?  Thanks.

fra589, 2018-09-25:
Hi @jwags123,  
  
In the original [gnea/grbl-Mega](https://github.com/gnea/grbl-Mega), hardware limits are disabled for the Ramps board in cpu_map.h. Interrupt code is missing for the Ramp board version, so the hardware limits work just for homing. 
  
This issue is added to my todo list, but I can't tell you when it will be implemented.  

@++;  
Gauthier.

jwags123, 2018-09-25:
Thank you for the quick reply @fra589!  No worries, I'll wait it out.  Just wanted to make sure it wasn't something I was doing incorrectly.  I appreciate what you do!

closed 2018-09-25 by jwags123

MichealNicht, 2019-12-10:
@fra589
 Hi, hope it's okay to ask this here.
I've been using the Mega 5X software for a few days, since I needed a 4th axis. It works great!

However, I really want to get the limit switches working, since I'm using big stepper motors, if the axis run into each other they will destroy the machine. 
Is this still on the TODO list, or implemented yet?
In GrblGru, I can see the limit switches getting triggered, but it doesn't give an alarm state. 

Also tried kahoch's (user from issue#11) mod, but that also doesn't seem to work somehow...

fra589, 2019-12-10:
Hi @MichealNicht,  
Hard limit alarm for Mega-5X is finished implemented, but need some tests before publish...
Please, be patient :-$  
@++;
Gauthier.

MichealNicht, 2019-12-10:
@fra589 Thanks! I'll keep an eye out in that case :)
Thanks for all your work!

fra589, 2019-12-15:
Hi all,
New version "v1.1m.20191215" with hardware limits available for RAMPS pushed today !
See new cpu_map.h file at line 321 to enable new feature.
@++;
Gauthier.

H-Palacio, 2021-03-05:
I have this setup in my DIY CNC

Mega2560 and Ramps 1.6 with Nema17 and 5/16" screw. It was working on Marlin 1 with swapped YZ drivers.

OpenBuilds Control don't show A and B configs, so just setup manually. It show undefined instead.
[ $$ ] $100=568.240 ;X-axis steps per millimeter
$101=568.240 ;Y-axis steps per millimeter
$102=568.240 ;Z-axis steps per millimeter
$103=568.240 ;undefined
$104=568.240 ;undefined
$110=600.000 ;X-axis maximum rate, mm/min
$111=600.000 ;Y-axis maximum rate, mm/min
$112=300.000 ;Z-axis maximum rate, mm/min
$113=600.000 ;undefined
$114=600.000 ;undefined
$120=10.000 ;X-axis acceleration, mm/sec^2
$121=10.000 ;Y-axis acceleration, mm/sec^2
$122=5.000 ;Z-axis acceleration, mm/sec^2
$123=10.000 ;undefined
$124=10.000 ;undefined
$130=220.000 ;X-axis maximum travel, millimeters
$131=200.000 ;Y-axis maximum travel, millimeters
$132=130.000 ;Z-axis maximum travel, millimeters
$133=220.000 ;undefined
$134=200.000 ;undefined
[VER:1.1l.20190605:]
[ $I ] [AXS:5:XYZXY]
[ $I ] [OPT:VNM,35,255,248]
[ $I ] ok
[20:31:26] [ connect ] Firmware Detected: grbl version 1.1l on COM9

I'm trying to assign hard limits pins to Ramps Limits.

But I notice this

// Define homing/hard limit switch input pins and limit interrupt vectors.
  // NOTE: All limit bit pins must be on the same port
  #define LIMIT_DDR       DDRB
  #define LIMIT_PORT      PORTB
  #define LIMIT_PIN       PINB
  #define X_LIMIT_BIT     4 // MEGA2560 Digital Pin 10
  #define Y_LIMIT_BIT     5 // MEGA2560 Digital Pin 11
  #define Z_LIMIT_BIT     6 // MEGA2560 Digital Pin 12
  #define LIMIT_INT       PCIE0  // Pin change interrupt enable pin
  #define LIMIT_INT_vect  PCINT0_vect
  #define LIMIT_PCMSK     PCMSK0 // Pin change interrupt register
  #define LIMIT_MASK ((1<<X_LIMIT_BIT)|(1<<Y_LIMIT_BIT)|(1<<Z_LIMIT_BIT)) // All limit bits

D10 is attached to X_Limit and also to Coolant/mist D10

// Define flood and mist coolant enable output pins.
  #define COOLANT_FLOOD_DDR   DDRB
  #define COOLANT_FLOOD_PORT  PORTB
  #define COOLANT_FLOOD_BIT   4 // MEGA2560 Digital Pin 10 - Ramps 1.4 12v output
  #define COOLANT_MIST_DDR    DDRH
  #define COOLANT_MIST_PORT   PORTH
  #define COOLANT_MIST_BIT    6 // MEGA2560 Digital Pin 9 - Ramps 1.4 12v output

Have someone achieved this?
Ramps +X is D3
Mega is PE5
Changing: DDRB to DDRE, PORTB to PORTE, PINB to PINE and the corresponding bits
Should it work??

fra589, 2021-03-05:
Hi @H-Palacio,  
  
The limit code definition you talkabout (you copyed/pasted lines from line 60 to 71 which are in the CPU_MAP_2560_INITIAL section) is not the one used by the RAMPS boards.  
This section is the Grbl's original one which can be used only for 3 axis.  
The cpu_map.h section used for the RAMPS board is the CPU_MAP_2560_RAMPS_BOARD which is from line 136 to 418.  
The choice of the section is made in config.h at line 44:
`#define CPU_MAP_2560_RAMPS_BOARD`  

For standard pinout definition, see the wiki here:  
https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout  
  
And for pinout modification howto:
https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h  
  
@++;  
Gauthier.

H-Palacio, 2021-03-10:
@fra589 thanks for your reply. 
I check your info, i'm sorry i didn't notice that.
The problem was the wiring on the limits, it was C-NC now it's working on/off in the troubleshooting section.
Thanks for your work

H-Palacio, 2021-04-22:
Hi @fra589  it's me again.

My cnc is working ok,  I made some test and parts without problems.
[image]
[image]

I don't know why, maybe I still didn't find the correct info. 
I need to manually zero XYZ, I have already wired correctly my limits, it works on OpenBuilds Controls, in troubleshooting section all of them (XYZ and probe) work on/off.
When trying to home all with the button, Z works up and down, then X moves and hit X limit and just freeze

fra589, 2021-04-24:
Hi @H-Palacio,  
  
>  When trying to home all with the button, Z works up and down, then X moves and hit X limit and just freeze
  
Without the complete axis configuration and homing configuration you use in your config.h file, I will not be capable to help you for patching yopur homing problem...  
  
@++;  
Gauthier.

abdelabdel1, 2021-05-06:
Bonjour Gauthier
Tout d'abord je tiens a te féliciter et te remercier pour ton portage.
J'essaie de me faire ma board mais je suis confronté a un problème.
i have moved all the stepper command like i need and all work fine but i have try to move the hard limit and the user control to the same port ( pcint 2) without succes.all the user control  work but no alarm appear when i change the state if the inputs pins. I have join the piece of the cpumap concerned, perhaps can you say me what's i have make wrong

// Define homing/hard limit switch input pins and limit interrupt vectors.
// NOTE: All limit bit pins must be on the same port
#define LIMIT_DDR       DDRK
#define LIMIT_PORT      PORTK

#define LIMIT_PIN       PINK
#define X_LIMIT_BIT     0 // MEGA2560 Digital Pin 10
#define Y_LIMIT_BIT     1 // MEGA2560 Digital Pin 11
#define Z_LIMIT_BIT     2 // MEGA2560 Digital Pin 12
#define LIMIT_INT       PCIE2  // Pin change interrupt enable pin
#define LIMIT_INT_vect  PCINT2_vect
#define LIMIT_PCMSK     PCMSK2 // Pin change interrupt register
#define LIMIT_MASK ((1<<X_LIMIT_BIT)|(1<<Y_LIMIT_BIT)|(1<<Z_LIMIT_BIT)) // All limit bits

  // Define homing/hard limit switch input pins and limit interrupt vectors.
  #define MIN_LIMIT_PORT_0 E
  #define MIN_LIMIT_PORT_1 E
  #define MIN_LIMIT_PORT_2 E
  #if N_AXIS > 3
    #define MIN_LIMIT_PORT_3 E
  #endif
  #if N_AXIS > 4
    #define MIN_LIMIT_PORT_4 E

  #endif
  #if N_AXIS > 5
    #define MIN_LIMIT_PORT_5 E // (Ramps Aux-1 D57)
  #endif
  #define MIN_LIMIT_BIT_0 3 // X Limit Min - Pin D3
  #define MIN_LIMIT_BIT_1 4 // Y Limit Min - Pin D14
  #define MIN_LIMIT_BIT_2 5 // Z Limit Min - Pin D18
  #if N_AXIS > 3
    #define MIN_LIMIT_BIT_3 3 // Axis number 4 : RAMPS AUX2 pin D42
  #endif
  #if N_AXIS > 4
    #define MIN_LIMIT_BIT_4 4 // Axis number 5 : RAMPS AUX2 pin D44
  #endif
  #if N_AXIS > 5
    #define MIN_LIMIT_BIT_5 5 // Axis number 6 : RAMPS AUX2 pin D57
  #endif
  #define _MIN_LIMIT_BIT(i) MIN_LIMIT_BIT_##i
  #define MIN_LIMIT_BIT(i) _MIN_LIMIT_BIT(i)
  #define MIN_LIMIT_DDR(i) _DDR(MIN_LIMIT_PORT_##i)
  #define MIN_LIMIT_PORT(i) _PORT(MIN_LIMIT_PORT_##i)
  #define MIN_LIMIT_PIN(i) _PIN(MIN_LIMIT_PORT_##i)

  #define MAX_LIMIT_PORT_0 E
  #define MAX_LIMIT_PORT_1 E
  #define MAX_LIMIT_PORT_2 E
  #if N_AXIS > 3
    #define MAX_LIMIT_PORT_3 E
  #endif
  #if N_AXIS > 4
    #define MAX_LIMIT_PORT_4 E
  #endif
  #if N_AXIS > 5
    #define MAX_LIMIT_PORT_5 E // (Ramps Aux-3 D58)
  #endif
  #define MAX_LIMIT_BIT_0 3 // X Limit Max - Pin D2
  #define MAX_LIMIT_BIT_1 4 // Y Limit Max - Pin D15
  #define MAX_LIMIT_BIT_2 5 // Z Limit Max - Pin D19
  #if N_AXIS > 3
    #define MAX_LIMIT_BIT_3 3 // Axis number 4 : RAMPS AUX2 pin D40
  #endif
  #if N_AXIS > 4
    #define MAX_LIMIT_BIT_4 4 // Axis number 5 : RAMPS AUX2 pin D59
  #endif
  #if N_AXIS > 5
    #define MAX_LIMIT_BIT_5 5 // Axis number 6 : RAMPS AUX2 pin D58
  #endif
  #define _MAX_LIMIT_BIT(i) MAX_LIMIT_BIT_##i
  #define MAX_LIMIT_BIT(i) _MAX_LIMIT_BIT(i)
  #define MAX_LIMIT_DDR(i) _DDR(MAX_LIMIT_PORT_##i)
  #define MAX_LIMIT_PORT(i) _PORT(MAX_LIMIT_PORT_##i)
  #define MAX_LIMIT_PIN(i) _PIN(MAX_LIMIT_PORT_##i)

  //  #define LIMIT_INT       PCIE0  // Pin change interrupt enable pin
  //  #define LIMIT_INT_vect  PCINT0_vect
  //  #define LIMIT_PCMSK     PCMSK0 // Pin change interrupt register
  //  #define LIMIT_MASK ((1<<X_LIMIT_BIT)|(1<<Y_LIMIT_BIT)|(1<<Z_LIMIT_BIT)) // All limit bits
  // Hardware limits interrupts are not implemented in the RAMPS version of grbl-Mega
  // since all the limit switch ports are not compatible with interrupts.
  #define DISABLE_HW_LIMITS_INTERRUPT

  // Enable Hardware limit support for RAMPS without using interrupt...
  // Warning! bouncing switches can cause a state check like this to misread the pin.
  // When hard limits are triggered, they should be 100% reliable.
  // The RAMPS_HW_LIMIT is implemented inside the stepper driver interrupt. Depending of your
  // hardware, this can affect the max speed possibility of movments
  // Disabled by default for performance optimization, uncomment to enable.
  //#define ENABLE_RAMPS_HW_LIMITS

  // Define spindle enable and spindle direction output pins.
  #define SPINDLE_ENABLE_DDR      DDRG

  // Define user-control CONTROLs (cycle start, reset, feed hold) input pins.
  // NOTE: All CONTROLs pins must be on the same port and not on a port with other input pins (limits).
  #define CONTROL_DDR       DDRK
  #define CONTROL_PIN       PINK
  #define CONTROL_PORT      PORTK
  #define CONTROL_RESET_BIT         3  // Pin A9 - RAMPS Aux 2 Port
  #define CONTROL_FEED_HOLD_BIT     4  // Pin A10 - RAMPS Aux 2 Port
  #define CONTROL_CYCLE_START_BIT   5  // Pin A11 - RAMPS Aux 2 Port
  #define CONTROL_SAFETY_DOOR_BIT   6  // Pin A12 - RAMPS Aux 2 Port
  #define CONTROL_INT       PCIE2  // Pin change interrupt enable pin
  #define CONTROL_INT_vect  PCINT2_vect
  #define CONTROL_PCMSK     PCMSK2 // Pin change interrupt register
  #define CONTROL_MASK      ((1<<CONTROL_RESET_BIT)|(1<<CONTROL_FEED_HOLD_BIT)|(1<<CONTROL_CYCLE_START_BIT)|(1<<CONTROL_SAFETY_DOOR_BIT))

abdelabdel1, 2021-05-06:
Bonjour Gauthier, tout d'bbord je tiens a te féliciter et te remercier pour ton portage;
I experiment a trouble when i attemp to compil the program, when i have enable the swith limit option.
please see the screnn capture of the problem
[image]

abdelabdel1, 2021-05-06:
Bonjour Gauthier, tout d'abord je tiens a te remercier pour ton portage.
 J'obtiens une erreur de compilation lorsque je valide "#define ENABLE_RAMPS_HW_LIMITS";
l'erreur est la suivante: 
#error "HW limits interrupts are not implemented"

En te remerciant d'avance

fra589, 2021-05-08:
Bonjour @abdelabdel1,  
  
L'erreur que tu obtiens est générée uniquement lorsque tu commente ```#define DISABLE_HW_LIMITS_INTERRUPT```.
Cela correspond à une ancienne portion de code qui ne peux plus être utilisée avec les cartes RAMPS. Je n'ai pas encore nettoyé cela, mais il ne faut pas y toucher car cela ne fonctionne pas.
Pour utiliser les limites matérielles avec grbl-Mega-5X, il faut juste valider ```#define ENABLE_RAMPS_HW_LIMITS``` sans commenter ```#define DISABLE_HW_LIMITS_INTERRUPT```.

@++;
Gauthier.

abdelabdel1, 2021-05-08:
Merci pour ta réponse, mais, je ne peux compiler le code uniquement lorsque #define DISABLE_HW_LIMITS_INTERRUPT est validé;
 Lorsque je valide  #define ENABLE_RAMPS_HW_LIMITS, je ne peux pas le compiler, et j'obtiens l'erreur cité ci dessus.

[image]

fra589, 2021-05-09:
Bonjour,  
  
Quel est le soft que tu utilises pour éditer et compiler? Je ne reconnais pas tes copies d'écran...  
Je pense que ton problème vient de la méthode qui génère peut-être des conflits entre différentes versions présentes dans différents répertoires.  
Le message d'erreur `#error "HW limits interrupts are not implemented"` est généré lors de la compilation uniquement si `#define DISABLE_HW_LIMITS_INTERRUPT` est décommenté.  
Dans ta version programme édité, tu ne dois pas avoir ce message car je vois que c'est bien en commentaire. Donc, ton compilateur utilise probablement un autre fichier que celui visible comme source.  
  
@++;  
Gauthier.

abdelabdel1, 2021-05-09:
Bonjour Gauthier
J'utilise AtmelStudio 7

fra589, 2021-05-09:
Ah :-( c'est pour ça que j'ai pas reconnu...  
  
Je développe sous Linux avec l'éditeur Geany, je compile avec avr-gcc par un simple `make clean && make` en ligne de commande. Quand c'est compilé, je pousse l'exe sur la carte Arduino avec avrdude.  
  
Ensuite, je vérifie que compilation sous Windows fonctionne uniquement avec la procédure d'origine de Grbl décrite dans le wiki : https://github.com/fra589/grbl-Mega-5X/wiki/Compiling-grbl-Mega-5X
  
Atmel Studio n'étant disponible qu'avec Windows, je vais avoir du mal à reproduire les erreurs que tu obtiens.  
Peux-tu essayer la procédure standard avec l'IDE Arduino pour vérifier ?  
  
@++;  
Gauthier.

abdelabdel1, 2021-05-09:
Comment faire avec l'IDE arduino ? il n'ya pas de .ino ou .pde dans le dossier ?

abdelabdel1, 2021-05-09:
J’arrive a le compiler  également avec VSCODE et PLATEFORMIO.
Merci beaucoup Gauthier, je ne comprenais pas d’où venait le problème et j’étais loin d'imaginer que cela pouvait provenir d'un problème de soft.
Aaaaah la French Touch !!!

abdelabdel1, 2021-05-09:
[image]

fra589, 2021-05-09:
Attention,

Tu ne dois pas modifier les lignes 200 à 206 du fichier cpu_map.h !
Cela doit rester comme ça :
```
  //  #define LIMIT_INT       PCIE0  // Pin change interrupt enable pin
  //  #define LIMIT_INT_vect  PCINT0_vect
  //  #define LIMIT_PCMSK     PCMSK0 // Pin change interrupt register
  //  #define LIMIT_MASK ((1<<X_LIMIT_BIT)|(1<<Y_LIMIT_BIT)|(1<<Z_LIMIT_BIT)) // All limit bits
  // Hardware limits interrupts are not implemented in the RAMPS version of grbl-Mega
  // since all the limit switch ports are not compatible with interrupts.
  #define DISABLE_HW_LIMITS_INTERRUPT
```

Cela provient du vieux code spécifique Arduino Uno (AT mega 328) et je n'ai pas encore eu le temps de nettoyer tout ce vieux code qui n'est plus compatible avec grbl-Mega-5X.

Pour activer les limites matérielles, il suffit simplement de décommenter la ligne 214, toujours dans cpu_maps.h comme ça :  
(lignes 208 à 214)
```
  // Enable Hardware limit support for RAMPS without using interrupt...
  // Warning! bouncing switches can cause a state check like this to misread the pin.
  // When hard limits are triggered, they should be 100% reliable.
  // The RAMPS_HW_LIMIT is implemented inside the stepper driver interrupt. Depending of your
  // hardware, this can affect the max speed possibility of movments
  // Disabled by default for performance optimization, uncomment to enable.
  #define ENABLE_RAMPS_HW_LIMI
```

Et c'est tout ce qu'il y à à faire !

@++;
Gauthier.

abdelabdel1, 2021-05-09:
fichier original avec seulement   #define ENABLE_RAMPS_HW_LIMITS validé 

[image]

abdelabdel1, 2021-05-09:
[image]

abdelabdel1, 2021-05-09:
[image]

abdelabdel1, 2021-05-09:
[image]

abdelabdel1, 2021-05-09:
[image]

[image]

abdelabdel1, 2021-05-09:
merci beaucoup!! l'erreur de compilation était du  a cela !!!