fra589/grbl-Mega-5X #19: Ramps 1.4 + LCD 2004

lgojniczek, 2018-10-20:
Hi

Is there any chance to support LCD 2004 on this great project?

fra589, 2018-10-21:
Hi @lgojniczek,

LCD 2004 support is not planned for this time...
My plan is to use cn5X++ packaged on SD card for Raspberry pi or equivalent micro linux plateform, with a 1024x600 LCD touch screen for for providing screen on my grbl-Mega-5X CNC.

@++;
Gauthier.

closed 2018-10-21 by fra589

thawkins, 2019-09-01:
I would be willing to work on this, I have several RAMPS CNC setups and I have recently purchased a bunch of 2004 LCD boards, There is support for the 2004 LCD in marlin that i can use as a guide, does anybody have a list of functions that would be required?

My initial List was:

1.  Home Machine
2.  Return to Zero
3.  Set Zero (All Axis)
4.  Set Zero (Z Axis only)
5.  Jog X (with Encoder)
6.  Jog Y (With Encoder)
7.  Jog Z (With Encoder)
8.  Jog A (With Encoder)
9.  Jog B (With Encoder)
10. Select and print from  SD Card. 

Once my cards arrive i can get started, but it may be a few weeks. 

I have submitted a pull request to add platformio support, which makes the building and debugging easier.