fra589/grbl-Mega-5X #20: 4 linear axis

lukino87, 2018-11-08:
Hi his is really a wonderful job that I would like to adapt to my cnc.
Would it be possible to go back to 4 linear and one rotary axes? if you are modic, I have to give thanks for the help.

lukino87, 2018-11-08:
What changes do I have to make?
@ fra589

fra589, 2018-11-09:
Hi @lukino87,
You just need to adapt the config.h file with the followings items :
`  #define N_AXIS 5            // Number of axes`
`  #define N_AXIS_LINEAR 4     // Number of linears axis
`
then, adapt the names of your supplementals axis :
- One of U, V, W letter for the 4th linear axis,
- one of A, B, C letter for the 5th axix.

If you need more information for the choice of thoses names, see http://linuxcnc.org/docs/2.7/html/gcode/machining-center.html.

@++;
Gauthier.

If you want to help, you can give what you want here:
Grbl-Mega-5X is an open-source project and fueled by the free-time of our intrepid administrators and altruistic users. If you'd like to donate, all proceeds will be used to help fund supporting hardware and testing equipment. Thank you!

[image]

lukino87, 2018-11-09:
thank you very much for the information and in the control exe program how can I change the movements of the letters of the axes

[Donate]<https://paypal.me/pools/c/842hNSm2It>

closed 2018-11-09 by fra589