fra589/grbl-Mega-5X #21: Problems with IJK comands.

Enducross, 2018-11-11:
Hello, I have tried your version of GRBL and it seems like a great idea. I have cloned the X and Y axes, and in linear movements it works correctly. But I have detected that in circular or arc motions, also with the commands g28, g30 and g10 (non modal gcode) it does not clone the movements of the axes and it does not succeed in realizing the trajectory well. 
There is some problem when grbl intérprets ijk commands. 
Can you think of a way to solve it? Thank you.

fra589, 2018-11-12:
Hi @Enducross,

You are right ! I published the version with axis cloning a little bit too fast :-(   
It still lacks some lines of code to be complete ... I will try to correct it as soon as possible.  

@++;  
Gauthier.

Enducross, 2018-11-12:
Hi, Thanks for the reply. I'm trying to correct the program but there are a lot of variables. If you can guide me with what to correct I will try to do it.
I already have change the spindle pwm pin location to pin D44 (pwm 5v).

fra589, 2018-11-13:
Hi,  
The work is in good progress, updates are essentially in gcode.c and in motion_control.c...
I'm almost done in gcode.c patch (~ 250 lines added), for motion_control.c, I need to make a major change in the mc_arc() function.  
I hope to publish corrected version soon (perhaps before the end of the week ?).  
@++;  
Gauthier.

RaphaelDives, 2018-12-01:
Hey,
Great work so far! Waiting for the new version, to use it on my MPCNC!
Thank you very much!
Raphael

Enducross, 2018-12-02:
Hi Raphael, I have posted on the MPCNC forum a modified versión for our machine, it works great with linear commands, so im working with laser raster.You can test it. 
Witout the IJK problems it will be the Best controller for our machines, so its worth waiting for the solution.

fra589, 2018-12-04:
Hi all,  
I just push a new version (VER:1.1i.20181204) for correcting this bug.  
But I have no operational machine at this time, so I can't test with real motors.  
@RaphaelDives & @Enducross please, test this new version and report here if it is OK (or not)...  
@++;  
Gauthier.

RaphaelDives, 2018-12-04:
Hey, Thank You very much!!
Couldn’t wait till tomorrow, so I put it on my MPCNC aaaaaand it works like a charm! 
(With real motors ;-)
Some circles at first and a little milling project afterwards! More testing will follow these days, but it looks very good at first glance!
Raphael

Enducross, 2018-12-06:
Hi, today I´ve been testing it, it works perfectly with IJK commands BUT , there´s a problem with homing cycles, I have modified config.h file in the following way:

#elif N_AXIS == 5 // 5 axis : homing
    #define HOMING_CYCLE_0 ((1<<AXIS_1)|(1<<AXIS_4))
    #define HOMING_CYCLE_1 ((1<<AXIS_2)|(1<<AXIS_5))

Then each motor goes perfectly to the min limit switch when homing, but when homing is done, it can´t interpret circles or arc movemnts.
I have tried to only put:
 
#elif N_AXIS == 5 // 5 axis : homing
    #define HOMING_CYCLE_0 (1<<AXIS_1)
    #define HOMING_CYCLE_1 (1<<AXIS_2)
but it only moves 1 stepper.

It would be great that when doing homing with the cloned axes, each stepper runs towards its limit switch (axis1 to limit1 and axis 4 to limint 4) at the same homing cycle. Can you fix it?

Thankyou for your hard work.

fra589, 2018-12-06:
Hi @Enducross,  
`#elif N_AXIS == 5 // 5 axis : homing`
`#define HOMING_CYCLE_0 ((1<<AXIS_1)|(1<<AXIS_4))`
`#define HOMING_CYCLE_1 ((1<<AXIS_2)|(1<<AXIS_5))`
is the right method to do cloned homing cycle.  
So, I don't understand why the circle and arc movements don't work after thoses homing cycle...

Can you post your config (lines 51 to 83 of config.h), and the $$ result of your grbl parameters ?

I need to investigate and simule your config to find your problem.

@++;
Gauthier.

Enducross, 2018-12-06:
Hi I think it isn´t a config problem, this is my config:

// Axis array index values. Must start with 0 and be continuous.
#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  #define N_AXIS 5            // Number of axes
  #define N_AXIS_LINEAR 3     // Number of linears axis
#else
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X'
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'X' // Letter of axis number 4
#endif
#if N_AXIS > 4
  #define AXIS_5 4
  #define AXIS_5_NAME 'Y' // Letter of axis number 5
#endif
#if N_AXIS > 5
  #define AXIS_6 5
  #define AXIS_6_NAME 'C' // Letter of axis number 6
#endif
#if N_AXIS > 6
  #error "N_AXIS must be <= 6. N_AXIS > 6 is not implemented."
#endif

And here you era the GRBL settings:

$0=10	Step pulse, microseconds
$1=255	Step idle delay, milliseconds
$2=0	Step port invert, mask
$3=28	Direction port invert, mask
$4=0	Step enable invert, boolean
$5=0	Limit pins invert, boolean
$6=0	Probe pin invert, boolean
$10=1	Status report, mask
$11=0.020	Junction deviation, mm
$12=0.002	Arc tolerance, mm
$13=0	Report inches, boolean
$20=0	Soft limits, boolean
$21=1	Hard limits, boolean
$22=1	Homing cycle, boolean
$23=31	Homing dir invert, mask
$24=200.000	Homing feed, mm/min
$25=2000.000	Homing seek, mm/min
$26=200	Homing debounce, milliseconds
$27=2.000	Homing pull-off, mm
$30=1000.	Max spindle speed, RPM
$31=0.	Min spindle speed, RPM
$32=0	Laser mode, boolean
$100=200.000	X steps/mm
$101=200.000	Y steps/mm
$102=3200.00 Z steps/mm
$103=200.000	A steps/mm
$104=200.000	B steps/mm
$110=7500.000	X Max rate, mm/min
$111=7500.000	Y Max rate, mm/min
$112=350.000	        Z Max rate, mm/min
$113=7500.000	A Max rate, mm/min
$114=7500.000	B Max rate, mm/min
$120=400.000	X Acceleration, mm/sec^2
$121=400.000	Y Acceleration, mm/sec^2
$122=400.000	Z Acceleration, mm/sec^2
$123=400.000	A Acceleration, mm/sec^2
$124=400.000	B Acceleration, mm/sec^2
$130=640.000	X Max travel, mm
$131=640.000	Y Max travel, mm
$132=120.000	Z Max travel, mm
$133=360.000	A degrees ¿? Maybe here is the problem
$134=360.000	B degress ¿?

I think the problem is in A and B configuration, they are configured in step per degree and they need to be configured in step per mm.

Thankyou!
Pablo

Enducross, 2018-12-06:
Hi Gauthier, all problems have been solved changing $133 and $134 . All works perfect!!!
This is the best software for a MPCNC! I´ll post about it in the MPCNC forum, I have modified cpumap.h to change the spindle PWM pin, it allows to control a TTL laser.

Thankyou so much for your work, I´ll work with your software from now.
If I found some bug I will tell you.

Bye, Pablo

closed 2018-12-07 by fra589

decarpentier, 2019-12-07:
I am having issues with ijk commands on my cloned y axis (y2). The motor does not energize. The same motor works fine in Marlin using the ijk commands on g2 and g3.