fra589/grbl-Mega-5X #23: Modification for 4 axis and 3 axis homming cycle, later add 4th homming cycle

luisluna2, 2018-11-19:
Hello dear @fra589 , how are you?, I hope fine!

I want to use your GRBL for 4 axis only, and for a special reason I need use it with 3 homing cycle because it is still about to reach the 4th axis, and you could tell me when the 4th axis arrives how to modify to have 4 homing cycle, could let me know what I have to modify to achieve it.

For you attention, thank you!

fra589, 2018-11-19:
Hi @luisluna2 ,  
Homing cycles are defined in the config.h file (line 132 to 170).
For N_AXIS == 4, the definitions are on lines 147 to 151.
@++;
Gauthier.

luisluna2, 2018-11-23:
> Hi @luisluna2 ,
> Homing cycles are defined in the config.h file (line 132 to 170).
> For N_AXIS == 4, the definitions are on lines 147 to 151.
> @++;
> Gauthier.

thank you dear, I'am checking this!

closed 2018-12-04 by fra589