fra589/grbl-Mega-5X #25: Fix HOMING_SINGLE_AXIS_COMMANDS bug

fra589, 2018-12-13:
Bug reported 4 days ago by @stackola in pull request

In motion_control.h, the 4th 5th and 6th axis homing cycles are defined as HOMING_CYCLE_A/B/C, but in system.c they were being referenced as HOMING_CYCLE_1/2/3
This lead to a compile time error when enabling the feature HOMING_SINGLE_AXIS_COMMANDS

Thanks for using grbl-Mega-5X and reporting bugs.  
By the way, your patch work with simple axis config, but it is not compatible with cloning ant renaming axis new functionality.  
So, I will not merge your pull proposal, I will correct this bug with cloning and renaming compatibility  ASAP.
Thanks for your report.
@++;
Gauthier.

_Originally posted by @fra589 in https://github.com/fra589/grbl-Mega-5X/pull/24#issuecomment-446114681_

fra589, 2018-12-13:
Hi @stackola,  
I just pushed a fix for this bug, but I don't have real CNC compatible for testing this...
Can you test and report if it's OK ?
Thanks,
@++;
Gauthier.

RaphaelDives, 2018-12-17:
Hey, 
is there some test-cycle for testing all the stuff? I would provide my MPCNC for testing, since I appreciate your work and I like to test new things. ;)
I'd say I'm not bad at programming, but I can't find the time to work me into the code, so maybe I could help this way… :)
Raphael

fra589, 2018-12-18:
Hi @RaphaelDives,  
Many thanks for your help proposal !  
Since my hardware is still under development, I can't make all the tests by myself... I also miss time, but for hardware construction. :smirk: :hourglass_flowing_sand:  
The test of this fix consists in activating the functionality HOMING_SINGLE_AXIS_COMMANDS in config.h (line 191) and to test the homing axis by axis as well as all the axes sets.  
@++;  
Gauthier.

RaphaelDives, 2018-12-19:
Hey,
Compiling with HOMING_SINGLE_AXIS_COMMANDS worked, homing with $HX and $HY worked for me, as well as my homing cycle via $H. I did some short projects afterwards and everything looked okay.
I did not test the $HZ yet, since I have no limit switch there.
I'm working with cloned axis on X and Y with one limit switch for each motor.
Cheers
Raphael

fra589, 2018-12-19:
Hi,
Many thanks for your help / reply, I can close this issue now 😃 
@++;
Gauthier.

closed 2018-12-19 by fra589