fra589/grbl-Mega-5X #26: 6 axis - XYZ linear, ABC rotational

johnopsahl, 2019-01-04:
My goal is to achieve 6 axis control (XYZ linear, ABC rotational) without the RAMPS board. I am using DM556 stepper drivers. I should be able to copy the "CPU_MAP_2560_RAMPS_BOARD" in the cpu_map.h and change the output pins correct? Anything else that you can think of that should be changed?

THORR-1, 2019-01-04:
I don't think there's any need to change anything in order to use the DM556 drivers, assuming you don't mind having somewhat messy wiring. If you look through the CPU map you'll see in the comments the actual Arduino pin (D# or A#) that corresponds to a given motor/function. You should be able to just hardwire the DM556 into these, though I could be wrong. I'm actually working on a project currently and intend to take that route, so let me know if you come across any issues and I'll do the same.

johnopsahl, 2019-01-05:
My plan is to update the CPU map so wiring is less messy.

THORR-1, 2019-01-05:
Gotcha, mind letting me know how it goes?

Another option would be to use a RAMPS board just for passthrough, and use header connectors where the step-sticks go.

fra589, 2019-01-05:
Hi @johnopsahl,  

> (...) I should be able to copy the "CPU_MAP_2560_RAMPS_BOARD" in the cpu_map.h and change the output pins correct? Anything else that you can think of that should be changed?

This should do the job for your need.  

> My plan is to update the CPU map so wiring is less messy.

Update the "CPU_MAP_2560_INITIAL" with the same syntax and organization as "CPU_MAP_2560_RAMPS_BOARD" to make non Ramps hardware compatible with the Ramps code is in my TODO list. Your idea is the begining of this job. After this, it will possible to simplify a lot of code by removing the non Ramps code. Let me know your wiring updates so I can use it for this job.  

@++;  
Gauthier.

johnopsahl, 2019-01-05:
I am working on the new wiring layout for 6 axis without RAMPS. Any idea why RAMPS 1.4 board uses analog pins for x and y step pulse? 
[image]

johnopsahl, 2019-01-05:
See attached .pdf document for my proposed 9 axis wiring layout. At this time, I am only looking to use 6 axes but I expect to need additional in the future so I included pin placeholders at this time. Is it even possible to use more than 6 with grbl-Mega-5X? 
[grbl-Mega-5X_wiring9axis.pdf](https://github.com/fra589/grbl-Mega-5X/files/2729794/grbl-Mega-5X_wiring9axis.pdf)

CzRobotics, 2019-01-06:
I like your project very much. I would also like to use pinout according to johnopsahl PDF.  

Ota

johnopsahl, 2019-01-06:
@fra589 

With only 8 axes the wiring is actually more convenient. See revised pin mapping for 8 axes attached.
[grbl-Mega-5X_wiring8Axis.pdf](https://github.com/fra589/grbl-Mega-5X/files/2730650/grbl-Mega-5X_wiring8Axis.pdf)

CzRobotics, 2019-01-06:
6 axes is ok for my purpose (8 would fit in the future)
The important thing at this point is how to change pinout.
Can you tell me where to find the pin definition? I'm not a programmer, but I would have tried it :-) I would save a lot of work with the unpleasant soldering of many wires :-)

johnopsahl, 2019-01-06:
@CzRobotics 

I created a fork of grbl-Mega-5X for the six axes cpu map without the RAMPS board here: [https://github.com/johnopsahl/grbl-Mega-5X](url)

I created a new cpu pin map in cpu_map.h called "CPU_MAP_2560_6X" that is consistent with the 8 axis pin map that I have proposed. Note that is it only for 6 axes at this time because grbl-Mega-5X currently only supports up to 6 axes. I also commented out the RAMPS board define in config.h and added a define for the new "CPU_MAP_2560_6X".

 I plan to test these changes sometime in the next two weeks when I get the chance. If you liked my wiring layout and can afford to wait a couple of weeks then you won't need to change any code because I will have already done the work for you. Otherwise, pinouts are defined in the cpu_map.h file. You can find pin port definitions by just searching for an Arduino Mega 2560 pinout diagram.

johnopsahl, 2019-01-15:
I have completed a round of testing and revised the cpu map "CPU_MAP_2560_6X" of my fork to better organize the wiring. I changed it so the step, dir, and enable output pins are adjacent and in line per each stepper driver. See rev A of the pin layout document for details
[grbl-Mega-5X_wiring8axis_RevA.pdf](https://github.com/fra589/grbl-Mega-5X/files/2757778/grbl-Mega-5X_wiring8axis_RevA.pdf)

Additionally, I have also added six hard limit switches with pin change interrupt.

Everything seems to be working at this point. My fork: [johnopsahl/grbl-Mega-5X](https://github.com/johnopsahl/grbl-Mega-5X)

@fra589 
Why include inputs for both min and max limits in grbl-Mega-5X? Why not force them to be wired in series and take only one input pin per axis?

walterwissmann, 2019-01-15:
> @fra589
Why include inputs for both min and max limits in grbl-Mega-5X? Why not force them to be wired in series and take only one input pin per axis?

A good question, John.
I think, Gauthier will have a better control on which direction an axis collapsed with his end...

Which gcode-sender do you have? Right now i have the bCNC on RasPi 3 B+

Walter

RaphaelDives, 2019-01-15:
Hello @johnopsahl, @walterwissmann 
as an educated guess, I would say the first reason is the original grbl code and the second, that grbl-mega-5x was developed for the RAMPS board, that is physically build with min an max limit switches for x,y,z axis.
In my opinion there a definitly cases where these limit switches make sense - but as far as I know, you can easily combine them by defining the same port and pin for each max and min limit or even for all switches together.
Cheers
Raphael

johnopsahl, 2019-01-15:
@walterwissmann 
I have been using [CNCjs](https://github.com/cncjs/cncjs) with my six axes machine.

@RaphaelDives 
The ramps board explanation makes sense to me. Thanks

walterwissmann, 2019-01-15:
Yes, for me also.

fra589, 2019-01-16:
Hi all,  

Raphael gave the right reasons :
1) The gnea/grblMega original project I forked have this behavior,
2) The Ramps card have those 6 pin group for min/max switch for X, Y and Z.
However, I think it's really not helpful, and I plan to use the 6 pins group for the 6 axis.

@++;  
Gauthier.

qiwalker, 2019-03-13:
@johnopsahl 

how to wire limit switch i use 6 axis grbl firmware but pin no respon when i try to trigger Pin
i use ramps board

walterwissmann, 2019-03-13:
When you use the original cpu.h map and follows the configuration: https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-pinout

then you should have NO-Switches to the rams-board.

Fusseldieb, 2019-03-29:
Hey @johnopsahl , I've tried out your fork but didn't get any of the motors to work.

Are the pin definitions right in your CPU_MAP_2560_6X? A motor is on pin 0 and at the same time other things also use 0. 0 in my mind is the serial connection and cannot be used.

Can you shed me some light?

Thank you

Fusseldieb, 2019-04-03:
> Hey @johnopsahl , I've tried out your fork but didn't get any of the motors to work.
>
> Are the pin definitions right in your CPU_MAP_2560_6X? A motor is on pin 0 and at the same time other things [...]

It seems like I figured it out.

Looking at the image from [here](https://forum.arduino.cc/index.php?topic=138519.0), you see all PORTs on the Arduino. Axis must all be on the same PORT probably due to performance/stability/sync gains.

So... The fork from @johnopsahl is a great start, but lack correct pins, only the pin comments and PORT letters are right (at the time of writing).

> My fork: [johnopsahl/grbl-Mega-5X](https://github.com/johnopsahl/grbl-Mega-5X)

What I've done is assign the right pin "numbers" to the respective variables and now the motors move, like:

```
  #define STEP_BIT_0 PA0  // X Step - Pin D22
  #define STEP_BIT_1 PA1 // Y Step - Pin D23
  #define STEP_BIT_2 PA6  // Z Step - Pin D28
```

If you question yourself "What the heck is **PA0**", well, simply put, it's pin D22 without the sugar coating. You can't type "D22" in there, because it won't work.

But don't be afraid, the image tells you all. PORT PA0...7 is in sequence like PA0, PA1, PA2, PA3... 

This means that:

```
D22 = PA0
D23 = PA1
D24 = PA2
D25 = PA3
...
```

I hope I helped some soul out there lmao

johnopsahl, 2019-04-04:
> Hey @johnopsahl , I've tried out your fork but didn't get any of the motors to work.
>
> Are the pin definitions right in your CPU_MAP_2560_6X? A motor is on pin 0 and at the same time other things [...]

@Fusseldieb 
Though I haven't done extensive testing with my 6 axis machine yet, I didn't have any issues getting the motors to run properly when running my fork. The structure of the code is to first define the ports and then define the pin bits. The same pin bit of 0 is used but for different ports. See my proposed 8 axis pin map .pdf document that I posted previously on this thread for clarification on ports, bits, and pin labels. Image search for "mega 2560 pin mapping" online for a visual.

johnopsahl, 2019-04-04:
@Fusseldieb 
I am excited that you are getting some use out of my fork. Hope the pin map is working for you now. The step pulse, direction, and stepper enable/disable pins shouldn't need to be on the same port. I am not deep enough into the code to be able to tell you why, but I can tell you that even the RAMPS board 5X cpu map doesn't keep each of them on the same port. The limits switch and auxiliary control switch pins each have to be on the same port and specifically on ports configured with pin change hardware interrupts.

qiwalker, 2019-04-15:
hi ... @johnopsahl 

can i use NC limit for 6 axis firmware ?

thanks
Rizki

Fusseldieb, 2019-04-16:
> can i use NC limit for 6 axis firmware ?

I am not johnopsahl, but yes, you can hook these all up and then invert the function of the limit switches.

To make these all work, you must also send the homing command with X Y Z A B C.

qiwalker, 2019-04-20:
Hi...
Thanks for the answer it work perfectly
One more question
Can i ported grbl on arduino due ?

Thanks
Rizki

fra589, 2019-04-20:
Hi Rizki,
Porting to Arduino Due is part of my TODO list 😃   
This should not be too difficult except to find the time to do it.
@++;
Gauthier.

qiwalker, 2019-04-21:
hi fra

I'm glad to hear that on your TODO list
cannot wait to use more of your creation :)
may I know when u will start to do it ?

thanks 
Rizki

closed 2019-06-07 by fra589