fra589/grbl-Mega-5X #27: Cannot homing when use 3 axis

qiwalker, 2019-01-05:
hallo, can you help me, i use grbl-mega 5x for 3 axis, it work awesome, but when i try to homing it cannot homing, please help and guide me to solve this problem

thanks :)

RaphaelDives, 2019-01-05:
Hello @qiwalker, 

it's a very good decision to use grbl-mega 5x! ;)

A little more details about your homing problems would be very convenient...
For example: Does the homing start? Is it homing to the right direction? Do you have any cloned axis? Have you changed the original code?

To adjust the homing cycle please refer to config.h:146ff 

Cheers
Raphael

qiwalker, 2019-01-06:
Hi @RaphaelDives

I only enable core xy in config.h
I have try to use 5axis and when i try to homing it work stepper move, but for 3 axis stepper not moving :(

Thanksss :)

fra589, 2019-01-06:
Hi @qiwalker,  

Are you sure you have homing enabled in your Grbl configuration ($22=1) ? 
Sometime, when you change the axis numbers or other compile option, you need to force clears and restores all of the EEPROM data used by Grbl by issuing the $RST=* command. Then, reconfigure all your specifics parameters.  

I don't have COREXY machine to test this configuration, so, when I forked from the original [gnea/grbl-Mega](https://github.com/gnea/grbl-Mega) for adding 5/6 axis with cloning, renaming, etc. functionalitys, I didn't make any hardware tests. I only studyed it in the theory.  
I will try to make some test to find what can happend to you...  
But since I am very busy right now, it may take a while.  
During this time, you can test with the original [gnea/grbl-Mega](https://github.com/gnea/grbl-Mega) to know if your problem was already present or if it comes from additions that I developed.  

@++;
Gauthier.

fra589, 2019-01-24:
Hi @qiwalker,
No reply from you ? Is this still an issue ?

qiwalker, 2019-01-28:
hi @fra589 

i already test from original GRBL-Mega i got same problem
i already solved the problem by comment all //#ifdef DEFAULTS_RAMPS_BOARD :(
and use

#define HOMING_CYCLE_0 (1<<AXIS_3)  // COREXY COMPATIBLE: First home Z
#define HOMING_CYCLE_0 (1<<AXIS_1)  // COREXY COMPATIBLE: First home X
#define HOMING_CYCLE_1 (1<<AXIS_2)  // COREXY COMPATIBLE: Then home Y

and alter in limi.c  line 347 changed sys.homing_axis_lock[idx] = axislock[idx]; to sys.homing_axis_lock[idx] = 73; to enable z axis homing

i have one more question and req, can GRBLMEGA-5X use for scara ?

Thanks
Rizki

closed 2019-04-06 by fra589