fra589/grbl-Mega-5X #28: not an issue (state of things)

walterwissmann, 2019-01-24:
Hello all,

After i have finished my MEGA-Shield i can help to do many tests for grbl with more axis ;)
Now i am here:
[image]

The shield has the pin-layout from John as wrote here up to 6-aixsis:
https://github.com/fra589/grbl-Mega-5X/issues/26

Then it should work in his case:
[image]

br
Walter

walterwissmann, 2019-02-04:
sorry, opened now ;)

walterwissmann, 2019-02-04:
How can i reopen it?

fra589, 2019-02-05:
Wow ! Amazing job :clap: :clap: :clap:

walterwissmann, 2019-02-05:
Thanks Gautier,

The plan is i will give the gerber files to a professional to let make the PCB. Then i will test them. I think in around 10 days i can say is the PCB ok or not ;)

The shield has the pin-layout from John as wrote here up to 6-aixsis:
#26
John sayes:
` I have completed a round of testing and revised the cpu map "CPU_MAP_2560_6X" of my fork to better organize the wiring. I changed it so the step, dir, and enable output pins are adjacent and in line per each stepper driver. See rev A of the pin layout document for details
[grbl-Mega-5X_wiring8axis_RevA.pdf](https://github.com/fra589/grbl-Mega-5X/files/2757778/grbl-Mega-5X_wiring8axis_RevA.pdf)

Additionally, I have also added six hard limit switches with pin change interrupt.
`
I think that is  good way.

BR
Walter

walterwissmann, 2019-02-19:
Hi,
a little update:

[image]

I think to the weekend it runs ;)

BR
Walter

n0th1n6, 2019-02-20:
Just subscribing here. When it work, do you have a plan to sell boards?

jflapao, 2019-02-20:
If you sell the board, what the price?
I'm interest in one board for mega

José Lapão

walterwissmann, 2019-02-20:
Ok, we see this board has potential.

But all the thinks about this board can i and / or fra589 not alone ;)
We need help.

BR
Walter

walterwissmann, 2019-02-22:
Hi,
 i have today done an addition to the FC.
This addon let me switch between automatic and hand steering the FC.
Also it has plugs for all cables and an On-Off-Switch.
[image]

All the connections to the MEGA-Shield and the AddOnFC i have tested and the lines are ok :)
[image]

BR
Walter

walterwissmann, 2019-02-24:
Hi,
It is not ready assembled yet, but i can say all the parts on the PCB and the AddOnFC workes great together :)
Here the curve of the PWM to 0-10V converter:
I have to linearise it a little bit more... but for now and for me it looks good. 
[image]

Could it be i found a bug in the grbl software?
I could not switch the revolution direction directly of the spindle from CW to CCW, but from CCW to CW is ok

CW to CCW:
M3 S10000
M4 S9000
M5
M4

Then CCW to CW:
M3 s10000

BR
Walter

fra589, 2019-04-06:
Hi Walter,

> Could it be i found a bug in the grbl software?
I could not switch the revolution direction directly of the spindle from CW to CCW, but from CCW to CW is ok

For me, switch from CW to CCW to CW... work well :  
```
Grbl 1.1i ['$' for help]  
[MSG:'$H'|'$X' to unlock]  
[VER:1.1i.20181204:]  
[AXS:5:XYZAB]  
[OPT:VNMH,35,255]  
ok  
ok  
$X  
[MSG:Caution: Unlocked]  
ok  
M3  
ok  
M4  
ok  
M3  
ok  
M5  
ok

```

@++;
Gauthier.

walterwissmann, 2019-04-06:
Hi Gauthier,
Hmm, nice to hear this... :) I have to search ;)

BTW: 
See at the PWM-Diagram above.
Would be nice, we could lineryse the PWM with software ;)

BR
Walter

fra589, 2019-04-12:
Hi Water,
Grbl software already have a linear model adjust function for PWM spindle :smile:   
See ENABLE_PIECEWISE_LINEAR_SPINDLE in config.h (parameters from line 658 to 680),  
There is a documented Python script to calculate values in doc/scipt/fit_nonlinear_spindle.py.
@++;
Gauthier.

walterwissmann, 2019-04-12:
Hi Gauthier.

Wow, thanks a lot! :-)

BR
Walter

closed 2019-06-07 by fra589