fra589/grbl-Mega-5X #30: Enabling Z touch probe

HLDMYBR, 2019-02-01:
I'm working on my MPCNC and have mega-5x running it. Auto square/homing is working fine. However I can't get the z touch plate to work. I use the probe widget in cnc.js, when the probe touches the z continues moving down without any response.

If I manually touch the z probe and type in '?' to the console I get this:
> ?
<Alarm|WPos:1218.000,608.000,-16.968,1218.000,608.000|Bf:35,254|FS:0,0|Pn:Z|Ov:100,100,100>

I think the Pn:Z is telling me it is detecting the z touch, since that is not there if the z isn't touching.

Any ideas to troubleshoot this?

fra589, 2019-02-02:
Hi @HLDMYBR 
You ar right, Pn:Z is the marker to say that the Z end stop is active... But...
In the original gnea/grbl-Mega, the hardware limits ars disabled with the CPU_MAP_2560_RAMPS_BOARD configuration (code for the interrupt definition is not writed for this config).
You can just use the limit switch to do the homing, after, you must to use software limits with a good configuration to ensure move inside limits.
It is in my TODO list to write this code... But I don't known when.
@++;
Gauthier.

HLDMYBR, 2019-02-02:
> In the original gnea/grbl-Mega, the hardware limits ars disabled

I have $21=1 (Hard limits enable, boolean)

Is this setting not enabled?

fra589, 2019-02-02:
It is in the code, not in the Grbl's config.

In the file cpu_map.h, we have in line #311 :
`  #define DISABLE_HW_LIMITS
`
Because in the limit.c file, the code is not compatible with the pin definition of RAMPS for enabling the pin change interrupt user by hardware limits detection.  
@++;
Gauthier.

ykutovoy, 2019-02-18:
My problem is this, $ 21 = 1, all end sensors are connected in series to form a closed loop, a break occurs when it is triggered, everything is connected to pin + X. In the DRO status, I see the activation of limit switches, but there is no stopping of the movement. I wanted to open a new request about the work of Hard limits, but carefully read this request and came to the conclusion that this is also my problem. Please explain if I use CPU_MAP_2560_RAMPS_BOARD, then Hard limits do not work?

fra589, 2019-02-19:
Yes, as I sayed before, hardware limits are not coded for the CPU_MAP_2560_RAMPS_BOARD configuration. It's in my TODO list, but I can't give a delay for it be usable.
Be patient and stay connected :-)
@++;
Gauthier.

ykutovoy, 2019-02-20:
Dear friend, if I use the definition of DEFAULTS_RAMPS_BOARD, but I do not use the RAMPS board. I have a lot of wires connected directly to MEGA. Is it possible to add definitions such as MYLIMITS +++ or MYLIMITS ---, which will allow the CPU_MAP_2560_RAMPS_BOARD section in the cpu_map.h file to replace the construction from "247 line" to "311 line" to the construction from "62 line" to "71 line", but using MIN_LIMIT_BITs(xyz) or MAX_LIMIT_BITs(xyz) respectively . And add changes to the limits.c file that will allow you to use the code from "221 line" to "232 line".

I hope my English is quiet understandable :)  
I'm not a programmer, so it's a difficult task for me. I would be appreciate if You help me. I use Yours firmware on real hardware and it's annoying to control limits by hand :)

fra589, 2019-02-21:
Hi @ykutovoy,  
  
Your idea is not so bad, but unfortunately for you, it can't either work.  
  
The construction of steppers commands registers and limit switches registers is not the same in CPU_MAP_2560_INITIAL definition than in the CPU_MAP_2560_RAMPS_BOARD definition, it is the reason for the hardware limit can't work actualy with CPU_MAP_2560_RAMPS_BOARD.  
  
Moreover, I have not yet programed the 6 axis mode for the CPU_MAP_2560_INITIAL (it's also in my TODO list)...  
  
While waiting for my futures work, you can use the logical limits ($20=1) with the limits switch on the min travel and a good configuration of the dimentions of your CNC machine ($130, $131, ..., $135).  
  
@++;  
Gauthier.

ykutovoy, 2019-03-28:
Dear friend, i've make fork of your repository, and make some changes in code to satisfy my needs. I'm far not a programmer so most of changes made by copy/paste of your code. In my case arduino mega without RAMPS board works perfect with 5AXISs and 3 hard limits.  I would happy if my changes  helps to anyone. By the way, it would be very helpful if you add some information about input pins in cpu_map file description. In case of using whole port for limit pins and whole but different port for user control it is needed to pay attention that not every port has external interrupts.

ykutovoy, 2019-03-29:
Here the links to video of working z-axis (simple on desk example)
first one, homing with hard limit (UGS)
[Video-1](https://youtu.be/7sbK3n84EEk)
and second z-probbing with touch probe(CNC.js)
[Video-2](https://youtu.be/AMnJ0kCC1iA)

closed 2019-04-06 by fra589