fra589/grbl-Mega-5X #32: connecting pwm/ttl laser

lesaux, 2019-02-24:
Hello, I have a blue laser like this one:

https://www.aliexpress.com/item/1m-1m-area-5000MW-big-DIY-laser-engraving-machine-500mw-2500mw-3000mw-diy-marking-machine-diy/32755630904.html?spm=a2g0s.9042311.0.0.38154c4dwb3t60

It is the "Full 7W" blue laser with a red case. It says both PWM and TTL in the description.

I'm a bit worried I could damage something while connecting it, so I would rather ask some advice here:
The code seems to mention to use pin D8 (I am currently using an mks base2 v1.2 https://www.thanksbuyer.com/3d-printer-motherboard-controller-mks-base2-v1-2-with-sd-card-for-diy-arduino-50647)
but this is the bed heater connector on my board. This doesn't feel right, and I'm afraid to fry my laser if I plug it there.

It feels like those smaller connectors A13 A15 A14 would be a better fit (same connector), I believe on a 3D printer they are used for the thermistors.

I'm looking for a bit of advice if anyone is familiar with this.

thanks.

walterwissmann, 2019-02-27:
Hi lesaux,

you need this firmware:
https://github.com/MarlinFirmware/Marlin

BR
Walter

lesaux, 2019-02-27:
Not sure why you recommend a firmware change. I have everything working fine with grbl-mega-5x on this board except for the laser pwm (I just need to plug signal and ground - my laser is externally powered).

n0th1n6, 2019-02-27:
You should implement a hardware solution that prevent the pin from
floating. It cant be fixed in firmware. The resistor is the easiest you can
do.

lesaux, 2019-02-28:
Thank you, I tried some resistors I had lying around on the signal and I see some progress. I will buy a proper 10k resistor tomorrow and see if it's better.

n0th1n6, 2019-02-28:
I just realize I replied on a wrong thread. But this still applies if you
are encountering the same problem as the other thread where the laser is
turned on just after applying power.

jflapao, 2019-02-28:
@lesaux Hi, I don't use the grbl on mega, because I never can made the end stops work on ramps 1.4, and size 3 years ago i use grbl on uno with cnc shield v3.0. On that i use the pwm pin 12 for run a ttl laser and a relay to turn on and off my dremel, i build a box with a power socket for the dremel and a socket for the laser. When i put the laser and the g-code is for laser the S go from 0 to 12000 the ttl works. When i connect the dremel the S goes directly to 12000 and the relay goes on. The laser drive turn on and off to, but no laser connected to wer, when the laser is the and is on full power (S12000) the relay come on to, but no dremel ther so i have all time the two ones connected to the arduino, but only the one is connected to the box works

closed 2019-04-06 by fra589

jransp, 2020-04-07:
PWM to TTL
PWM signal is the spindle **negative** signal wire

[image]