fra589/grbl-Mega-5X #34: Configure Laser-PWM to D6

Mad-Onion, 2019-03-04:
I'd like to use pin D6 for my laser-pwm-signal because I don't need a mosfet-controlled power-source as the laser has its own driverboard. I modified the ramps-section of `cpu_maps.h` correspondingly:
```
  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT   3 // MEGA2560 Digital Pin 6
```

Unfortunately I don't get any pwm-output - no matter if in lasermode or not. Did I make anything wrong?

walterwissmann, 2019-03-05:
Hi `Mad-Onion,

have you set the right register to PIN 3?

regards
Walter:
`

Mad-Onion, 2019-03-05:
Hi Walter,

that might be the little detail I forgot about. I haven't configured PWM-timers by registers for years :-D
I will have a look at it later and give you feedback.

Best regards
Tobias

Mad-Onion, 2019-03-05:
This is my current configuration, but instead of using PWM the output moves between 0 and 5 volts...

```
  // Advanced Configuration Below You should not need to touch these variables
  // Set Timer up to use TIMER4B which is attached to Digital Pin 8 - Ramps 1.4 12v output with heat sink
  #define SPINDLE_PWM_MAX_VALUE     1024.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
  #ifndef SPINDLE_PWM_MIN_VALUE
  #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif
  #define SPINDLE_PWM_OFF_VALUE     0
  #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

  //Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
  #define SPINDLE_TCCRA_REGISTER    TCCR4A
  #define SPINDLE_TCCRB_REGISTER    TCCR4B
  #define SPINDLE_OCR_REGISTER      OCR4A
  #define SPINDLE_COMB_BIT          COM4A1

  // 1/8 Prescaler, 16-bit Fast PWM mode
  #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
  #define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
  #define SPINDLE_OCRA_REGISTER   OCR4A // 16-bit Fast PWM mode requires top reset value stored here.
  #define SPINDLE_OCRA_TOP_VALUE  0x0400 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   DDRH
  #define SPINDLE_PWM_PORT  PORTH
  #define SPINDLE_PWM_BIT   3     // MEGA2560 Digital Pin 6
```

walterwissmann, 2019-03-06:
Hmm,
my code on PIN D8 with PH5 and T4C:

  // Advanced Configuration Below You should not need to touch these variables
  // Set Timer up to use TIMER4C which is attached to Digital Pin 8
  #define SPINDLE_PWM_MAX_VALUE     1024.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
  #ifndef SPINDLE_PWM_MIN_VALUE
    #define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
  #endif
  #define SPINDLE_PWM_OFF_VALUE     0
  #define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)
  #define SPINDLE_TCCRA_REGISTER    TCCR4A
  #define SPINDLE_TCCRB_REGISTER    TCCR4B
  #define SPINDLE_OCR_REGISTER      OCR4C
  #define SPINDLE_COMB_BIT          COM4C1

  // 1/8 Prescaler, 16-bit Fast PWM mode
  #define SPINDLE_TCCRA_INIT_MASK 	((1<<WGM40) | (1<<WGM41))
  #define SPINDLE_TCCRB_INIT_MASK 	((1<<WGM42) | (1<<WGM43) | (1<<CS41))
  #define SPINDLE_OCRA_REGISTER   	OCR4A // 16-bit Fast PWM mode requires top reset value stored here.
  #define SPINDLE_OCRA_TOP_VALUE  	0x0400 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

  // Define spindle output pins.
  #define SPINDLE_PWM_DDR   		DDRH
  #define SPINDLE_PWM_PORT  		PORTH
  #define SPINDLE_PWM_BIT   		5 // MEGA2560 Digital Pin 8

workes great...
How have you measure the output voltage?
Could it be, in lasermode is the PWM-output a On/Off-Switch?

BR
Walter

walterwissmann, 2019-03-09:
Update:
Lasermode On($32=1): 
no PWM
Spindel Enable: high with M3
Spindel Dir: high with M4
M5 both low

Lasermode off ($32=0):
PWM with S5000 (example)
Spindel Enable: high with M3
Spindel Dir: high with M4
M5 both low

BR
Walter

Mad-Onion, 2019-03-15:
Sorry for my late response, problem was already solved [here](https://github.com/gnea/grbl-Mega/issues/55#issuecomment-403261212). You have to watch out as it is a 8bit-timer instead of 10bit as the D8-pin.

closed 2019-03-15 by Mad-Onion