fra589/grbl-Mega-5X #36: Lose steps on b axis

jflapao, 2019-03-08:
Hi every one, I'm using the mega 5 axis on a experimental cnc.
Normaly i use a UNO with a cnc shield and work perfectly.
I use the $1=255 to have all the time the stepp motors with power (I have a fun attached to each one to keep cool)
I want to move to the mega to have 6 axis on my cnc xyz and a head with the c and a and a extra one for one lathe that i want to be interpolated with the others 5.
I'm testing the mega firmware and i verified some issues.
1° The $1=255 don't work like in the uno, in the uno all the motors keep with power all the time, with the mega from time to time the motors lose power and is not all at the same time, imagine one circle 1st the X the the y, then Z and so one.
2° the b axis lose power when work I have made this lines to test the head
G0 B135
G0 A360 B-135
G0 A0 B135
G0 B0
whit the G0 movement 95% the time do the job but if i do the same test but in G1 movemente even with F5000 the B lose power in the middle of the job (is random when lose power some time is in the beginning other's in the middle some time simple don't work)

At this time only have the head connected to the mega, so i think is not problems with the power source

fra589, 2019-03-09:
Hi @jflapao,  
  
I just checked the activation / deactivation code of the motors (stepper.c 254-267 & 326-339), it is very simple and I did not find what could cause this problem. I think rather a hardware problem ... Have you the possibility to disconnect the enable/disable pins and put it the good value of voltage (GND or +5, depending on your drivers) to be sure that your motors are always enabled and if the software is involved or not?  
  
@++;
Gauthier.

walterwissmann, 2019-03-10:
Hi @jflapao,

I think the same as Gauthier and have another point:
You say: I have a fun attached to each one to keep cool.
you should decrease the motor current at the Stepper Driver. When the MotorCurrend are to high, the Stepper Driver goes in a short emergency mode to protect itself...

Also have  look to $0: https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration

BR
Walter

jflapao, 2019-03-10:
Hi @walterwissmann,

The question of having a fan on each engine relates to the $ 1 = 255 (Step idle delay, milliseconds) setting. If we have a standard CNC with 3 X, Y, and Z axes where they are linear and there are no involuntary movements of not being jammed, there is no need to keep the motor always on load, even if stopped, but if we have a Z head that have the axis C and A, or even the A and B, we must have constant current in these motors or due to the weight of the spindle that axis will move involuntarily. Imagine B as an axis that moves between -90 and +90 or -135 and +135 as I want to do with mine where 0 is pointing down, so gravity helps, and engine B needed no load to is set to 0, but motor A or C because the weight is not centered on the axis of rotation would rotate unintentionally. But imagine that instead of a cutting edge tool we had a blade B would have to start working at -90 or +90 ai to keep B at those grades it is necessary to always have load. Actually there are things that can be modified in my CNC is true, change the drives that are at the moment the DRV8825 for some that keep the motor fixed even without power, true, but that does not justify why B does not always work, even with the configuration of $ 1 = 25 or even less, nor does it justify the reason why $ 1 = 255 do not keep the motors all connected simultaneously, since the GRBL for the UNO does, I already tested to have the CNC turned on for 72H without being to work, the motors heat up yes - hence the fans - but they are all locked all the time, in the GRBL MEGA not and it is soon as soon as the hardware

[image]
[image]
[image]
[image]

This is the kind of head that i want do to my CNC, and on top of this I want put a lathe like this i did
[image]
[image]
[image]

walterwissmann, 2019-03-10:
As i say,
take the motor current with the poti on the 8825 PCB a little bit down ;)
The Chip on that PCB goes in a little emergency halt.

Also have you played with the $0 ?

btw. : can not see the pictures

jflapao, 2019-03-10:
Hi @fra589 

My drivers at this moments are DRV8825 and I don't know do that but I have mesure the voltage on the enable pin with a multimeter  and have a 4.98V when is no activity on the motors, and 0.03V when is motors activity.

walterwissmann, 2019-03-10:
[image]
With that Poti you have to decrease the current to the Motor

jflapao, 2019-03-10:
Hi again @walterwissmann 

[image]
Lathe in test
[image]
This is with a milling down tool
[image]
With a laser
[image]
This is the head for the 5/6 axis.
I have to change the B axis (rotation over Y axis) for the C axis (rotation over the Z axis) but at this time the A axis are the C axis and the B axis are the A axis, because I write the code manualy is no problem
[image]
[image]
[image]

walterwissmann, 2019-03-10:
Nice work ;)
and otherwise: see above.

jflapao, 2019-03-10:
> With that Poti you have to decrease the current to the Motor

I know that, the pois is the drives ar the same that I use in the UNO, so I don't see the reason to change this value becouse is work perfectly on the UNO, I have done lots of jobs with the hardware like he is and work, but because i want made a 5/6 axis CNC for research porpose I have change to the MEGA 5/6 axis firmware, and I fund for now some insues

walterwissmann, 2019-03-10:
Do it! as i say!

jflapao, 2019-03-10:
@walterwissmann 

OK then what value should I have?

walterwissmann, 2019-03-10:
Take it step by step a little bit down.

walterwissmann, 2019-03-10:
Without words:

[image]

jflapao, 2019-03-10:
https://youtu.be/7ykaemQS-MU
One video with the head
https://youtu.be/zQ1WAqQauSI
One video with the lathe with the laser

walterwissmann, 2019-03-10:
nice work :)

Before i came to MEGA 5x i have also used the UNO. All it worked fine :) After i switched to the MEGA 5X  the mashine have had also lost steps. I decreased the current. Now all is fine ;)

fra589, 2019-03-11:
Hi,  
Instructions for setting the Pololu's current limit are available on the DRV8825 product page :  
https://www.pololu.com/product/2133  
@++;
Gauthier.

jflapao, 2019-03-12:
Hi, @walterwissmann @fra589 

Yes is some news, I have change the current on the Pololu's drivers and for now the drive and the motor don't heat all much nothing.

I can only work on the project on weekends because my job take me to much time, but I realize that the A axis (the C on my machine - rotation over the Z axis) when is interpolated with the B axis at a small amount of degrees ( I have tried with 5, and 10 degrees) is not smooth, have some bumps, but can be illusion because the structure is week.
I have to change some acceleration on the config and see what happens, then I give some news.

@fra589 

I don't understand nothing of code (all much nothing) but how can i change the axis letters to match my machine like you see on my test head I need the C axis (rotation over Z) to rotate 360º and a A axis (rotation over X) to rotate between +135º and -135º when the 0º is the spindle face down. I intend to put end stops on this axis to home them, but on the config the your A axis have a 360º rotation and your B axis have 180º rotation, how can i setup this to be +135º and -135º?

Thanks

fra589, 2019-03-12:
Hi @jflapao,  
Internaly, axis are known with their number. In config.h, you can change corresponding axis names by updating definitions of AXIS_1_NAME, AXIS_1_NAME,..., AXIS_N_NAME (line 60, 62, 64, 71, 75 & 79).  
  
360° and 180° are only default values for (respectively) axis number 3 (default to A) and number 4 (default to B), these defaults are defined in default.h and you can adjust it by using Grbl config variables $133 & $134.  
  
@++;  
Gauthier.

walterwissmann, 2019-03-12:
@jflapao:
with witch gcode sender do you send the code to grbl 5x?

i will also do some tests at more axis in the next weeks...

fra589, 2019-03-12:
Hi @walterwissmann,
You can use cn5X : https://github.com/fra589/cn5X, the gcode sender that I currently write.
@++;
Gauthier.

walterwissmann, 2019-03-12:
Yes i have seen it, Gauthier ;)
I am hot to install it on raspi 3 b+ ;)
For now i take a look on them at Windows...

And my first tests with mega 5x i have done with bCNC on Raspy 3 b+ ...
Also a great tool but for 3-Axis.
Autolevel, toolchange, probe/camera Alignment, Simulation, CAM-Tools, Editor, Terminal, Pendant, Plugins, Scripting and a big plus (it comes from Linux itself): remotedektop :)
https://github.com/vlachoudis/bCNC

jflapao, 2019-03-12:
@ walterwissmann

I have used the cnc5x from @ gauthier but I have a lots of problems with the connections and because my laptop is 4k the screen stay a lots small, i use the bcnc. You only see the information of the 3 axis X,Y, and Z but all the axis works, all the photos and videos i send is performed by bcnc

jflapao, 2019-03-12:
At this time I'm working, until 10pm in Portugal, when I arrive at home I can show you a video with bcnc working with the 5 axis. My brother they know code and is trying change the bcnc to 5 axis for me. If he can do we put im on github

walterwissmann, 2019-03-12:
Ohhhh yes, please:-)

walterwissmann, 2019-03-25:
@jflapao 
Any news?

BR
Walter

jflapao, 2019-03-25:
@walterwissmann 

Hi, is on the good way, is all ready show the axis, but miss some information, and we need to create the axis buttons, for my hardware, is on hold, this month i have a lots of work on my job, and i have no time, maybe next month.

closed 2019-06-07 by fra589