fra589/grbl-Mega-5X #37: A axis soft limit issue

hzxiaoyu, 2019-03-14:
XYZ AXIS with soft limits is OK,but A axis can't move after homing. How to fix it?

fra589, 2019-03-14:
Hi @hzxiaoyu,  
I need more info to help you...  
What is the configuration of your homing cycle?  
Does your home switch are at min or max value?  
What is the A axis move (value) than you try?  
When you try to move the A axis, do you have an error message?  
...  All details you can add :-)
@++;
Gauthier.

hzxiaoyu, 2019-03-15:
@fra589 
Hi, I set xy min   and z max , for A & B is 0 position. I changed A axis homing dir mask, now it can move + dir now , but it can't move more than 360,if I set $133=360, and the - dir not work.

fra589, 2019-03-15:
Hi @hzxiaoyu,
This is the normal behavior of standard CNC mahines:  
On normal CNC, end stops are always on the max axis positions, so, in machine coordinates system, the move destinations are always negatives, and must stay between 0 and -max_travel(axis) unless you change the homing dir mask, wich tell to the system that the homing switch is on min position.  
In your case, with $133=360 and homing dir mask inverted, the machin position of A axis must stay between 0 and 360.
It is in my TODO list to disable soft limit if max travel is 0, but it's not coded yet :-)
@++;
Gauthier.

fra589, 2019-05-15:
HI @hzxiaoyu,  
In the new version (1.1k.20190514) that I published yesterday, the new soft limit enhancement is implemented. :smile: 
The soft limit is disables for axis when the max_travel parameter ($130 to $135) is equal to 0 for this axis.  
In your case, with $133=0, the A axis will be able to move in + or - dir without limit (except the processor limit for float numbers as large as 3.4028235E+38 and as low as -3.4028235E+38)
@++;
Gauthier.

closed 2019-05-15 by fra589