fra589/grbl-Mega-5X #38: 6-Axis Prototype Breakout Board

PulseTrainHat, 2019-03-14:
Just received the PCB back for first prototype 6-Axis Breakout Board.

Have gone for Open Collector Buffers on each output and Opto Isolated inputs for each Limit input.

[image]

This is based on Ramps 1.4 configuration for pin out.

Also there are headers for a calibration PCB to be connected to the breakout board.
This calibration circuit is for any Axis that has two motors on it and will home each individually which will line them up correctly. 

Now to find time to solder up and test :)

Cheers

Sean

RaphaelDives, 2019-03-14:
Hey there, looks great! But all the work soldering SMD's…. Have fun! 😎

PulseTrainHat, 2019-03-14:
We have solder stencil for it and also Re-Flow oven, so not too bad a job hehe :)

raenji-sk, 2019-04-03:
Hello Sean, any update on price / availability?
Will it support inductive endstops as your GRBL Breakout/Shield for Arduino Uno CNC Controller Kit?
BTW that one works great, but I could use the auto squaring feature for the Y axis.

Andrej

closed 2019-06-07 by fra589