fra589/grbl-Mega-5X #39: Mega Ramps 1.4 CNC XYZ motors do not move

walterbux, 2019-04-01:
Hello,
I have built a 5 axes CNC XYZBC with Mega and RAMPS 1.4. Works perfectly with 4 of 5 axes with Repetier FW. No motion of XYZ motors with grbl-Mega-5X. Tested with GrblPanel apparently shows no error in the dialogue but no motion. The motors are Nema 23 with DQ542MA drivers.
Any advice ?
Thanks for the very interesting work and project.

Walter

fra589, 2019-04-01:
Hi Walter,  
Have you make a complete factory reset ($RST=*) after changing the firmware ?  
Is it possible than some Repetier FW params are stored in the eeprom and are incompatible with Grbl...  
Grbl does not reset the eeprom by itself.   
@++;  
Gauthier.

walterbux, 2019-04-01:
Hi Gauthier,
happy to hear from you !
No, I have not made any reset.
I retry and give you an answer..
See you soon.

Walter

walterbux, 2019-04-01:
Hi Gauthier,
I have sent the reset. Same behaviour, no motion of x, y, z axes.
No error in the dialogue.

Walter

walterbux, 2019-04-01:
When I connect the usb port I hear the noise that make motors when energized: a little shot but after no motion. As a note I have made a standard connection of the drivers: I attach the schematic. I have not the second Y motor. The extruders are not connected.
Walter
[image]

fra589, 2019-04-01:
Hi Walter,
Since the signals (step pulse) are +5v impulses, I think you have wrongs wiring for your drivers.
The good wiring should be this :  
[image]
@++;  
Gauthier.

walterbux, 2019-04-01:
The driver I/F allows my wiring..
Walter

[image]

walterbux, 2019-04-01:
And even your advised wiring.

Walter

[image]

walterbux, 2019-04-02:
Hello Gauthier,
problem solved with the commands:
$2=7
$3=7
$4=1

Now the motors XYZ are working ...
:) 👍

fra589, 2019-04-02:
Cool ! Issue closed :smiley:

closed 2019-04-02 by fra589