fra589/grbl-Mega-5X #40: discuss about backlash

walterwissmann, 2019-04-01:
Hi,

@Schildkroet hase done routines for backlash: https://github.com/Schildkroet/GRBL-Advanced

Would it be nice to have also this routines for MEGA 5X ?

BR
Walter

fra589, 2019-04-01:
Hi Walter,
I think it is a very good idea :)
Thi is added to my TODO list... But the list is long.
@++;
Gauthier.

jgalvarezr, 2020-10-03:
Hi...
I have been pending of this case waiting for its solution. My CNC has a lot of inertia and fails at high speeds.

I understand that this is altruistic work, but I would like to know if there will be a solution.

Thanks for your excellent work.

fra589, 2020-10-04:
HI @jgalvarezr,  
  
Since I'm retired, summer is more convenient for journey than for computer programing 😃   
If backlash in always in my TODO list, it's a pretty complicated thing to implement and it's not a priority over other features that I intend to write next winter. So, I can't tell you that it will be available before sevral month.  
However, this feature should help solve mechanical backlash problems, not for inertia solving problem.  
If your CNC have a lot of inertia and fail at high speed, you need to adjust the acceleration parameters of your axis (parameters $120 to $125).  
I had similar problems when I started out, and fine-tuning maximum speeds and accelerations allowed me to solve them.  
  
@++;  
Gauthier.

walterwissmann, 2020-10-05:
@jgalvarezr:
I can confirm Gauthier's arguments.

I would also like to mention:
When you have problems at high speed, you can also play with the current of your stepper drivers...

Bests
Walter

jgalvarezr, 2020-10-11:
I don't know if it will be appropriate to continue this thread, but I need help to solve this problem. Accelerations work well in most cases. If I make a Y + movement and then an XY + the X motor does not accelerate, it tries to move instantly at the displacement speed, at that moment steps (all steps) skip. This occurs in some combinations of movements but not in all.

fra589, 2020-10-11:
Hi @jgalvarezr,  
  
It will be better for all that you open a new thread for this specific case.  
Don't forget to post your config ($i, $$, etc.).  
A maximum of info is needed to save time.  
  
@++;  
Gauthier.

davidlehrian, 2023-04-30:
I have implemented backlash on this code base as per https://github.com/shooter64738/grbl-mega. My changes are here https://github.com/davidlehrian/grbl-Mega-5X. If you are interested I can create a pull request and you can integrate these changes into your code base.

fra589, 2023-05-02:
Hi @davidlehrian,  
  
Thanks for your proposal, backlash is still in my TODO list...
I'm traveling all the May, I'll watch this when I get back.  
  
@++;  
Gauthier.

Matthias199, 2024-11-06:
@fra589 still on your todo list ?

fra589, 2024-11-07:
Hi @Matthias199,  

Yes, it's still on the list, but I'm pretty busy on another projects...

@++;
Gauthier.

davidlehrian, 2024-11-07:
Hello @Matthias199,
I have a branch of this code where backlash has been integrated and is available at https://github.com/davidlehrian/grbl-Mega-5X-backlash. I have been using it and it works great.

Matthias199, 2024-11-08:
@davidlehrian Yeah i know that!
PLZ again the question was "did you think of an auto backlash compensation with the endstops"

This could help many people without tools to streamline the calibration.
Since voron uses klicki probes ... the regular endsops should be okey