fra589/grbl-Mega-5X #41: Communication errorr using cn5X

walterbux, 2019-04-02:
Hello Gauthier,
I receive this error message when connecting to the MEGA.
If I try another sender like GrblPanel I have no error while connecting.
What to do ? May I change some interface setup in cn5X ?

[image]

walterbux, 2019-04-02:
In debug mode I have ...

[image]

fra589, 2019-04-03:
Hi @walterbux,  
At first look, it seems you use a Windows version that is not the up to date development one...  
Can you post the "About cn5X" screenshot please ?
@++;
Gauthier.

walterbux, 2019-04-03:
Here it is..
Moreover I see the same behaviour described in cn5X issue 1:
https://github.com/fra589/cn5X/issues/11
"...starting first the serial monitor of arduino ide then its software instead the whole connects and works without any error...".

Walter

[image]

walterwissmann, 2019-04-03:
Hi Walter,
I think, that is an 'Earth Issue' ;)

BR
Walter <- ( Namebrother ) :)

walterbux, 2019-04-03:
I don't have a picture: I connect my laptop battery powered to a mega + ramps board with only the usb cable. no other power than that usb provided and the issue is there.
But, if I open Arduino correctly configured to flash grbl to Mega with the correct port number and I open the serial monitor, I see the grbl answer. I close the Arduino IDE. I open cn5X and everything is ok. no error.
If I power on the pc and simply open cn5X and connect the error appears.
My os is the last version of Windows 10.
I have installed Python 3.7 64 bits.

Thank you again for your work.

Walter

walterwissmann, 2019-04-03:
Ahh, ok.

Could you please install this little serial terminal program and test it with the same procedure?
https://github.com/pyserial/pyserial/blob/master/serial/tools/miniterm.py

fra589, 2019-04-03:
> @walterbux wrote :  
> But, if I open Arduino correctly configured to flash grbl to Mega with the correct port number and I open the serial monitor, I see the grbl answer. I close the Arduino IDE. I open cn5X and everything is ok. no error.
> If I power on the pc and simply open cn5X and connect the error appears.
> My os is the last version of Windows 10.
> I have installed Python 3.7 64 bits.
> (...)

Ho, interesting behavior !  
I'll be able to test this tonight, and perhaps understand this issue soon...

By the way, the latest version is 0.3.g, but changes since 0.3.f don't affect initial connection sequence. Therefore, the bug must always be present.

@++;  
Gauthier.

walterbux, 2019-04-03:
Both before and after pc restart no error or anomaly...

Walter

[image]
[image]

walterwissmann, 2019-04-03:
Awesome :)
Now, Gauthier should have enough info to find the bug ;)

BR
Walter

walterbux, 2019-04-03:
Yes, very good :), many thanks. but do not forget to give me the link to download the last revision of cn5X sw.

Walter

fra589, 2019-04-04:
Hi Walter(s) ;-)  
It seem to be a cn5X specific problem on Windows...  Can't reproduce on Linux, Perhaps the code have too speed for W$ :-)  
Can you try the latest cn5X version (0.3.h) from here : https://github.com/fra589/cn5X ?  
And if the problem persist, post the new debug informations.  
@++;  
Gauthier.

fra589, 2019-04-04:
Hi @walterbux,
But what about the new Debug outputs ? I added some new message in the grblComSerial.py...
Now, when it work fine, the new debug output is :
```
cn5X++ (v0.3.h) : Starting debug.
grblCom.startCom(self, ttyUSB0, 115200)
grblComSerial.__openComPort(self)
grblCom.on_sig_connect(self, True)
grblComSerial: comPort ttyUSB0 open, RC = True
grblComSerial : Wait for Grbl init... T = 1554390449681 ms...
grblComSerial : Wait for Grbl init... T = 1554390449957 ms, Nothing to read on serial.
grblComSerial : Wait for Grbl init... T = 1554390450232 ms, Nothing to read on serial.
grblComSerial : Wait for Grbl init... T = 1554390450508 ms, Nothing to read on serial.
grblComSerial.__openComPort() : Buffer recu : "b"\r\nGrbl 1.1i ['$' for help]\r\n[MSG:'$H'|'$X' to unlock]\r\n""
grblComSerial : Grbl init string received in 1077 ms, OK.
grblCom.on_sig_init(self, Grbl 1.1i ['$' for help])
```
I don't nees image, just copy/past the full new debug output text of cn5X connection attempt from Debug tab...

@++;
Gauthier.

walterbux, 2019-04-04:
Here it is.
I think nothing special.

I add that in the old version I have tried to extend the timeout from 5 to 20 s. Same behaviour.

[image]
[image]

fra589, 2019-04-05:
Hi Walter,
Since I am not able to reprocuce your problem on my Windows system, I am forced to grope...
The only thing I am sure, is that on your system, Grbl don't send his initialisation string as he normaly do on mine.
In new cn5X version (0.3.i), I added a first timeout of 1500 ms. If I don't receive the Grbl init string before this first timeout, I try to send a soft reset to Grbl, which must force him to send his init string. 

Can you try this new version ?
@++;
Gauthier.

walterbux, 2019-04-05:
Hi Gauthier,
I understand you very well, no problem.. It is strange that you can not reproduce the error on your windows system because mine has nothing special..
I don't know where is the prolem but as signalled in the previous message I have tried to extend the timeout from 5 to 20 s inside the grblComSerial.py file, see attached point, without any result. Same behaviour therefore there is some condition that prevents the opening of the port except if I open before the same port with the Arduino Ide Serial monitor.
See you soon.

Walter

[image]
. Same behaviour.

fra589, 2019-04-05:
When you open the port with arduino serial monitor, does it show you it's initialisation string as showed in the following picture?    

[image]
  
Have you tested the cn5X version that I posted this morning with a soft reset added in the initialisation sequence ?  

@++;
Gauthier.

walterbux, 2019-04-05:
Hi Gauthier,
I repeat every test step for completeness:
1-pc boot
2-open a cmd window as admin
3-cd where it is .. python cn5X.py
4-select the arduino port
5-open
6-the cn5X app thinks some seconds and after shows the usual error (tested with version cn5X-0.3.h, last in github) your msg this morning saying "Can you try this new version ?" had nothing attached.

AFTER I start Arduino IDE
I do nothing else but opening serial monitor configured to send "NL + CR" !!!!!!! at 115200 baud.

Immediately grbl sends its message exactly that of your picture.

AFTER I reopen your app cn5X and magically it works !!!

Try a version that sends NL + CR just after the port is opened !!

I hope to have been clear.
See you soon.

Walter

fra589, 2019-04-05:
Yes, It's clear.
But this morning, I pushed the version 0.3.i on the github repository! => 2 files are modified from 0.3.h :  cn5X_config.py and grblComSerial.py... 
This version send a Ctrl + X to the serial after opening the serial port. This should do the job like a CR+LF...
If not, you could test replacing the line number 245 :  
`self.__sendData(REAL_TIME_SOFT_RESET)`
by this one :  
`self.__sendData("\r\n")`
@++;
Gauthier.

walterbux, 2019-04-05:
Wow Gauthier !!!

It works !!! :) :)

Your Ctrl + X is good as CR + LF !!
Thank you !
It has a good experience to work with you !

Walter

p.s. see you on next issue :)

walterbux, 2019-04-05:
It has been a good.. to be quick I lose the words.

Walter

closed 2019-04-06 by fra589