fra589/grbl-Mega-5X #42: Min Limits on RAMPS

walterbux, 2019-04-04:


walterbux, 2019-04-04:
Hi Gauthier,
while trying to debug my limit switches (in grbl I set $10=16 or 18) I find the following:

Xmin limit mechanical NO works perfectly
Ymin limit does not work
Zmin limit does not work
Ymax limit NO works as Ymin
Zmax limit NO works as Zmin
Looking at cpu_map.h the pins indicated in the comment are correct.
I am sure the HW connections are ok because the same limits work perfecly with Repetier FW for 3d printer configuration on the same machine.
I don't understand what is happening.
Any advice ?

walterbux, 2019-04-04:
I try to reword the problem after some testing:
Apparently both Min and Max limits triggering are reported in a unique status bit sending the command
"?"
so if I close the Ymin end-stop OR I close the Ymax end-stop the reporting is always Y.
I was interested in reusing the Ymax and Zmax pins for Amin and Bmin Limits triggering (A and B) axes. Is there any way to do so ?
Thanks.

Walter

fra589, 2019-04-04:
Hi @walterbux 
This behavior is normal since in the original Grbl code, there is only one input pin for both min and max limit switch. In the grbl-mega, both signals trigger the same code.
If you move in the direction of the max the end-stop reportin mean the max othewhise, you move in the min direction, so is the min switch triggered :-)
@++;
Gauthier.

walterbux, 2019-04-04:
I understand but for compatibility wrt the 3D printer it wuould be nice to use MAX Limit inputs for A,B and C axes. Do you know where to modify to implement this, only in cpu_map.h ?

Thanks.

Walter

fra589, 2019-04-05:
Hi Walter,
Yes, you can do it in cpu_map.h, then, you need to suppress/disable with a #ifdef preprocessor directive, all references to MAX_LIMIT* in limit.c and setting.c.
@++;
Gauthier.

walterbux, 2019-04-05:
Hi Gauthier,
I have solved my personal problem by switching pins only: I have used in my 3D printer Xmax pin for other purposes therefore for compatibility I have only the Ymax and Zmax pins to be used for A and B axes.
I simply switched the Ymax and Zmax pins with Amin and Bmin ... and voilà les jeux sont faits, rien ne va plus.. see attached file.
I agree with you that this may be not so clean but I don't use max pins at all ..
In the future it may be good to allow people to change pin usage without modifying many files around the code.

Thanks for all.

Walter

[cpu_map.txt](https://github.com/fra589/grbl-Mega-5X/files/3047890/cpu_map.txt)

closed 2019-04-05 by fra589