fra589/grbl-Mega-5X #43: DRO for 4 and 5 axis

qiwalker, 2019-04-12:
Hi Fra

i got some problem here for value of 4 and 5 axis
when i jog machine 1mm value in my DRO is 1.012 (just in 4 and 5 axis)
for x, y ,z axis value is correct only show 1.000
kindly please advice for this value of 4 and 5 axis

thanks 
Rizki

fra589, 2019-04-12:
Hi Rizki,
Since a stepper motor can only move step by step, depending of step by unit configuration ($103 and $104 for axis 4 and 5), the motor move will go to the step with the nearest value you requested.
@++;
Gauthier.

qiwalker, 2019-04-21:
Hi fra

The problem i found in DRO for 4 and 5 axis 
For the step is good precision on 1 degree but DRO display 1,012

Thanks
Rizki

fra589, 2019-04-22:
Hi Rizki,
Can you post your $$ Grbl params ?
@++;
Gauthier.

fra589, 2019-05-08:
Hi @qiwalker  
Still have your problem ?
If yes, post your $$ Grbl params (and all other precisions that can help us to find the solution), if no, close this issue.
@++;
Gauthier.

qiwalker, 2019-05-13:
Hi fra

Its oke now
Thankyou very much
Apologize for late replay

Thanks
Rizki

closed 2019-05-13 by qiwalker