fra589/grbl-Mega-5X #44: Random issue

digitalmess03, 2019-04-12:
I'm not sure what issue I am currently dealing with. LaserWeb4 or GRBL-Mega-5x firmware issue. I'm thinking this is probably more of a GRBL/LW4 issue but trying to collect as much debug as possible and hoping you might be able to help narrow this down?

I have a custom built laser engraving unit, Mega 2540 with the Ramps 1.4 board, 4 drivers, 2 dual Y, 1 X, 1Z. The cloned driver works as expected so a big thank you for working that out. This attaches to a Raspberry Pi 3 via USB (Has touch screen interface attached, probably means nothing but for the sake of full disclosure). The rPi 3 is running LaserWeb4 and is joined to my wireless network. My workstation is hard wired to the network and has the LaserWeb4 Windows Bin running as the head.

Everything works manually. I can connect to the device, manually jog the unit full up and full down, left/right/ Z up and down. no problem at all... from 1 mm/min to 7000 mm/min it runs good manually. When I load an image and generate the code for 1000mm/min everything is good. When I send it to the unit. It always freaks out, some times it moves diagonally gets to the point of where it should start burning with laser and makes gawd aweful sounds from the steppers. It'll either lock or just get stuck studdering in one specific direction. Only one time did I get it to start burning properly and that was when it was set to 100mm/min. Other than that one time I get random lock out errors. My current image that is continuously causing it is 270mmx300mm.

error: 9 - G-code locked out during alarm or jog state
ALARM: 8 - Homing fail. Cycle failed to clear limit switch when pulling off. Try increasing pull-off setting or check wiring.

I have no homing/limit switches, no soft limits. Not entirely sure what exactly is going on. Any help you could provide would be greatly appreciated.

 $0=10
 $1=25
 $2=0
 $3=3
 $4=0
 $5=0
 $6=0
 $10=3
 $11=0.020
 $12=0.002
 $13=0
 $20=0
 $21=0
 $22=0
 $23=0
 $24=100.000
 $25=500.000
 $26=250
 $27=1.000
 $30=1000
 $31=0
 $32=1
 $100=44.440
 $101=44.440
 $102=2560.000
 $103=44.440
 $104=44.440
 $110=3000.000
 $111=3000.000
 $112=1000.000
 $113=3000.000
 $114=3000.000
 $120=1500.000
 $121=1500.000
 $122=300.000
 $123=1500.000
 $124=1500.000
 $130=800.000
 $131=800.000
 $132=100.000
 $133=800.000
 $134=800.000

walterwissmann, 2019-04-13:
Hi,
Sounds like not connected limit input pins...
Ok the lines over the PCB are like antennas. 
Could you please connect the Limit-pins to ground or +5V with a resistor 1k to test it?

BR
Walter

fra589, 2019-04-13:
Hi @digitalmess03,
Sound from steppers can also come from step lost, try using lower acceleration value ($120 to $124).
Be carrefull to have the same parameters on clonned axis.
@++;
Gauthier.

walterwissmann, 2019-04-13:
yes, of corse ;)

digitalmess03, 2019-04-13:
@fra589 @walterwissmann Thanks for the suggestions, out of the house right now, but I'll try the accel. change first and then the limit pins. I had used the same GRBL config from my Arduino Uno w/Protoneer CNC Shield that worked without issue previously. On the ramps board I have 3 jumpers on each stepper for 1/16 micro stepping incase that changes your guys thoughts any.

digitalmess03, 2019-04-14:
Dropped the Accel. to 50.000 for each axis and it stopped the steppers squealing and running off in abnormal directions. Realized after the fact that it was supposed to be 150.000 not 1500.000. Thank you for pointing out my oversight! 

Have another weird issue that with the laser switched off (hardware switch) and the barrel connector plugged securely in, the LED light on the ramps board indicates that the laser should be on, but if it's slightly plugged in, it doesn't turn on unless I send M3 S###. I have a 12v barrel connected to a switch, to a 12v ps. The other end goes to a blackbuck m8 r3.1 board 12v + and - respectively. The PWM is connected to the Ramps D8 + to Buck Mod IN and D8 - to Buck GND. Thoughts on what is occurring?

closed 2019-05-07 by digitalmess03