fra589/grbl-Mega-5X #47: Ramps 1.6 A and B axes never activated

marcow1601, 2019-05-03:
I am currently trying to get grbl Mega 5X running on an Arduino Mega + Ramps 1.6 setup and am hopelessly lost in finding an initial issue.

I have flashed the software onto the Arduino without changing anything. I have plugged in all 5 DRV8825 drivers in the correct orientation and plugged in all 5 motors.

On power-up, all 5 motors engage (they hold their position actively).
If I launch bCNC and try jogging the x-, y- and z-axis, the steppers attached to the X,Y,Z-drivers move without any issues.

I can't by any means however get the steppers attached to the E0 and E1 drivers to move.

I would be very very happy about any hits what might be wrong here.

walterwissmann, 2019-05-03:
Hi Marco,
bCNC is only for three axis for now ...

But you can do tests with Fra's gCode-Sender:
https://github.com/fra589/cn5X

BR
Walter

marcow1601, 2019-05-03:
Hey Walter,

thank you for your super quick reply.

I actually don't require 5 individual axes for my machine, but only the regular 3 linear axes with the X and Y axis having 2 motors. This is why I was expecting the A and B motors to simply mirror the motion of the X and Y axis motors respectively.

Do I have to configure anything special in the Mega 5X software to enable this behavior?

walterwissmann, 2019-05-03:
No...

You can double the motors by hardware ;-)

I am not a specialist for the ramps board, but all drivers with CLK, EN and Pulse can be take parallel on the same hardwaresignal ;-)

Are there Jumpers to do that?

marcow1601, 2019-05-03:
Okay, but doubling the motors by hardware makes me loose the ability to control the two motors individually in software which would be bad. I still need to be able to control the E0 and E1 motors individually during the homing sequence.

walterwissmann, 2019-05-03:
They are not parallel on one axis?

BTW: https://github.com/fra589/grbl-Mega-5X/issues/28

marcow1601, 2019-05-03:
Actually, I just stumbled upon Issue #4 which ends in a software solution implemented by Gauthier giving the functionality I am requiring, I unfortunately must have missed that when reading the issues for the first time.

I will try this right now and then report the current state!

walterwissmann, 2019-05-03:
Ok. The followers will thank you;-)

marcow1601, 2019-05-03:
Reading Issue #4 carefully actually solved all the problems I had very quickly.

After renaming axis_4 to 'X' and axis_5 to 'Y' in config.h the E0 and E1 drivers now copy the X and Y moves respectively, just how I intended them to do.

Thank you so much for your assistance, Walter, and thanks Gauthier for the great software.

closed 2019-05-03 by marcow1601

rikkie, 2019-05-04:
Hi Marco this is the changed code for my homebuilt dual y-axis laser engraver
https://github.com/rikkie/GRBL-DUAL-Y-Axis