fra589/grbl-Mega-5X #48: grbl ramps 1.4 5 axis (x,y cloned) plus appropriate homing

alexpGH, 2019-05-12:
Hi all,
I try to figure out how to achieve an MPCNC dual endstop appropriate solution based on grbl.
I'm a bit confused about the branches, version etc. and would be glad for any advice.
My aim is to allow for the MPCNC setup (described below) as contibution to any branch as close as possible to the main stream of grbl.
My impression is that this would be the branch to use, due to the 5 axis extension. Is this correct?
However, I see old open pull requests - so would I have a chance to get the modification in here if successfully implemented?
  
**What would we need for MPCNC (dual endstop)**

- The mpcnc uses cloned x and y axis. Details can be found [here](https://www.v1engineering.com/auto-square-dual-endstops/)
I think I read somewhere that the 5ax extension allows for cloning - so would this be easily achieved? If so, any pointer where to read about it?

- Z needs to be able to move without homeing (this is a simple config issue, I read somewhere)

- The hard part (an I saw it requested here already) would be the auto squaring.
The guys solved it for marlin and I would try to convert it to grbl.
Strategy is to move the cloed axis bot towards endstops. If either fires, stop that motor, let the other continue. So this would mean to temporarily 'break' the cloning. Any chance to achieve this?

A last question: I saw a comment by chamnit about the upcoming rework of grbl for ARM support. So I wonder on which grbl version is this here based?

RaphaelDives, 2019-05-12:
Hey there,
since this is a topic that comes up once in a while, please check older posts for details. ;)
Long story short, it’s worth trying it. 
(Someday, when I have too much time, I will write some words about this evergreen...)

> Hi all,
> I try to figure out how to achieve an MPCNC dual endstop appropriate solution based on grbl.
> I'm a bit confused about the branches, version etc. and would be glad for any advice.
> My aim is [...]
Yes, just use the standard version by fra589...

> **What would we need for MPCNC (dual endstop)**
> 
Endstops? ;) (Just because I’m curious... what do we need dual endstops? I have one on each axis (min) and never missed the max yet. Or am I mistaken here? Or do you mean the thing about squaring?)
I used the min and max endstop header on the ramps board for minx1/minx2 and miny1/miny2. Just take care that the max definitions points to other pins... (I also wrote that before... but it’s a nasty mistake that keeps me searching for one day.)

> * The mpcnc uses cloned x and y axis. Details can be found [here](https://www.v1engineering.com/auto-square-dual-endstops/)
>   I think I read somewhere that the 5ax extension allows for cloning - so would this be easily achieved? If so, any pointer where to read about it?

That’s implemented and works like a charm. See older posts, I have written about this two or three times. (I could do it again, if you want me to do. ;)

> * Z needs to be able to move without homing (this is a simple config issue, I read somewhere)

Just leave Z out of the homing cycle... (Although I’m  currently working on an endstop for my Z - for safe moving height and for my cross laser and camera. ;)

> * The hard part (an I saw it requested here already) would be the auto squaring.
>   The guys solved it for marlin and I would try to convert it to grbl.
>   Strategy is to move the cloned axis both towards endstops. If either fires, stop that motor, let the other continue. So this would mean to temporarily 'break' the cloning. Any chance to achieve this?
> 
That’s also implemented. Just set the correct pins and homing cycle... In my case it was necessary to switch the order of x and y homing. I think I have one of the bigger MPCNCs.

> A last question: I saw a comment by chamnit about the upcoming rework of grbl for ARM support. So I wonder on which grbl version is this here based?

I would say on the previous one. :P Don’t take this serious, I really have no idea. But most important, I’m very very happy for this piece of software and for my case it’s working perfectly!

Happy milling!

alexpGH, 2019-05-13:
Hi Raphael,
thanks for your quick and highly appreciated answer!
As Gauthier stated as comment to your linked answer 'your reply is perfect'.
I will test that. Can I 'copy' your recipe and write it up in the mpcnc forum (referencing at you), where I started a FOSS toolchain tutorial?
Thank you!
Alex

fra589, 2019-05-13:
Hi @RaphaelDives,
Thanks for answering so well!  
Your help for after sales service is welcomed :smile: 
@++;
Gauthier.

closed 2019-05-13 by fra589

RaphaelDives, 2019-05-13:
Hi @alexpGH,
I have no problem with that - like I said: If I had the time I would do a documentation about configuring the grbl-mega-5x for MPCNC. (but job, family and all unexpected stuff… ;)
The most important things are up there… If you run into problems feel free to ask, maybe I can remember my solution. ;)
Btw. I'm running my MPCNC with the ramps board and a raspberry with bCNC. I tried the Freecad thing myself, but I was not confident with the results. (I like Freecad very much for 3d-print modeling though…)
In the end I stumbled about CamBam (http://cambam.info/) and up to now it's a perfect solution for me. 

@fra589 - Where should I send the bill? 😀 

Cheers
Raphael

alexpGH, 2019-05-22:
Ok, 
i started som docu on gh pages https://alexpgh.github.io/foss-toolchain-mpcnc/
Which version of Freecad did you try? The newer ones?