fra589/grbl-Mega-5X #49: Invert Limit Pin problem:

CzRobotics, 2019-05-14:
Please help. I have MEGA 2560 with grbl-Mega-5X in default settings (downloaded as is). When is set Invert limit pins $ 5 = 0, everything is OK. 
But there is a problem when $ 5 = 1 is set. The system does not respond to a change the pin's status. Whatever the pin state is Low or Hi,  GRBL always reports Alarm on XYZAB pins!!! 
UGS Platform status: [verbose] <Idle | WPos: 3.9370,0.0000,0.0000,0.0000,0.0000 | FS: 0.0.0 | Pn: XYZAB>

Some solution ? 

Standard Grbl MEGA https://github.com/gnea/grbl-Mega  work properly ...

fra589, 2019-05-14:
Hi @CzRobotics,

HO ! This is a bug !!!
You're lucky, I had some time this afternoon, and I was able to fix it :smile:  
I just pushed VER:1.1k.20190514  
Can you please test it and tell me if it is OK ?
@++;
Gauthier.

CzRobotics, 2019-05-14:
Wow ... great news in such a short time. I'm going to test it ...

CzRobotics, 2019-05-14:
I did a basic test with a default configuration and a 6-axis configuration - 6 Linear and the pins react as they should ... BIG THANKS ...
In the next few days I will try 5 axes - limit switch on a specific machine. I will also change the mapping of all pins. I'll let you know what the result will be ...

closed 2019-05-15 by fra589