fra589/grbl-Mega-5X #50: renaming axis does not rename status report

sinfocomp, 2019-05-15:
i have renamed axis to  Z X Y instead of X Y Z however if i move Z on de GUI it shows a X move, phisically it moves the correct axis however the reported move is on another axis

RaphaelDives, 2019-05-15:
What benefit do I have by changing the names over connecting my motors to different pins? ;) 
(If that is not possible, then I would try to swap the pins in cpu_map.h:151ff...)
Cheers Raphael

fra589, 2019-05-15:
Hi @RaphaelDives,
You are right, but renaming axis in config.h is more simple than swap ports and pins in cpu_map.h...

In fact, in my final think, if axis are renamed or cloned in config.h, actual standards GUI that don't known those new grbl-Mega-5X functionalities will don't work correctly.

@sinfocomp made me think of this potential problem, so, I added 2 new options in config.h :  
`#define SORT_REPORT_BY_AXIS_NAME` wich make the outputs of axis values reports in axis name order,
`#define REPORT_VALUE_FOR_AXIS_NAME_ONCE` wich make that the output value of one axis_name appear only one time if the axis is cloned.

Those two options are not enabled by default.
I pushed this new version (1.1l.20190515) few minutes ago.
@sinfocomp please, test it and tell me if this work well for you.

@++;
Gauthier.

sinfocomp, 2019-05-16:
> Hi @RaphaelDives,
> You are right, but renaming axis in config.h is more simple than swap ports and pins in cpu_map.h...
>
> In fact, in my final think, if axis are renamed or cloned in config.h, actual [...]

Thanks for the quick answer and help, i will give it a try and get back

sinfocomp, 2019-05-16:
Excellent work, tested with various clone and renames of axis and worked ok
Closing issue

closed 2019-05-16 by sinfocomp