fra589/grbl-Mega-5X #51: how to clone y axis and auto square , Please help

sinfocomp, 2019-05-15:
Hello, i need to have dual Y axis with auto square feature however i dont find the info on how to setup like this

RaphaelDives, 2019-05-15:
Hello @sinfocomp,
here: https://github.com/fra589/grbl-Mega-5X/issues/48#issue-443119303
and here: https://github.com/fra589/grbl-Mega-5X/issues/4#issue-321316303
Good luck. ;)
Cheers
Raphael

sinfocomp, 2019-05-16:
so to be able to auto square menas to have limits at same place to make square gantry ?
is like having adjustable bum screws but this time adjusable limits on each side of dual motor axis
am i right ?

RaphaelDives, 2019-05-16:
I don't really know what bum screws are, but I would say yes. ;) 
I'd say it's just to ensure parallelism of the gantry. Therefore I have my limit switches on the corners, that should hopefully be square and trigger plates on each side of my gantry, so both (cloned) steppers can perform homing independently. 
Cheers
Raphael

sinfocomp, 2019-05-16:
ok, got it working, closing Thanks

closed 2019-05-16 by sinfocomp

coilheaddesigns, 2019-07-05:
@sinfocomp what did you change. I'm not winning.
My homing is inverted but jogging is correct

fra589, 2019-07-06:
Hi @coilheaddesigns,
You should learn to read :smiley:   
Look in the wiki (https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration), you may find the answer with the grbl parameter: $ 23 - "Homing dir invert, mask"
@++;  
Gauthier.

coilheaddesigns, 2019-07-06:
But it's only the cloned axis that homes in the wrong direction.

coilheaddesigns, 2019-07-06:
@fra589 yeah you're right think I've found the error on my side, apologies.

I'll play around and will get it working👌