fra589/grbl-Mega-5X #52: Feature request, support ESC RC servo spindles

sinfocomp, 2019-05-18:
look here its done on esp32 port of grbl, could it be implemented on this branch ?
[https://github.com/bdring/Grbl_Esp32/issues/148]

fra589, 2019-05-20:
Hi @sinfocomp,
As @langwadt tell you on your issue you posted on the original [gnea/grbl-Mega](https://github.com/gnea/grbl-Mega/issues/99), you can probably already do that with the right combination of PWM frequency and min/max value...
The grbl-Mega-5X goal is to provide a firmware for milling machines more like industrial machines. Currently, the driving of model-making servos is not in my TODO list.
I think [the post you quoted](https://github.com/bdring/Grbl_Esp32/issues/148) should give you enough information to do it by yourself.
@++;
Gauthier.

closed 2019-05-20 by fra589

enriquesmw, 2021-05-16:
Hi @sinfocomp, you should check this version: https://github.com/alnwlsn/grbl-Mega-5X-servos (thanks to @alnwlsn) . It's a fork of the current version (thank's @fra589 for this also), maybe the fork should be updated, I'm still not figuring out how to make it works (like with the https://github.com/robottini/grbl-servo of @robottini),

I hope this helps your project,

Thanks to all :)