fra589/grbl-Mega-5X #53: ENHANCEMENT, Can support for MPG wheel bee added to manually jog axis and set zero (g92) from controller instead of pc, And support for reading potentiometers for feed override ?

sinfocomp, 2019-05-21:


sinfocomp, 2019-05-23:
posted a bounty for $100 if someone is willing to implement this
https://www.canwork.io/jobs/public/enhancement-to-33
counter offers welcomed

fra589, 2019-05-23:
Hi @sinfocomp,
I don't think it will be a good id to ask to Grbl firmware to do à lot of thing like this you want.
Grbl must stay as light as possible to keep all its performance.
I think the functions you want must to be assigned to a CNC control panel. 
My CNC software control panel ([cn5X](https://github.com/fra589/cn5X)) is writen to be embeded on Rasbery PI or equivalent with a touch screen and to be dedicaced to one CNC milling machine. In the future, this type of function can certainly be implemented on [cn5X](https://github.com/fra589/cn5X).
@++;
Gauthier.
P.S. by the way, the link you posted above don't work...

closed 2019-05-23 by fra589