fra589/grbl-Mega-5X #54: (It s no a problem) 9 axis with aux 4

lukino87, 2019-05-27:
hi according to her it would be possible to modify the firmware to add 3 more linear axes using aux4 as you did for the sixth axis? the axes would not go in function all together but before 5 then 4.I saw more or less the files to be modified could route me in this? thanks in advance and congratulations for your work

fra589, 2019-05-27:
Hi @lukino87,  
Use config.h to do that :  
line 53-54
```
#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  #define N_AXIS 6            // Number of axes
  #define N_AXIS_LINEAR 6     // Number of linears axis
#else
```
lines 71, 75 & 79
```
  #define AXIS_4_NAME 'U' // Letter of axis number 4

  #define AXIS_5_NAME 'V' // Letter of axis number 5

  #define AXIS_6_NAME 'W' // Letter of axis number 6
```
Then, define your homing cycle (line 184 to 189)   

Then, it should be work like you whant.
I don't have tested this configuration, please, report here your experience :-)

@++;
Gauthier.

fra589, 2019-05-27:
Hi again @lukino87,  
OUPS ! I read your question too rapidly, my reply was for 6 linears axis :-(
For the moment, I think it's not possible to drive more than 8 axis (I don't known if driving 8 axis is possible, but in theory, yes) because each axis is driven by one bit of CPU register, and the CPU is only 8 bits...

To make Grbl up to 9 axis or more, you need to make a port to 16, 32 or 64 bits platform.
Another solution should be to use 2 Arduino Mega...

In reply to your original question, simply search for `#ifdef N_AXIS` in all .h and .c files then make complement for axis 7 to 8.  
@++;
Gauthier.

closed 2019-05-27 by fra589

lukino87, 2019-05-27:
thanks to the fast response actually 8 would be enough I will certainly try to get to 8 and share the experience ... I like using two arduino mega as a solution but I think it is difficult for my possibilities she would have an idea how to interact in that

Ottieni Outlook per Android<https://aka.ms/ghei36>

Closed #54<https://github.com/fra589/grbl-Mega-5X/issues/54>.

fra589, 2019-05-27:
8 axis is in my TODO list, but not in pole position :-)
@++;
Gauthier.

lukino87, 2019-05-27:
I go to work and I share my progress with her I will work on the 8 axes for a single board. and besides I thought about the solution with two cards using master and slave according to her is possible

Ottieni Outlook per Android<https://aka.ms/ghei36>