fra589/grbl-Mega-5X #55: 4 linear axis gantry machine timing is off.

rlwoodjr, 2019-06-02:
I was testing this version to do a self squaring gantry. I set the config file as shown below.

*******************************************************************************
  #define N_AXIS 4            // Number of axes
  #define N_AXIS_LINEAR 4     // Number of linears axis
#else`
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
#define AXIS_1_NAME 'X' // Axis names must be in X, Y, Z, A, B, C, U, V & W.
#define AXIS_2 1
#define AXIS_2_NAME 'Y'
#define AXIS_3 2
#define AXIS_3_NAME 'Z'

#if N_AXIS <3
  #error "N_AXIS must be >= 3. N_AXIS < 3 is not implemented."
#endif
#if N_AXIS > 3
  #define AXIS_4 3
  #define AXIS_4_NAME 'X' // Letter of axis number 4
******************************************************************************
If the AXIS_4_NAME is A, then it works fine, but if the AXIS_4_NAME X, the timing is off.

If  AXIS_4_NAME is A, then to move 30" takes 15 seconds at 120 inches per minute.
If  AXIS_4_NAME is X, then to move 30" takes 22 seconds at 120 inches per minute.

The stepper accelerations and resolution is the same for both Axes.

RaphaelDives, 2019-06-03:
Hello @rlwoodjr,
as far as I recall, there are more parameters to set for the axes. I would carefully check the $$ output and look for following numbers like $100-$104 and set the paraemters accordingly. (For example MAX Rate $110-$114...)

MAYBE this can help. ;)
Good luck!

Cheers Raphael

rlwoodjr, 2019-06-03:
Hello @RaphaelDives, thanks for the reply.

All of the velocity, acceleration resolution, travel parameters were set correctly. Once I discovered the issue I compiled the code only changing the A to X.   And then back again.

If AXIS_4_NAME is A, then to move 30" takes 15 seconds at 120 inches per minute.
If AXIS_4_NAME is X, then to move 30" takes 22 seconds at 120 inches per minute.

I even ran with the 2560 only (no stepper motors connected) and the timing is off if  AXIS_4_NAME is X,

I set UGS Platform to run 30" and 120 inches per minute.  and clicked the X jog button.

fra589, 2019-06-04:
Hi @rlwoodjr,  

Effectively, it seems to be a bug in motion speed control...
I need to search it... It may take a while, be patient :-)

Please, can you post your $$ config output? So I can reproduce with the same config as yours.

@++;
Gauthier.

rlwoodjr, 2019-06-04:
Hello @fra589 

Here are the settings:

**** Connected to COM22 @ 115200 baud ****
Grbl 1.1l ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.020    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 500.000    (Homing locate feed rate, mm/min)
$25 = 2000.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 71.111    (X-axis travel resolution, step/mm)
$101 = 71.111    (Y-axis travel resolution, step/mm)
$102 = 400.000    (Z-axis travel resolution, step/mm)
$103 = 71.111   
$104 = 8.889   
$110 = 20500.000    (X-axis maximum rate, mm/min)
$111 = 20500.000    (Y-axis maximum rate, mm/min)
$112 = 4000.000    (Z-axis maximum rate, mm/min)
$113 = 20500.000   
$114 = 1440.000   
$120 = 2300.000    (X-axis acceleration, mm/sec^2)
$121 = 2300.000    (Y-axis acceleration, mm/sec^2)
$122 = 1000.000    (Z-axis acceleration, mm/sec^2)
$123 = 2300.000   
$124 = 100.000   
$130 = 1200.000    (X-axis maximum travel, millimeters)
$131 = 1200.000    (Y-axis maximum travel, millimeters)
$132 = 125.000    (Z-axis maximum travel, millimeters)
$133 = 1200.000   
$134 = 180.000

fra589, 2019-06-05:
Hi @rlwoodjr,  
I pushed a fix for this issue (version 1.1l.20190605), can you test it and report to config resolution please ?
@++;
Gauthier.

rlwoodjr, 2019-06-05:
Hi @fra589 

I connected and did testing without stepper motors and it seemed to work.  I will try to connect stepper motors later and post an update. 

Thanks for the quick work.

rlwoodjr, 2019-06-06:
Hi @fra589 ,

The timing is good with 3 to 6 axes configurations  with stepper motors connected. Now to try to optimize for max step frequency. Again, thanks for the updated code.

closed 2019-06-06 by fra589