fra589/grbl-Mega-5X #56: (Not an issue)   I used the CPU_MAP to keep the Step/dir on the Same port

rlwoodjr, 2019-06-03:
I am not very good at programming but I did edit the CPU_MAP in hopes to increase the jitter-free pulse generation frequency.  The pin changes seem t work, but sadly it did not seem to help.  

Should any of the pins or ports be off limits? Or does it not help to have these on the same port?

Description  | D Pin  | Pin | Address
X1 step | D24 | 24 | A2
Y step | D25 | 25 | A3
Z step | D26 | 26 | A4
X2 Step | D27 | 27 | A5
A Step | D28 | 28 | A6
B Step | D29 | 29 | A7
X1   direction | D30 | 30 | C7
Y   direction | D31 | 31 | C6
Z   direction | D32 | 32 | C5
X2   direction | D33 | 33 | C4
A   Direction | D34 | 34 | C3
B   direction | D35 | 35 | C2
X1   enable | D44 | 44 | L5
Y   enable | D45 | 45 | L4
Z   enable | D46 | 46 | L3
X2   enable | D47 | 47 | L2
A   enable | D48 | 48 | L1
B   enable | D49 | 49 | L0
Spindle   enable | D4 | 4 | G5
Spindle   direction | D5 | 5 | E3
Spindle   PWM | D8 | 8 | H5
Coolant   mist | D9 | 9 | H6
Coolant   flood | D6 | 6 | B4
X1 Max | A0 | 54 | F0
X1 Min | A1 | 55 | F1
Y Max | A2 | 56 | F2
Y Min | A3 | 57 | F3
Z Max | A4 | 58 | F4
Z Min | A5 | 59 | F5
X2 Max | A6 | 60 | F6
X2 Min | A7 | 61 | F7
A Max | D10 | 10 | B4
A Min | D11 | 11 | B5
B Max | D12 | 12 | B6
B Min | D13 | 13 | B7
Probe | A15 | 69 | K7
Reset   switch (soft reset) | A9 | 63 | K1
Feed   hold | A10 | 64 | K2
Cycle   start | A11 | 65 | K3
Safety   door | A12 | 66 | K4

rlwoodjr, 2019-06-03:
Here is my add to the CPU_MAP file.

#ifdef CPU_MAP_2560_BOBSCNC // (Arduino Mega 2560) FOR KL7 SERIES  BobsCNC
#include "nuts_bolts.h"

// Serial port interrupt vectors
#define SERIAL_RX USART0_RX_vect
#define SERIAL_UDRE USART0_UDRE_vect

// Define ports and pins
#define DDR(port) DDR##port
#define _DDR(port) DDR(port)
#define PORT(port) PORT##port
#define _PORT(port) PORT(port)
#define PIN(pin) PIN##pin
#define _PIN(pin) PIN(pin)

// Define step pulse output pins.

#define STEP_PORT_0 A
#define STEP_PORT_1 A
#define STEP_PORT_2 A
#if N_AXIS > 3
#define STEP_PORT_3 A // Axis number 4 
#endif
#if N_AXIS > 4
#define STEP_PORT_4 A // Axis number 5 
#endif
#if N_AXIS > 5
#define STEP_PORT_5 A // Axis number 6 
#endif
#define STEP_BIT_0 2  // X Step - Pin D24
#define STEP_BIT_1 3  // Y Step - Pin D25
#define STEP_BIT_2 4  // Z Step - Pin D26
#if N_AXIS > 3
#define STEP_BIT_3 5 // Axis number 4 Step - Pin D27
#endif
#if N_AXIS > 4
#define STEP_BIT_4 6 // Axis number 5 Step - Pin D28
#endif
#if N_AXIS > 5
#define STEP_BIT_5 7 // Axis number 6 Step - Pin D29
#endif
#define _STEP_BIT(i) STEP_BIT_##i
#define STEP_BIT(i) _STEP_BIT(i)
#define STEP_DDR(i) _DDR(STEP_PORT_##i)
#define _STEP_PORT(i) _PORT(STEP_PORT_##i)
#define STEP_PORT(i) _STEP_PORT(i)
#define STEP_PIN(i) _PIN(STEP_PORT_##i)

// Define step direction output pins.
#define DIRECTION_PORT_0 C
#define DIRECTION_PORT_1 C
#define DIRECTION_PORT_2 C
#if N_AXIS > 3
#define DIRECTION_PORT_3 C // Axis number 4 
#endif
#if N_AXIS > 4
#define DIRECTION_PORT_4 C // Axis number 5 
#endif
#if N_AXIS > 5
#define DIRECTION_PORT_5 C // Axis number 6 
#endif
#define DIRECTION_BIT_0 7 // X Dir - Pin D30
#define DIRECTION_BIT_1 6 // Y Dir - Pin D31
#define DIRECTION_BIT_2 5 // Z Dir - Pin D32
#if N_AXIS > 3
#define DIRECTION_BIT_3 4 // Axis number 4 Step - Pin D33
#endif
#if N_AXIS > 4
[... 148 lines cut ...]
#define SPINDLE_PWM_MAX_VALUE     1024.0 // Translates to about 1.9 kHz PWM frequency at 1/8 prescaler
#ifndef SPINDLE_PWM_MIN_VALUE
#define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
#endif
#define SPINDLE_PWM_OFF_VALUE     0
#define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)

//Control Digital Pin 6 which is Servo 2 signal pin on Ramps 1.4 board
#define SPINDLE_TCCRA_REGISTER    TCCR4A
#define SPINDLE_TCCRB_REGISTER    TCCR4B
#define SPINDLE_OCR_REGISTER      OCR4C
#define SPINDLE_COMB_BIT          COM4C1

// 1/8 Prescaler, 16-bit Fast PWM mode
#define SPINDLE_TCCRA_INIT_MASK ((1<<WGM40) | (1<<WGM41))
#define SPINDLE_TCCRB_INIT_MASK ((1<<WGM42) | (1<<WGM43) | (1<<CS41))
#define SPINDLE_OCRA_REGISTER   OCR4A // 16-bit Fast PWM mode requires top reset value stored here.
#define SPINDLE_OCRA_TOP_VALUE  0x0400 // PWM counter reset value. Should be the same as PWM_MAX_VALUE in hex.

// Define spindle output pins.
#define SPINDLE_PWM_DDR   DDRH
#define SPINDLE_PWM_PORT  PORTH
#define SPINDLE_PWM_BIT   5 // MEGA2560 Digital Pin D8

#endif

RaphaelDives, 2019-06-03:
Hey @rlwoodjr,
I don't really understand what you're up to, but I don't think that changing the port makes any difference, unless you change the code itself…
Cheers Raphael

Please use the "Insert code" button ("<>") for inserting code. 😎

rlwoodjr, 2019-06-03:
My thought is if all the step pins were on the same port, then it would save a clock cycle or two. each step signal.  In the CPU_MAP.h it states:

`// Define step pulse output pins. NOTE: All step bit pins must be on the same port.`

fra589, 2019-06-04:
Hi @rlwoodjr,  
  
> // Define step pulse output pins. NOTE: All step bit pins must be on the same port.  

The comment you talk about is for non RAMP mapping. This constraint was from original grbl-Mega before the RAMP version. The RAMP version add new pin definition method wich freed this constraint. It just affect the pin definition, the step pulse are sended by the same interrupt handler routine, independently of the pin definition.
You can found more info about the max step frequency of GRBL here : (https://github.com/grbl/grbl/issues/41)  
  
@++;  
Gauthier.

rlwoodjr, 2019-06-04:
Hello @fra589 

It might be interesting to note that the grbl1.1e on the Uno will run 36kHz max step frequency. I can run the Mega code at 23 kHz max step frequency.  Not sure if it is the added axes or if it is something I am doing wrong.

Here is a video link of the CNC moving 1200 inches per minute. [https://www.youtube.com/watch?v=0lUSA1DxqGQ](url)

fra589, 2019-06-05:
Hi @rlwoodjr, 
See also in the stepper.c file, the max step frequency ans speed limitation are explained in comments...
Sure, the added axis also add computing cycles CPU, but I can't explain myself the big difference you have.
Perhaps you can search in the stepper.c file where to optimize the pulse frequency?
@++;
Gauthier.

rlwoodjr, 2019-06-06:
Hi @fra589  Thanks for the information.

closed 2019-06-06 by rlwoodjr