fra589/grbl-Mega-5X #61: Reprap discount smart controller lcd compatiblity

coilheaddesigns, 2019-07-02:
Good day.
Is there a way to get lcd compatability for the lcd display from reprap.
Perhaps just basic jog and home and set zero.
With grbl position on the main display.

fra589, 2019-07-03:
Hi @coilheaddesigns,  
it's not in the pipes to implement an LCD display on grbl-Mega-5X as I think it may slow it down too much.  
My [cn5X project](https://github.com/fra589/cn5X) is attended to run on Rasberry PI with touch screen and so, to be a complete solution to drive 3 to 6 axis CNC machine with all it's functionalities with a medium size screen showing all relevant information in real time, and touch buttons and menus to jog and control al the grbl-Mega-5X functionalities.

@++;
Gauthier.

coilheaddesigns, 2019-07-03:
OK cool no problem than you for the response

closed 2019-07-03 by coilheaddesigns

coilheaddesigns, 2019-07-05:
Would it be possible to copy the code from issue #77 to you version or will it be to difficult for someone with beginner coding skills.

I have to start getting into c++ for this hobby of mine😂

fra589, 2019-07-05:
Hi @coilheaddesigns,  
I think it's not really within the reach of a beginner.  
I have a good programming experience: I worked on the development of CAD software and post-processors since 1993 (I would be retired at the end of this month), and it took me several weeks / months to do the reverse engineering of the grbl original, in order to be able to evolve it to what it became in the 5X version.  
But if you have courage, it will be a great adventure for you to do it.  
@++;
Gauthier.

coilheaddesigns, 2019-07-05:
Than you for you advice it is much appreciated