fra589/grbl-Mega-5X #62: Clone axis

coilheaddesigns, 2019-07-02:
Hey

Where do I clone axis.
Need to clone Y axis

fra589, 2019-07-03:
Hi @coilheaddesigns,  

To have a 3 axis with the Y axis cloned :  
Edit the config.h file, 
On the line 53, use :  
` #define N_AXIS 5 // Number of axes`
and on the line number 71, type :  
` #define AXIS_4_NAME 'Y' // Letter of axis number 4`
  
Then, the axis Y will be cloned on the E0 axis on RAMPS

@++;
Gauthier.

coilheaddesigns, 2019-07-03:
OK cool

Thank you very much.

The how do I enable the other axis for my rotary setup?

fra589, 2019-07-03:
In config.h, define the number of axis you need on line 53 (N_AXIS) (max 6 axis), then give the name of each axis you need. If you give the same name to more than one axis, this named axis will be cloned.

For example :
#define N_AXIS 5 // Number of axes
```
(...)
#define AXIS_1_NAME 'X'
(...)
#define AXIS_2_NAME 'Y'
(...)
#define AXIS_3_NAME 'Z'
(...)
#define AXIS_4_NAME 'Y'
(...)
#define AXIS_5_NAME 'A'
```
  
Then, you will have 5 reals axis with 4 logical :  X, Y cloned on axis number 2(Y) and 4(E0), Z and the rotational A.
  
@++;  
Gauthier.

coilheaddesigns, 2019-07-03:
Brilliant thank you for the response will setup tonight when I get home.

coilheaddesigns, 2019-07-03:
When I do it like that the Y motor on E0 engages but does not move

fra589, 2019-07-04:
First, verify parameters of your axis : step/unit, max speed and acceleration...

coilheaddesigns, 2019-07-04:
[Config setup](https://imgur.com/a/0E12Hxh)

coilheaddesigns, 2019-07-04:
Still only engages y no movement

fra589, 2019-07-04:
See the wiki page of grbl configuration (when grbl is running) here :  
https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration
And post the output of the "$$" grbl command...

coilheaddesigns, 2019-07-04:
Got it running

coilheaddesigns, 2019-07-04:
Homing cloned axis not homing

coilheaddesigns, 2019-07-04:
The cloned axis moves in the Same direction with jog but on homing it does not move

coilheaddesigns, 2019-07-05:
I have confirmed all cloned axis $values have been replicated

fra589, 2019-07-05:
Please, take a look at previous issues and their replies before asking...
https://github.com/fra589/grbl-Mega-5X/issues/51
https://github.com/fra589/grbl-Mega-5X/issues/48
https://github.com/fra589/grbl-Mega-5X/issues/4
@++;
Gauthier.

closed 2019-07-05 by coilheaddesigns

cybertza, 2019-12-26:
$23=11
Seems to have fixed this for me:
`
00001111
'

'
0000AZYX
'

'
00008421
`