fra589/grbl-Mega-5X #65: Mega5X not compiling

JeffHank1, 2019-08-06:
I am trying to load Mega5X grbl 1.1l onto an Arduino Mega 2560 using Arduino IDE 1.8.8 without success. I had not modified the file in any way. The initial errors had to do with the Compile Time comments at the end of the Config.h file with complaints of undefined variables. Deleting those comments allowed it to progress a little further but now the compiler is reporting:  ‘N_Axis was not declared in this scope’ followed by a succession of N_Axis related variable complaints. 
I realize this description is a quite sketchy but rather than copy and paste pages of error reporting, I wonder if I have to fill in some parameters in the various .h files before I try to compile Mega5X grbl. Thanks for any suggestions

fra589, 2019-08-06:
Hi @JeffHank1,  
Have you seen the grbl's compilation procedure via the Arduino IDE from the original gnea/grbl project ?
cf. : https://github.com/gnea/grbl/wiki/Compiling-Grbl
@++;
Gauthier.

JeffHank1, 2019-08-06:
Yes. Thanks for the reply. I followed that exactly. I also made sure that
the Arduino file contained the Grbl library from the mega 5X edge download
and no others. That’s what is puzzling me. I managed to get it to compile
and upload before and now the procedure stops at compile error 1. The
source of the errors is the Compile Time comments in config.h. It seems to
me that all the variables are defined in the other files.

The other Mega 2560 board is being used successfully with Grblgru FYI.
Thanks for making this version of Grbl

Jeff

JeffHank1, 2019-08-07:
If I try to compile Mega5X 'as is' making sure to have the Arduino Sketch file on my root drive and deleting any other Sketch files - particularly on Windows OneDrive I get these error messages:
Arduino: 1.8.8 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
(After the verbose listing of files processed comes these  errors. No changes to the original GRBL  Mega 5X file Ver 1.1l)
In file included from C:\Documents\Arduino\libraries\grbl/config.h:32:0,
                 from C:\Documents\grbl-Mega-5X-edge\grbl\examples\grblUpload\grblUpload.ino:1:
C:\Documents\Arduino\libraries\grbl/grbl.h:70:4: error: #error "Required HOMING_CYCLE_0 not defined."
   #error "Required HOMING_CYCLE_0 not defined."
C:\Documents\Arduino\libraries\grbl/grbl.h:98:4: error: #error "WCO refresh must be greater than one."
   #error "WCO refresh must be greater than one."
C:\Documents\Arduino\libraries\grbl/grbl.h:101:4: error: #error "Override refresh must be greater than zero."
   #error "Override refresh must be greater than zero."
Using library grbl in folder: C:\Documents\Arduino\libraries\grbl (legacy)
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

It seems the program is looking for defined values but the ones mentioned seem to already be defined.

fra589, 2019-08-07:
Hi Jeff,
I use Linux to develop and compile grbl-Mega-5X without the Arduino IDE... So, It's possible I made a mistake for the Arduino compile procedure. I will test it today or tomorrow (I need to put a Windows box in configuration to test it, it may take a while)
@++;
Gauthier.

RaphaelDives, 2019-08-07:
Is there a new grbl-mega?! :D
I'm compiling with Arduino and Win10... I will test the actual Version this evening and report back!
Cheers, Raphael

JeffHank1, 2019-08-07:
Thank you. Even posting a hex file for 1.1l would be great. The latest one I could find is 1.1h and it looks like you’ve fixed some bugs since. Linux, Python and the CN5X app all make sense now ; )

PulseTrainHat, 2019-08-07:
I compiled on a new Windows 10 machine yesterday with no issues.

Downloaded the files, copied just the GRBL folder into the C:\Users\sean\Documents\Arduino\libraries\ folder

Installed Arduino IDE from Windows Store.

loaded the C:\Users\sean\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino into the IDE, made sure that GRBL was ticked under Sketch>Include Library and Compiled successfully.

The only time I have seen similar problems and errors you are seeing is if I have more than one type of GRBL in the libraries folder. So I just make sure I have the one version of GRBL  that I am compiling in that folder,

Cheers

Sean

JeffHank1, 2019-08-07:
Thanks Sean. I tried that Arduino version as well just now. I got rid of any and all Arduino libraries and added the GRBL Mega5X GRBL file as a library to the Arduino IDE and got the same result as above. Weird. Probably Windows 10 so I'll try on a Mac.

JeffHank1, 2019-08-07:
So I just tried GRBL Mega 5X with a Macbook Pro and a older copy of Arduino IDE 1.8.5 and it all compiled. I've uploaded it to my Mega 2560 and will sail onwards to multiple axis success!

fra589, 2019-08-08:
Hi Jeff,  
  
I just compile grbl-Mega-5X on fresh install of Arduino 1.8.8 without problem.  
I seems that your Win10/Arduino install is not the standard one.  
The standard location of Arduino's libraries is in the folder :  
`<USER_HOME_FOLDER>\My Documents\Arduino\libraries`  
The error message you posted tell about `C:\Documents\Arduino\libraries`...  
Perhaps is there a mystake in the Arduino's libraries search path, perhaps Arduino IDE found more than one version of grbl in your system...  
  
Please, an you post the integral detailed output of the compilation ?  
  
@++;  
Gauthier.

JeffHank1, 2019-08-08:
Thank you for your kind offer. I think Windows is my problem to solve. Your efforts are better invested in the great software you make : )
Thanks again for Grbl Mega 5X. It’s working well with Grblgru.

closed 2019-08-08 by JeffHank1