fra589/grbl-Mega-5X #67: Self-squarring feature

akzmonster, 2019-08-29:
Hello Gauthier, thanks so much for responding to me. I have been going through the wiki looking for any documentation on it, but I can't seem to find it. I'm assuming that motor A and B would need to be cloned. Thanks.

fra589, 2019-08-29:
Hi @akzmonster,  
  
The cloning principle is quite simple and can be done in the config.h file:  
To clone both X and Y axis, leave the N_AXIS variable to 5, then use :  
```
#define AXIS_4_NAME 'X' // This clone the X axis on the 4th axis (E0 on RAMPS)
#define AXIS_5_NAME 'Y' // This clone the Y axis on the 5th axis (E1 on RAMPS)
```
  
For auto-squarring, this an be done at homing with a config like this:  
```
#define HOMING_CYCLE_0 (1<<AXIS_3) // Home Z axis first to clear workspace.
#define HOMING_CYCLE_1 ((1<<AXIS_1)|(1<<AXIS_4)) // Home X axis
#define HOMING_CYCLE_2 ((1<<AXIS_2)|(1<<AXIS_5)) // Home Y axis
```
  
You can see more info in old closed issues...  
But you are right, writing detailed infos about this in the wiki is necesary.  

@++;  
Gauthier.

akzmonster, 2019-08-29:
Perfect, thanks so much!

closed 2019-09-09 by fra589