fra589/grbl-Mega-5X #69: ALARM 9  on Z-homing

Tiktiki, 2019-09-20:
I have been scratching my head for over a week now on this issue.

When ever I try to home the Z-axis, if the limit switch is not triggered within 7mm of travel, I get ALARM 9  triggered.

Every axis moves in the right direction (and I tried to reverse everything possible including the axis direction and homing direction, physically inverting the stepper plug, ect.).

However, if the Z-axis is within the 7mm distance from the carriage position, it triggers, bounces and homes all the axis properly. Soft limits work as they should and yes I have set the max travel properly (in my case I have 107mm of travel in Z).

Once I get it to home properly, everything works as intended past that point, and sending a G28 from any position goes back to the proper zero on all axis.

fra589, 2019-09-21:
Hi @Tiktiki,  
  
Strange behavior... ??? Never happened for me?!?!?
Can you post $$ output and config.h axis and homing definition?  

@++;  
Gauthier.

Tiktiki, 2019-09-24:
Here are the config.h file and settings.

$0=10
$1=255
$2=0
$3=3
$4=0
$5=0
$6=0
$10=3
$11=0.020
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=0
$24=200.000
$25=2000.000
$26=200
$27=2.000
$30=24500
$31=0
$32=0
$100=100.000
$101=100.000
$102=1600.000
$103=100.000
$104=100.000
$110=4500.000
$111=4500.000
$112=800.000
$113=4500.000
$114=4500.000
$120=50.000
$121=50.00
$122=50.000
$123=50.000
$124=50.000
$130=1152.000
$131=612.000
$132=107.000
$133=1152.000
$134=612.000

// Define CPU pin map and default settings.
// NOTE: OEMs can avoid the need to maintain/update the defaults.h and cpu_map.h files and use only
// one configuration file by placing their specific defaults and pin map at the bottom of this file.
// If doing so, simply comment out these two defines and see instructions below.
//#define DEFAULTS_GENERIC
//#define CPU_MAP_2560_INITIAL

// To use with RAMPS 1.4 Board, comment out the above defines and uncomment the next two defines
#define DEFAULTS_RAMPS_BOARD
#define CPU_MAP_2560_RAMPS_BOARD

// Serial baud rate
// #define BAUD_RATE 230400
#define BAUD_RATE 115200

// Axis array index values. Must start with 0 and be continuous.
#ifdef DEFAULTS_RAMPS_BOARD
  // 4, 5 & 6 axis support only for RAMPS 1.4 (for the moment :-)...)
  #define N_AXIS 5            // Number of axes
  #define N_AXIS_LINEAR 3     // Number of linears axis
#else
  #define N_AXIS 3 // Number of axes = 3 if not DEFAULTS_RAMPS_BOARD
#endif

#define AXIS_1 0        // Axis indexing value. Must start with 0 and be continuous.
[... 111 lines cut ...]
  #define HOMING_CYCLE_1 ((1<<AXIS_1)|(1<<AXIS_2))  // OPTIONAL: Then move X,Y at the same time.
  // #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
#endif // DEFAULTS_RAMPS_BOARD

// NOTE: The following are two examples to setup homing for 2-axis machines.
// #define HOMING_CYCLE_0 ((1<<AXIS_1)|(1<<AXIS_2))  // NOT COMPATIBLE WITH COREXY: Homes both X-Y in one cycle.

// #define HOMING_CYCLE_0 (1<<AXIS_1)  // COREXY COMPATIBLE: First home X
// #define HOMING_CYCLE_1 (1<<AXIS_2)  // COREXY COMPATIBLE: Then home Y

// Number of homing cycles performed after when the machine initially jogs to limit switches.
// This help in preventing overshoot and should improve repeatability. This value should be one or
// greater.
#define N_HOMING_LOCATE_CYCLE 1 // Integer (1-128)

// Enables single axis homing commands. $HX, $HY, and $HZ for X, Y, and Z-axis homing. The full homing
// cycle is still invoked by the $H command. This is disabled by default. It's here only to address
// users that need to switch between a two-axis and three-axis machine. This is actually very rare.
// If you have a two-axis machine, DON'T USE THIS. Instead, just alter the homing cycle for two-axes.
// #define HOMING_SINGLE_AXIS_COMMANDS // Default disabled. Uncomment to enable.

// After homing, Grbl will set by default the entire machine space into negative space, as is typical
// for professional CNC machines, regardless of where the limit switches are located. Uncomment this
// define to force Grbl to always set the machine origin at the homed location despite switch orientation.
// #define HOMING_FORCE_SET_ORIGIN // Uncomment to enable.

fra589, 2019-09-25:
Hi @Tiktiki,  
  
More and more weird?!?!?!  
  
I understand your CNC is a 3axis with 2 motors on X, 2 motors on Y and 1 motor on Z... So, this config should be work fine.  
Have you previously tested another Grbl version or another axis number configuration on your board?  
Sometime, the EEPROM is corrupted by a previous old firmware or another firmware configuration (differences of config.h, cpu_map.h, default.h, ...)  
So, perhaps you need to clears and restores all of the EEPROM data used by Grbl by issuing the $RST=* Grbl command...  
It's already happened to me this kind of problem.  
Keep me informed.  
  
@++;  
Gauthier.

Tiktiki, 2019-09-25:
I tried clearing the EEPROM, in fact I tried 3 different MEGA. Never got GRBL to work on this, it was running Marlin before and I hated it...

fra589, 2019-09-25:
My own CNC machin also use TR8*2 screw for all axis... 
I had to reduce licrostepping to 1/4 to resolve frequency problems with a max speed of about 800 mm/mn.
This leaves 2.5 micron accuracy. Which is more than enough for my own use.

Tiktiki, 2019-09-25:
I have lowered my setup to full-step on the Z-axis to test and it fixed the issue.
This means that the maximum mechanical resolution would be in the 4-8mm per turn lead screw or belt setup. This is really weird considering the exact same setup works perfectly in Marlin (even more precise actually, it was working fine in 1/32 micro-stepping), so it should not be a limitation of the 8-bit hardware, but something in the code that is weird.

It might be related to the way the watchdog counts the cycles or distance traveled by the machine. I will re-install the 1/16 micro-stepping setup and see if there is a relation between the actual traveled distance when I get the error and the limitations of the 8-bit architecture. (I suspect there is a stack overflow situation because the setup requires too many step pulses, but did not dig into that section of the code yet).

fra589, 2019-09-26:
Hi @Tiktiki  

> I suspect there is a stack overflow situation because the setup requires too many step pulses  

I'm rather thinking of a problem of exceeding the maximum call frequency of interrupts timers ... With a large number of steps per unit and a high speed, the previous interruption may not be completed when the next one is called. The ATMega processor architecture does not allow it.  
  
Given your homing speed settings ($24 & $25) with 1/32 microstepping and TR8*2, you need :
```
$24=200.000
$25=2000.000
```
```
f = (MICROSTEPS_XYxSTEP_REVS_XY/MM_PER_REV_XY)x(UNIT_PER_SECOND)
f = (32x200/2)x(2000/60) = 106666,6667 Hz 
```
More than 100kHz for Homing seek, the frenquency is really too high !!!
Given the acceleration limitation ($122=50 for your Z axis), the speed is not maximum at the beginning of the movement, which could explain that the axis can move up to about 7mm before the speed becomes too important.
With full steps, the frequency decreases to about 3 kHz... ATMega can do better :-)
  
Maybe you should go back to a better microstepping (1/4 or 1/8) and decrease the maximum speed.
  
On my machine, with same TR8*2 single entry screw as yours, I optimized with these settings:
```
Microstepping = 1/4
$24=25 (Homing feed, mm/min)
$25=500 (Homing seek, mm/min)
$102=400 (Z steps/mm)
$112=750 (Z Max rate, mm/min)
$122=50 (Z Acceleration, mm/sec^2)
```  
  

@++;
Gauthier.

closed 2019-11-19 by fra589