fra589/grbl-Mega-5X #70: Feature request: Maslow style motor controls

akzmonster, 2019-10-16:
Hey Gauthier,

First of all, allow me to butter your biscuit. You have been very helpful and knowledgeable and I greatly admire and envy your skill. GRBL is so huge and complex and I would have loved very much to be a part of it.

Are you familiar with a Maslow CNC? It's similar to a CoreXY but uses the shared hypotenuse of two motors from which the spindle hangs. The images are far more effective. Well, I've been noodling around the internet trying to find a way to make something work as simple as GRBL can be to use. I would like to build a Maslow-style (polargraph-style) CNC using GRBL Mega and Ramps 1.4. This would allow full sheet CNC capabilities with minimal floor space and minimal motors all in the GRBL family. I have checked out "GRBL Mega Wall-Plotter" here on Github but there is little progress there and I am learning GRBL very slow. I am a beginner programmer and would love to contribute but I seem to be spinning my wheels. I figured that one will never get what they don't ask for so I thought that I could at least try.

So, is it possible to add that functionality to grbl-Mega-5X? Even if not but you tell me how that functionality could be added I could try and implement it myself using the existing code as a start. I know this is probably asking a lot but I can't seem to find any project that works:
- I have found some that use G-codes but also requires servos (light duty pen plotters)
- I have found some that use steppers but don't use standard G-codes
- I have found some that use steppers and/or g-codes but aren't made for mega/ramps
- The rest are not open source or are proprietary :(

bdurbrow, 2019-10-24:
Do a search across the source files for COREXY. The coordinate frame translations are implemented at those locations; so it would be just a matter of creating a compile-time switch (for example, #define MASLOW) and adding the requisite transforms in the same spots as for COREXY.

There are a few concerns that I have with this though - not insurmountable issues, just things to watch out for:

1) You will probably not be able to get reasonable performance out of motors that can be driven by by Stepstick-sized drivers. I would expect to have to use outboard drivers for this - so unless you are using my fork of Grbl which supports 3D-printer-style user interface hardware; there's no need for the actual RAMPS board itself. Instead you can either connect directly to the Arduino Mega with 'DuPont' style 0.100"-pin wires; or use a screw-terminal shield (mechanically this will be the more reliable of the two, of course) in order to route the step and direction signals to the stepper drivers. Also, you won't want to run the stepper drivers on 12v - that will limit the top speed of them, and make rapids painfully slow.

2) Pretty much by definition, a Maslow CNC has no rigidity. I might seriously think about adding a third control line to it to fully constrain the position of the head in the X-Y plane.

3) Depending on the kind of cuts you intend to make, it may be worthwhile to build a vacuum table for it to keep cut-out parts from separating from the table and jamming the head. Attaching some sort of high-friction material (perhaps paint it with flooring-rated paint with coarse sand mixed in?) would probably be a good idea also; just be sure that the result is reasonably rigid (so the foam rubber drawer liners probably isn't the best choice).

P.s. - it's not ready yet - as I still have testing to do - but I do have a breakout board that is designed to support outboard motor drivers; has IO for the same user-interface hardware that works with my version of Grbl-mega; and has pinouts largely compatible with a RAMPS card (the LCD pin mapping is a little different than a RAMPS+RepRapDiscount Smart Controller stack uses). I intend to release gerbers for it as soon as I've confirmed that it works right.

😄

fra589, 2019-10-26:
Hi,  
Thanks @bdurbrow for the reply, This is the good method to implement the good @akzmonster idea.  
Unfortunately, I have not enough time to do it by myself, but perhaps I will can help by responding if you have precises questions on grbl-Mega-5X internal functioning.  
See you on this feature request for the continuation of this subject :smile: 
@++;  
Gauthier.