fra589/grbl-Mega-5X #71: G0 should find the shortest path on Rotary Axis (wrap-around coordinate system ?)

henpemaz, 2019-10-18:
Consider I'm on a XYZA machine with a rotary axis (A) that can spin without any limits on both directions. After doing a "spiral" operation, this A axis might be on some crazy high work coordinates like 36000 degrees (100 turns of that axis). Now we need to get it back to 0 somehow.

On this situation, a "G0 A0" command will spin the A axis 100 times to get it back to it's origin, which is "correct" but sometimes just a waste of time. A different approach would be to reset that axis using "G92 A0" if we're in fact at the origin (a multiple of 360 degrees), but if we're not, we'd need to do the math on how much we need to move before resetting that coordinate, or what's our "equivalent" current position and G92 to that.

Shouldn't GRBL in this situation (G0 on a rotary axis) be able to find out the shortest movement it can do to get that axis to an arbitrary position that's "equivalent" to the commanded position ? This might not we suitable for every machine, some have wires or other limitations, but for some others, it would make things a lot simpler.

I'm new to 4-axis and I'm writing my Gcode for testing by hand for now, so I might be missing on some clever tricks that might solve the problem. If so, please share them. BUT if that's not the case, then I think it should be a quite important option/feature for a 4-axis machine controller.

PS: Super merci pour ce firmware magnifique, ça a acceleré beaucoup mon project :)

jflapao, 2019-10-18:
Hi, I work with a industrial CNC from the brand Pedrini, and the machine have a rotary axis called B should be C because is over the Z axis (is a table) but Pedrini called B, and have a indunction cell to reset the encoder of the B axis, we can perform the many degrees we want the we the the zero procedure to the axis, when she find the induction cell she stop and reset the axis,  think that should be one way to you reset that A axis. You can put a induntion cell, because you have no limits you turn the cell of, and when you need the zero procedure you activit and do it .

bdurbrow, 2019-10-24:
Hmm... not sure what the best implementation would be... either a compile-time switch to make G0 always behave that way; or adding a G-code to do it (G0.1???).

fra589, 2019-10-26:
hi @henpemaz,
Thank you for raising this behavior... Yes, I will add this topic in the todo list.  
Actually, the only solution is to calculate the number N of revolutions and to use G92 to offset the angle coordinate by N*360.
@jflapao, since we don't have encoder, I think your proposal is not in the good direction.
@bdurbrow, I think Both solutions are usables : compile-time(s) switch(s) to modify G0 (and perhaps G1) behavior, or add G0.1 / G1.1 to move to nearest position modulo 360.
@++;  
Gauthier.

Flying-Hammer, 2020-04-03:
Hem.... Why not just use a $H A ?

@fra589  ???

fra589, 2020-04-04:
Bonjour [Jean-Claude](@Flying-Hammer),

Ben parce-que $H effectue la prise d'origine (homing), c'est à dire un double déplacement jusqu'à activer les commutateurs de butée des axes.
Et là, on ne veux pas se déplacer, on veut simplement pouvoir corriger la coordonnée courante d'un axe rotatif (la mémoire interne qui stocke la position de l'axe) modulo 360.

@++;
Gauthier.

Flying-Hammer, 2020-04-04:
Enfin bon je lis quand-même

>  ""G0 A0" command will spin the A axis 100 times to get it back to it's origin, which is "correct" but sometimes just a waste of time"

S'il a fait 100 tours, je comprends bien qu'il n'ait pas envie de les faire dans l'autre sens juste pour revenir à zéro. S'il a (comme moi) un endstop sur son axe, faire un homing remet l'église au centre du village en moins d'un tour de diviseur.... Et comme il travaille par rapport à un zéro pièce (qui n'est d'ailleurs pas forcément confondu avec le zéro machine du diviseur) il peut repartir facilement sans avoir à attendre 100 tours pour rien.
Enfin bon moi ce que j'en dis hein....
Dans le fraisage j'y connais que dalle !

fra589, 2020-04-07:
Oui, mais `G0A0` n'est qu'un exemple parmis tant d'autres...
Un autre exemple pourrait être `G1A2` si on a besoin d'usiner jusqu'à un angle de 2...
Les besoins sont infinis, et $HA ne répond qu'au seul cas de figure d'un retour à l'origine, et encore, a condition d'avoir un capteur d'origine sur l'axe rotatif A, ce qui est bien rare, surtout avec un axe non limité. Il faudrait en plus que la précision du dit capteur soit au moins aussi bonne que la précision de positionnement des moteur pas à pas, ce qui est loint d'être le cas dans la majorité des machines DIY...
En plus, il faut avoir activer l'option `#define HOMING_SINGLE_AXIS_COMMANDS`...
Effectivement, si toutes ces conditions sont réunies et le besoin seulement le retour à l'origine, alors la commande $HA fera le boulo.
Mais pour tout le reste, il faudra coder de nouvelles fonctionalités.
@++;
Gauthier.

Flying-Hammer, 2020-04-07:
J'ai longtemps réfléchi à "comment faire un endstop pour le diviseur" sachant qu'avec une came ça va pas du tout, c'est pas précis. J'ai donc monté sur l'axe de rotation un disque avec une fine-fine fente radiale. Le disque tourne dans l'entrefer d'un capteur optique à fourche, la répétabilité est diabolique.
Nota : tout ce que je dis au-dessus ne concerne que MON application, j'ai bien sûr défini le single truc, ça marche mais je ne prétends pas que c'est l'Evangile.
@++
L'JCG

henpemaz, 2020-04-09:
Salut!

Jusque là mon axe A n'a pas de commutateur et le homming n'est pas confinguré. C'est un mandrin à quatre mors qui peut tourner librement sans impediments, donc j'avais choisi de ne pas mettre un switch, en plus parce que je n'ai pas les modeles avec roulette. En fin je croyais que je n'avait pas besoin. 

J'utilize par fois ce axe pour usiner des filetages avec un v-bit 60º (je n'ai pas un tour 😢), et la passe finale je la répète avec des offsets pour arriver a un bon ajustement. Le problème est que chacun de ces passes veut démarrer a A0, et le passe précédent n'a pas fini sur un multiple de 360.

Cette fonctionnalité ("Wrapped Rotary Axis") a été demandée quelque fois chez LinuxCNC avec des diferéntes specifications mas je crois qu'elle n'est pas implementée, bien que elle existe dejá dans quelques autres controlateurs commercielels.

Pour le moment j'accepte que chaque passe prends 30 secondes en plus, c'est plus facile que faire des ajustements avec G92.

jflapao, 2020-04-10:
@fra589 Sorry I'm away for long time
"@jflapao, since we don't have encoder, I think your proposal is not in the good direction."
But when you do the homing procedure to find the "zeros" of all the axis no mather if is X,Y,Z,A,B,C you need a hard limit right can be a switch or a cell induntion so on the C axis of the Pedrini like a say she turn, and turn, and turn to do the job, when she to the job we have 2 option:
One: We see how much degrees she move and give the order to go back this degrees, or 
Two: We to the homing of the C axis (on Pedrini is the B axis) and she move in positive motion until find the induction cell that's is your hard limit swich stop and at the same time reset the axis coordinates to "zero".

So is exact the same way you do when your GRBL machine stop in the middle of the work area, you turn off the power when back on the GRBL don't know here is it because YOU DON'T HAVE ANY ENCODER you need to do the homing procedure to all the axis.

I think is a good ideia to do the job, and the front end of the GRBL, your cn5X for example should be have some buttons to do this kind of procedure, in one area of definitions or other place have one ares to do the "zeros" of the all axis - one of the time start with Z, then X, them Y, then C, then A and/or B in my experience work with industrial CNC machines with encoders, and servo motors this is the best way. Pedrini have encoders and form times to times we need to do the zeros of the axis, and this is the sequence that pedrini tell us to do, the GMM have servo motors we don't need to do the zeros no mather if you turn the power down move the machine by hand and back on the machine always know the position , we only need to do the zeros if we have change one drive because was broken or if we switch the spindle because burn or something like that.

For me the best way to reset a rotatory axis without limits, is have one induction cell on the zero position and when needed do the zero procedure to that axis to reset it

jflapao, 2020-04-10:
@Flying-Hammer , @henpemaz , @fra589 I know this is a forum of a french guy, but I think the best way to communicate is english, I'm portuguese and I don't write here in portuguese right!!!!!

fra589, 2020-04-10:
Bonjour,

_(Traduction exceptionnelle de courtoisie en Anglais ci-dessous...)  
(Exceptional courtesy translation to English below ...)_
  
  
Vu qu'ici, nous sommes, sur les pages concernant les problèmes du projet dont JE suis le développeur, je suis Français et je préfère utiliser le Français quand c'est possible, notamment lorsque mes interlocuteurs sont eux même Français.

Etant donné que c'est ma langue maternelle, je suis beaucoup plus précis et performant dans cette langue que dans n'importe quelle autre.
J'accepte d'utiliser l'anglais car j'y suis malheureusement obligé, et que je sais quand même un petit peu le faire. Mais cela me demande d'avantage d'efforts et de travail et forcément, mes réponses s'en trouvent moins précises et performante.
Je fais de gros efforts pour cela. Vous pouvez donc vous même, soit faire quelques efforts pour traduire mes réponses dans votre langue (Google ne le fait pas si mal que cela) soit ne pas lire mes réponses Francophones.
Lorsque je participe aux discussions sur d'autres pages, je me plie aux langages employés, j'y écrit généralement en Anglais ou en Français suivant ce qui est possible, mais je me débrouille pour traduire quand leurs propriétaires utilisent d'autre langues. Il ne me viendrait pas à l'idée de critiquer celà. Et effectivement, si vous utilisiez le Portugais ici, je ne vous répondrais pas.
Les programmes que je crées et que je donne gracieusement à la comunauté sont libres, cela se traduit par "Free", ce qui peut avoir différents sens en Anglais... Mais cette liberté, et non cette gratuité, vous donne toute liberté de ne pas l'utiliser et d'aller voire ailleurs si ma langue ne vous plait pas!
@++;  
Gauthier.
<hr />
Hello,

<div align="center">Exceptional courtesy translation</div>

Since here we are, on the pages concerning the problems of the project of which I am the developer, I am French and I prefer to use French when possible, especially when my interlocutors are themselves French.
Since it is my mother tongue, I am much more precise and efficient in this language than in any other.
I agree to use English because, unfortunately I am obliged to, and I know a little bit how to do it. But this requires more effort and work and necessarily, my answers are less precise and efficient.
I am making great efforts for this. You can therefore yourself, either make some efforts to translate my answers into your language (Google doesn't do that bad) or not read my French-speaking answers.
When I participate in discussions on other pages, I bend to the languages ​​used, I generally write in English or French depending on what is possible, but I manage to translate when their owners use other languages. It would not occur to me to criticize this. And indeed, if you used Portuguese here, I would not answer you.
The programs that I create and that I give free of charge to the community are free (in the sense of freedom), the English "Free" translation can have different meanings... But this freedom, and not this gratuitousness, gives you all freedom not use it and go somewhere else if you don't like my language!
@++;
Gauthier.