fra589/grbl-Mega-5X #72: Onboard tachometer

thawkins, 2019-10-22:
I have some code, originaly built against grbl-mega, which adds an onboard tachometer to grbl-2560 varients. 

The system works with a  cheap chinese tcrt5000 based opto sensor ($2)  and a small piece of reflective tape applied to the collet barrel.  I 3d printed a mount that attaches to the bottom of a 52mm spindle. My current implementation uses timer5 and D11 on the ramps card.  It seems to be acurate up to about 20k rpm. (My spindle toos out at about 19.5k.) I have checked it using a syandalone optical tachometer and it seems to track the same resukts with remarkable accuracy. 

Currently it only adds an RPM field to the reports.c status line, but it could be used with a PID loop to allow setting of true rpm speeds via tbe S command using feedback from the tach. Response time is about 5 seconds, that is mostly due to the 4 sample moving averaging process designed to eliminate jitter. 

If i prepare a pull request would that be acceptable?

fra589, 2019-10-26:
Hi @thawkins,
Thanks for your interest in grbl-Mega-5X and proposing this evol.
In your pull request, don't forget to add the config.h #define items for enabling thoses functionalities, the code must not be changed if thoses option are not enabled.
I'm a little worried about the impact of this feature on grbl-Mega-5X performance, especially with the announced response time of about 5 seconds... We will see what it is, we will have to document it and maybe we can optimize it together.
@++;  
Gauthier.

MARIOBASZ, 2020-02-06:
Hi @Thawkins, can you tell me the changes to add tachometer?
I have little knowledge of programming, however, due to perseverance, I get results (some complements for bCNC, for example), external help will facilitate my task and I can solve it more quickly.
I am in the final stage of my universal ATC economic project, and taking the data from one or two sensors (optical and hal in principle) is something I want to add.
Greetings, Mario

thawkins, 2020-02-06:
My mods are in

https://github.com/thawkins/grbl-Mega

It based on a slightly older version.  I will dig out my notes that i made
at the time, i got diverted to setting up a startup, so i have not had much
time.