fra589/grbl-Mega-5X #73: Homing Y-X

Drew3164, 2019-10-26:
Hi, All

I'm having problems home X-Y I can manual move them but when it comes to homing just Z will home and not X-Y they seem to be making a high humming nose after homing. 

[image]

fra589, 2019-10-26:
Hi @Drew3164,  
It look like your hardware is unable to accept the speed and/or the acceleration you ask to.
perhaps the power is not enough for the need.
Try to lower speed ($24, $25 and $110, $111) or accelerations ($120, $121) parameters.
@++;  
Gauthier.

Drew3164, 2019-10-26:
Thanks for your help but still not working?

[image]

Drew3164, 2019-10-26:
Thanks for you help but its till not working?

[image]

Drew3164, 2019-10-26:
oh thanks I got it so what I did was I had to change it so that it was not trying to home X,Y at the same time instead one at a time seems to work now. 

[image]

fra589, 2019-10-27:
Hi @Drew3164,
Viewing your $$ output config, I think the individuals max speed parameters are too high and the Arduino exceed the maximum call frequency of interrupts timers when you try to move X and Y simultanousely.
With a large number of steps per unit and a high speed, the previous timer interrupt may not be completed when the next one is called. The ATMega processor architecture does not allow it.
Your parameters seems that the max speed for individuals axis are 2000 mm/mn. With 2067 steps per mm, you need nearly 70kHz for driving the max speed.
In the homing process, the global speed of movment is defined by the $24 parameter (200 mm/mn in your case), this limit the speed on individual axis, in your case, this need about 7kHz, your Arduino can do this without problem. But if you whant to drive the 2 X and Y simultanously, the frequency will be multiplied by 1.414 (diagonal of the square), this can overshoot the max possibility of the hardware.
You need to define 200mm/mn on each axis, then, the system will not overshoot his capacity.

@++;
Gauthier.

closed 2019-11-19 by fra589