fra589/grbl-Mega-5X #74: Problem with homing

kikirikiki, 2019-10-27:
Hi, I have a 4 axes configuration and when I start the homing cycle, the axe A start turning looking for a limit switch. When it find it, the motor start to turn in the opposite direction and it doesn't stop anymore. What is the problem in my configuration?
Thanks a lot!

fra589, 2019-10-27:
Hi, 
If you don't post your config.h axis and homing definitions and your $$ output parameters, I will have difficulties to help...
I am not diviner.
@++;
Gauthier.

kikirikiki, 2019-10-27:
Hi, 
thanks for your answer. How can I extract the config.h file from my arduino?

kikirikiki, 2019-10-28:
Here is my config.h file and the $$ output parameters.

[image]

[config.zip](https://github.com/fra589/grbl-Mega-5X/files/3778356/config.zip)

Thanks a lot!

fra589, 2019-10-28:
Hi @kikirikiki,
The output of $$ seems to have parameters a little crazy...
- $24 and $25 : homing feed and homing seek are too high with 1000 millimeters by minute, at this speed, the switch detection precision will not be enough to ensure accurate positions. Standards values are 25 mm/mn for $24 and 500 for $25 (°/mn for rotationals axis).
- the $27 value is probably those wich is your problem... When the homing movment reach the end stop switch, the movment start in the oposite direction for a distance of this value. In your case, it need to go to 1000° = 2.78 turns. But in the $133 parameter (max travel of axis 4), you specify 10 degres ! It's impossible to go for about 3 turn with only 10 degres of max travel :-)
- For the end, 50000mm/mn for your linears speeds ($110 to $112) can be impossible to obtain : Risk of exceeding the maximum call frequency of interrupts timers on Arduino. With a large number of steps per unit and a high speed, the previous timer interrupt may not be completed when the next one is called. The ATMega processor architecture does not allow it. The frequency is calculed as this :
f = STEP_PER_UNIT * UNIT_PER_SECOND
For you : f = 80 * 50000 / 60 = 66,7 kHz => it's about ten times too much for what the Arduino can do!
Same for the accel values ($120 to $122) = 1g: You will need very powerful engines to get there and I doubt that it is possible.
@++;
Gauthier.

kikirikiki, 2019-10-30:
Hi @fra589,

thanks a lot for your Answer. I could fix my problem and now everything is working!
Thank you very much!

Juan

closed 2019-10-30 by fra589