fra589/grbl-Mega-5X #76: Grbl Mega 5X and Laser PWM/TTL

JeffHank1, 2019-11-10:
I am trying to get a laser running on my RAMPS 1.4 board. I have seen Pinouts referring to both Pin 44 and Pin D8 to provide a PWM signal for a laser but neither seems to work. My laser is 12v with a third wire for TTL control It works properly on an Uno but I am having trouble convincing it to run on the RAMPS + Arduino Mega. I have tried connecting the TTL input to both Pin 44 in AUX 2 and Pin D8+ but the laser remains fully powered On. The power leads are connected directly to 12V+ and Gnd. $32=1
I have seem a comment suggesting:
Goto cpu_map.h, L361, change it to TCCR5B seems to fix this issue.
#define SPINDLE_TCCRB_REGISTER TCCR5B
Is this a fix for my problem? Am I missing something? Thank you.

JeffHank1, 2019-11-11:
I found a solution here: https://github.com/gnea/grbl-Mega/issues/55#issuecomment-403261212
I edited the cpu_map.h file, recompiled the Hex file in Arduino IDE and uploaded it with Xloader...easy
Seems to work very well. Somehow I can live with only 255 power levels...

fra589, 2019-11-24:
Hi @JeffHank1,
I'm glad you found the solution, do not forget to close the issue when is done.
@++;
Gauthier.

closed 2019-11-24 by fra589